ev3dev/ev3dev #237: Is it possible to install QtSixA from source to have dualshock 3 controller on ev3dev?

artpoz, 2015-01-01:
Hi
I want to write script in python that control motors using ps3 controller as joystick

1. I've installed python (2.7) and pygame on ev3dev 
2. I wrote a simple script to check if I have joystick connected,

    from pygame import joystick
    joystick.init() # main joystick
    try:
      j = joystick.Joystick(0)
      j.init()
      print 'Enabled joystick: ' + j.get_name()
   except pygame.error:
      print 'no joystick found.'

3. I've tried to install QtSixA from source on ev3dev

    wget http://sourceforge.net/projects/qtsixa/files/QtSixA%201.5.1/QtSixA-1.5.1-src.tar.gz
    tar xfvz QtSixA-1.5.1-src.tar.gz
    cd QtSixA-1.5.1/sixad
    make
    mkdir -p /var/lib/sixad/profiles
    checkinstall

but I have error with make 
root@ev3dev:~/QtSixA-1.5.1/sixad# make
-bash: make: command not found

Im using ev3dev-jessie-2014-10-07.img with apt-get update 
it shows (uname -rv)
3.16.1-4-ev3dev #2 PREEMPT Sun Sep 28 19:57:58 CDT 2014

What should I do to have gamepad visible in python? How to build QtSixA ?

moriarty, 2015-01-01:
Hi, 

I can't answer your question, I haven't used the QtSixA. But I can help you with the make error.
```
sudo apt-get install build-essential
```

But I think you should take a look at brickstrap if you're going to do a lot of building directly on the ev3, and instead build inside the brickstrap environment on a real pc and then either move the files over to the ev3 or create an sd card image with all the things you require. 

The link to the brickstrap instructions is [here in the getting started section](http://www.ev3dev.org/docs/getting-started/#writing-programs) on ev3dev.org. Or directly here on the older [github wiki](https://github.com/ev3dev/ev3dev/wiki/Using-brickstrap-to-cross-compile-and-debug)

dlech, 2015-01-01:
Is the controller USB or Bluetooth? If it is USB, what does `lsusb` say? If it is Bluetooth, did you pair it and connect it first? You can also check `dmesg` to see if it was recognized as an input device.

Also, you are several kernel versions behind - although I don't think it will make a difference for this. Try running `apt-get install linux-image-ev3dev` to upgrade to the most recent kernel. After this, `apt-get dist-upgrade` will keep the kernel upgraded.

moriarty, 2015-01-02:
A little off topic, but it might help so I thought I'd mention it. It doesn't use QtSixA or help you install it from source, but it might help you get the Dualshock 3 working on the ev3. 

I've used my Logitech usb joypad to control a motor on the ev3... with so many caveats that I decided not write it as a comment but [put it here](https://github.com/moriarty/ros-ev3/blob/master/ev3_ros_joypad.md)

Long story short, I use [ROS](http://www.ros.org), which provides on github
[ros-drivers](https://github.com/ros-drivers) a [joystick_drivers](https://github.com/ros-drivers/joystick_drivers) package which supports many common joypads, including the ps3 six axis. Using ROS is probably huge overhead for what you're doing... However...

I mention it because their ps3 joypad driver code is very basic and has minimal requirements, they have a ps3joy.py and sixpair.c which are ros independent and only require ```libusb-dev joystick bluez python-bluez``` 

Looking at the ROS ps3joy code & README the only thing they have used from ROS is the ```rosrun``` and ```catkin```. The first can be replaced by just running the ps3joy.py & sixpair programs directly and the second is a cmake based build system, which can be avoided if you just make their sixpair.c directly.

Hopefully this helps and doesn't waste your time browsing through heaps of code and documentation.

Also, from the QtSixA sourceforge comments, if you're using gcc > 4.7
```
cptny
Needs to include unistd.h in shared.h to compile in recent systems.
Posted 08/13/2014
```

artpoz, 2015-01-03:
Little update :)
I resigned from QtSixA and Python. I did it with **brickrc** from this project:
http://www.pabr.org/bricks/brickfpv/brickfpv.en.html

I just needed gamepad part of the project. Pairing worked well using **sixpair** , but brickrc connected with gamepad didn't move motors.  Fortunately source is included, so I had to modify output_ev3dev.c (with support of tacho-motor) and compile on brick with this:

gcc --std=gnu99 -Wall -o2 input_bthid.c output_ev3wifi.c output_ev3dev.c brickrc.c -o brickrc

I don't know why motors are shaking while stop, but the most important problem - gamepad connection directly to the EV3 brick - is solved :)

dlech, 2015-01-03:
Be sure to close this issue since it is solved. If you need help with the motor control just open a new issue.

closed 2015-01-03 by artpoz