ev3dev/ev3dev #238: Where to put own script on ev3dev startup?

artpoz, 2015-01-03:
It's not an issue but a question of linux beginner.

I've got program to control my robot and I want to start it automatically after boot ev3dev.
I need it for testing in outside conditions with no wifi, laptop, usb etc...
Where should I put my script or command to run? 

Is it possible to run own program by pressing some combination of buttons on brick?

dlech, 2015-01-03:
There is not a way to run programs from the brick itself yet. That is issue #129.

To get a program to run at boot, you can create a systemd service. For example, save this file as `/etc/systemd/system/myprogram.service`.

```
[Unit]
Description=My Program
After=multi-user.target
[Install]
WantedBy=multi-user.target
[Service]
Type=simple
ExecStart=/home/user/bin/myprogram
```

`myprogram` of course can be replace by the name of your program. Also, `ExecStart` is not run in a shell and there is no `PATH` environment variable, so you can't take any shortcuts. For example, if you program is a python script, you would have to write `ExecStart=/usr/bin/python /home/user/myprogram.py`.

artpoz, 2015-01-03:
I've created in  `/etc/systemd/system/brickrc.service`
with

    [Unit]
    Description=My Program
    After=multi-user.target

    [Install]
    WantedBy=multi-user.target

    [Service]
    Type=simple
    ExecStart=/root/brickfpv-1.0/src/brickrc -v /root/brickfpv-1.0/src/ps3_tank.config

But it didn't started after reboot :(

Just to be sure if program is ok I added beep sound at the beginning. When I log as root and write
`root@ev3dev:~# /root/brickfpv-1.0/src/brickrc -v /root/brickfpv-1.0/src/ps3_tank.config`
I can hear that sound. What am I missing here? Any ideas?

dlech, 2015-01-03:
I forgot a very important step. You have to enable the service.

   systemctl enable brickrc.service

artpoz, 2015-01-03:
It works. Big thanks! Im closing issue.

closed 2015-01-03 by artpoz

sharived, 2015-03-30:
I got this working as well, but I've been struggling with getting the output of my script that (I believe) is going to stdout.  I've successfully redirected stderr using &> in my script - as I've already spent several hours on this, was wondering if anyone can give me some pointers.  Note I've tried setting StandardOutput= in my service file, but not exactly sure where to put it or what it should really be set to.

In my case 
ExecStart=/bin/bash /root/script.sh

and script.sh is
#!/bin/bash

sudo /usr/bin/python3 /root/test.py &> /root/out.txt

dlech, 2015-03-30:
systemd automatically logs the output of a service. just run `systemctl status -l my.service` and you will see the output. You don't need to redirect stderr in your script or use StandardOutput=

sharived, 2015-03-30:
unfortunately my stdout still doesn't show up in the systemctl status - if I run the script manually it is printed, but not if I run as a service.  I do get the stderr output though.  Any other suggestions - I guess I can put all my output to stderr, but it seems wrong...

dlech, 2015-03-30:
Check `journalctl` instead.

http://unix.stackexchange.com/questions/20399/view-stdout-stderr-of-systemd-service?rq=1

sharived, 2015-03-30:
Unfortunately this has the same issue - I'll just write directly to a file instead of stdout

giovannilugaresi, 2019-06-07:
> I forgot a very important step. You have to enable the service.
> 
> ```
> systemctl daemon-reload
> systemctl enable brickrc.service
> ```

I have tried this but it gives me permission denied. Running it from sudo the "systemctl daemon-reload" works correctly but the "systemctl enable brickrc.service" does not. Gives me this error:

Failed to execute operation: No such file or directory

Any ideas?

Big thanks in advance!