ev3dev/ev3dev #239: Upgrade throws errors

antosop, 2015-01-03:
image: ev3dev-jessie-2014-12-01
kernel: 3.16.1-7-ev3dev #2 PREEMPT Tue Nov 25 11:24:38 CST 2014

http://pastebin.com/raw.php?i=vm0zE5gm

<h2>Problems</h2>
1. The upgrade tries to use apt-utils which is not present.
2. dbus does not seem to be upgrading correctly which could be the cause of some bluetooth errors. 

Also It seems that /etc/bluetooth/main.conf has nothing but comments in them. Is this intended?

dlech, 2015-01-03:
1. `apt-utils` is optional, so this is a warning, not an error.
2. You always need to run `apt-get update` before `apt-get upgrade`. It is probably trying to install a package version that no longer exists online in the debian package repository.
3. The file `/etc/bluetooth/main.conf` comes from the `bluez` package and has not been modified, so this is expected.

antosop, 2015-01-03:
OK, just checking. I did run update first if that changes anything.

dlech, 2015-01-03:
Run `apt-get install -f` to patch things up and then run `apt-get update` and `apt-get upgrade` again.

dlech, 2015-01-03:
Looks like you are not the only one with the dbus package problem: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=774124

rhempel, 2015-01-03:
I ended up having to do:

```
apt-get install libdbus-1-3
```
And then the upgrade worked.

closed 2015-01-04 by antosop