ev3dev/ev3dev #242: kernel "Internal error: Oops" on setting dc-motor (for PF-motor)

BertLindeman, 2015-01-05:
Running `Linux ev3dev 3.16.1-8-ev3dev #2 PREEMPT Sat Dec 27 14:31:18 CST 2014 armv5tejl GNU/Linux`

tail of dmesg on FIRST attempt:

```
[73124.079058] lego-port port5: Trying to register an invalid motor type on port5.
```

tail of dmesg on SECOND attempt:

```
[73310.109824] lego-port port5: Added new device 'outB:rcx-motor'
[73310.361696] dc-motor motor0: Bound to device 'outB:rcx-motor'
[73311.421093] Unable to handle kernel NULL pointer dereference at virtual address 0000022c
[73311.421150] pgd = c090c000
[73311.421176] [0000022c] *pgd=c3991831, *pte=00000000, *ppte=00000000
[73311.421243] Internal error: Oops: 17 [#1] PREEMPT ARM
[73311.421267] Modules linked in: rcx_motor dc_motor_class nxt_i2c_sensor servo_motor_class i2c_legoev3 rcx_led cfg80211 rfcomm bnep usb_f_eem g_ether usb_f_rndis u_ether libcomposite configfs hci_uart bluetooth rfkill nfsd nxt_analog_sensor lego_sensor_class 8192cu(O) snd_legoev3 snd_pcm suart_emu snd_timer snd soundcore legoev3_ports lego_bus lego_port_class legoev3_battery legoev3_bluetooth legoev3_analog fuse
[73311.421614] CPU: 0 PID: 17052 Comm: cat Tainted: G           O  3.16.1-8-ev3dev #2
[73311.421647] task: c395db00 ti: c39a8000 task.ti: c39a8000
[73311.421749] PC is at ev3_output_port_get_command+0xc/0x14 [legoev3_ports]
[73311.421805] LR is at command_show+0x20/0x4c [dc_motor_class]
[73311.421845] pc : [<bf033224>]    lr : [<bf28119c>]    psr: a0000013
sp : c39a9e58  ip : c39a9e68  fp : c39a9e64
[73311.421871] r10: c3a50018  r9 : 00000001  r8 : 00001000
[73311.421897] r7 : c13dca40  r6 : c3a68000  r5 : c03e168c  r4 : c3a68000
[73311.421922] r3 : bf033218  r2 : c3a68000  r1 : bf281b68  r0 : 00000000
[73311.421950] Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[73311.421978] Control: 0005317f  Table: c090c000  DAC: 00000015
[73311.422005] Process cat (pid: 17052, stack limit = 0xc39a81c0)
[73311.422031] Stack: (0xc39a9e58 to 0xc39aa000)
[73311.422065] 9e40:                                                       c39a9e7c c39a9e68
[73311.422120] 9e60: bf28119c bf033228 bf28117c bf281b68 c39a9e94 c39a9e80 c022d89c bf28118c
[73311.422172] 9e80: c31da120 c03e168c c39a9ebc c39a9e98 c010d244 c022d884 c31da120 00020000
[73311.422224] 9ea0: c39a9f80 00000000 00000001 b6bd5000 c39a9ecc c39a9ec0 c010bb58 c010d1d0
[73311.422276] 9ec0: c39a9f1c c39a9ed0 c00d0660 c010bb3c 00000000 00000817 c31da150 c3b80d20
[73311.422328] 9ee0: c0518fa8 c39a9fb0 00000000 00000000 c39a9fac c3b80d20 c13dca40 c39a9f80
[73311.422380] 9f00: 00020000 c00096c4 c39a8000 b6bd5000 c39a9f4c c39a9f20 c010c524 c00d04c4
[73311.422431] 9f20: c39a8000 00000000 00020000 c3b80d20 b6bd5000 c39a9f80 c00096c4 00000000
[73311.422484] 9f40: c39a9f7c c39a9f50 c00ae998 c010c4e4 c00c99b0 c00c9934 c3b80d20 c3b80d20
[73311.422536] 9f60: 00020000 b6bd5000 c00096c4 00000000 c39a9fa4 c39a9f80 c00aebbc c00ae908
[73311.422585] 9f80: 00000000 00000000 00020000 00020000 b6bd5000 00000003 00000000 c39a9fa8
[73311.422636] 9fa0: c0009560 c00aeb84 00020000 00020000 00000003 b6bd5000 00020000 000291cc
[73311.422684] 9fc0: 00020000 00020000 b6bd5000 00000003 7fffe000 00000000 00000000 00020000
[73311.422733] 9fe0: 00000000 beb18284 00014854 b6e7f0ec 60000010 00000003 00000000 00000000
[73311.422746] Backtrace:
[73311.422879] [<bf033218>] (ev3_output_port_get_command [legoev3_ports]) from [<bf28119c>] (command_show+0x20/0x4c [dc_motor_class])
[73311.422988] [<bf28117c>] (command_show [dc_motor_class]) from [<c022d89c>] (dev_attr_show+0x28/0x4c)
[73311.423012]  r4:bf281b68 r3:bf28117c
[73311.423089] [<c022d874>] (dev_attr_show) from [<c010d244>] (sysfs_kf_seq_show+0x84/0xf0)
[73311.423108]  r5:c03e168c r4:c31da120
[73311.423191] [<c010d1c0>] (sysfs_kf_seq_show) from [<c010bb58>] (kernfs_seq_show+0x2c/0x30)
[73311.423212]  r10:b6bd5000 r8:00000001 r7:00000000 r6:c39a9f80 r5:00020000 r4:c31da120
[73311.423335] [<c010bb2c>] (kernfs_seq_show) from [<c00d0660>] (seq_read+0x1ac/0x3fc)
[73311.423398] [<c00d04b4>] (seq_read) from [<c010c524>] (kernfs_fop_read+0x50/0x15c)
[73311.423417]  r10:b6bd5000 r9:c39a8000 r8:c00096c4 r7:00020000 r6:c39a9f80 r5:c13dca40
[73311.423487]  r4:c3b80d20
[73311.423560] [<c010c4d4>] (kernfs_fop_read) from [<c00ae998>] (vfs_read+0xa0/0x148)
[73311.423578]  r10:00000000 r8:c00096c4 r7:c39a9f80 r6:b6bd5000 r5:c3b80d20 r4:00020000
[73311.423686] [<c00ae8f8>] (vfs_read) from [<c00aebbc>] (SyS_read+0x48/0x84)
[73311.423704]  r10:00000000 r8:c00096c4 r7:b6bd5000 r6:00020000 r5:c3b80d20 r4:c3b80d20
[73311.423813] [<c00aeb74>] (SyS_read) from [<c0009560>] (ret_fast_syscall+0x0/0x2c)
[73311.423830]  r7:00000003 r6:b6bd5000 r5:00020000 r4:00020000
[73311.423911] Code: e89da800 e1a0c00d e92dd800 e24cb004 (e590022c)
[73311.427826] ---[ end trace f41781e7b8cca372 ]---
```

Scenario to set PF-motor on outB
Running as root:
```
cd /sys/class/lego-port/port5
echo rcx-motor > mode
```

The THIRD time hangs the user session, does not respond to ^c
Does not end on kill -9
In fact better: reboot..

Problem does not  immediately occur after the `echo`
but at the moment I look at the values:
```
cat /sys/class/dc-motor/motor0/*
```
And then each time I do the `cat`

dlech, 2015-01-07:
Fix released in kernel v3.16.7-ckt3-ev3dev1

This was actually 2 separate bugs. One for the kernel oops and one for the mode not setting on the first try.

closed 2015-01-07 by dlech

BertLindeman, 2015-01-07:
Thanks David, 
Confirmed: both problem gone now.