ev3dev/ev3dev #243: Hard freeze/hang issue with NXTcam/I2C devices (have to remove battery)

dsharlet, 2015-01-07:
I'm trying to get NXTcam to work in C++ on ev3dev. I've basically got it working (it's giving me tracking data), but I have a really annoying problem: after closing my program using NXTcam, there is a sporadic issue where the brick freezes hard, such that I have to remove the battery to reboot it. No SSH/user interface responds to any kind of input.

An interesting thing I found is that this commit:

https://github.com/dsharlet/DeltaRobot/commit/c842ef9c86aeae7781d57df6fc605d4984b5cb62

Seemed to significantly reduce the probability of this occurring, but it still does happen. I can't say my sample size is very large (basically an evening of having to remove the battery 20-30 times) so it could just be random chance.

I've also been working on programming another I2C device, one based on the 8574 serial IO expander. I haven't seen any similar issues there, though I'm using a slightly different API to talk to it (still using open/ioctl, but using the I2C_SMBUS commands instead of the I2C_RDWR command, I wasn't able to get NXTcam to work with the former so I switched to the latter).

I just ran a full apt-get update/upgrade and I still see the same behavior.

I have two motivations for the issue: 

- Reporting this problem regarding hard freezes/hangs. This seems like a serious bug. I really don't think I'm doing anything crazy enough to justify this as working behavior.
- Find out if anyone else has built a driver/code to support NXTcam in ev3dev?

dlech, 2015-01-07:
I do actually have a mindsensors.com NXTCam, so I am planning on writing a driver. I am going to try to use the [lego-sensor class](http://www.ev3dev.org/docs/drivers/lego-sensor-class/). Hopefully I can make it work.

The problem is:

    #define I2C_RDWR	0x0707 /* Combined R/W transfer (one STOP only) */

The key words being "one STOP only". The EV3 I2C implementation currently does not know how to not stop between read and write, so this is probably where the hang up is coming from. So, you need to do your read and write separately. In the kernel, we use `i2c_smbus_read_i2c_block_data()`, so if this function is available, use it. Otherwise you will have to write and read in separate functions.

dsharlet, 2015-01-08:
Thanks for the tip. It makes sense. I modified my code to call ioctl twice, once for each packet, and it seems to be working. However, the issue was sporadic, so only time will tell :)

It's good to hear you are thinking about building an NXTcam driver, it will surely be much better than my attempt. 

BTW, I just took a quick look at the lego-sensor docs you linked. The interface seems sufficient to support nxtcam. However, an issue I would have with using that is: the docs say the minimum polling period is 50 ms, but I want to poll the camera at at least 30 Hz (33 ms polling period), the rate that the camera claims to perform tracking. I will probably just continue to use my nxtcam class (if it is indeed working now without hanging) so this doesn't bother me too much, but something to keep in mind for others :)

dlech, 2015-01-08:
I am reopening this since a hard lockup like this should be fixed. Performing an unsupported I2C operation should just return an error, not lock up the EV3.

Also, the 50ms hard limit was arbitrary - just to keep people from entering an unreasonable number and getting errors as a result. If I did my math right, 30ms should be OK instead of 50ms, so we can change that.

dsharlet, 2015-01-17:
I'm still having intermittent hard lockups very similar to the ones I originally reported here. They're much more rare, but they do still occur. I can only assume that they are similar to this issue because of how similar the causes/symptoms are to the original bug. 

They're rare enough that it's just mildly annoying that they happen, but it isn't easy to remove the battery from my robot so it's a bit of a pain.

I can't provide much more information than I did before. The only change is I'm now only using one packet per I2C_RDWR. Let me know if there is any other information I can provide that will help with this.

dlech, 2015-01-17:
You can press and hold the back and enter buttons to reset instead of removing the battery.

BertLindeman, 2015-01-18:
Had a few hard-hangs lately (once every few days)

Using `3.16.7-ckt3-ev3dev1`
During this time I have an NXT-US sensor and an NXT-sound sensor routinely connected.
They just are connected, no user action done to them, except to look at the status, like 

```
cat /sys/class/lego-sensor/sensor0/*
cat /sys/devices/platform/legoev3-ports/lego-port/port3/*
```
But  I did *not* do this at the time the lockup occured.

To recover from the hang I pressed and held the EV3-buttons `back` and `enter` 
until the leds turned red and the display shows `Starting`
But the status did not change from `Starting` so had to remove / re-place the battery.to restart.

I do not know how to get info about `what task was active` at a previous hangup. 

Will keep these sensors OFF the EV3 for a few days to see if the hang still occus.

BertLindeman, 2015-01-19:
Hang-up re-occured here. 
NO sensors connected since previous remark.

My last actions were:
```
echo auto > /sys/devices/platform/legoev3-ports/lego-port/port4/mode
```
to remove the PF-leds on outA

and
```
journalctl -f -k
```
to see what messages came-up on plugging in a minitacho into port outD.
I am not really sure if the freeze was immediate or that it took some seconds.

Last lines on two separate sessions:
```
09:21:09 root@ev3dev:~# ll /sys/devices/platform/legoev3-ports/lego-port/port4/
total 0
lrwxrwxrwx 1 root root      0 Jan 18 16:25 device -> ../../../legoev3-ports/
-r--r--r-- 1 root ev3dev 4096 Jan 18 16:25 driver_name
-rw-rw-r-- 1 root ev3dev 4096 Jan 18 16:36 mode
-r--r--r-- 1 root ev3dev 4096 Jan 18 16:25 modes
drwxr-xr-x 4 root root      0 Jan 18 16:36 outA:rcx-led/
-r--r--r-- 1 root ev3dev 4096 Jan 18 16:25 port_name
drwxr-xr-x 2 root root      0 Jan 18 16:25 power/
--w--w---- 1 root ev3dev 4096 Jan 18 16:25 set_device
-r--r--r-- 1 root ev3dev 4096 Jan 18 16:25 status
lrwxrwxrwx 1 root root      0 Jan 18 16:24 subsystem -> ../../../../../class/lego-port/
-rw-r--r-- 1 root root   4096 Jan 18 16:24 uevent
09:21:14 root@ev3dev:~# echo auto > /sys/devices/platform/legoev3-ports/lego-port/port4/mode
09:21:30 root@ev3dev:~#

```

```
Jan 19 09:20:22 ev3dev kernel: tacho-motor motor4: Bound to 'outD:ev3-tacho-motor'.
Jan 19 09:20:22 ev3dev kernel: lego-port port7: Added new device 'outD:ev3-tacho-motor'
^C
09:20:34 root@ev3dev:~# journalctl -f --unit=kernel
-- Logs begin at Sat 2000-01-01 01:00:08 CET. --
^C
09:21:40 root@ev3dev:~# man journalctl
09:22:42 root@ev3dev:~# journalctl -f -k
```

Press and hold the back and enter buttons only stops linux and the brick shows "Starting" for over 10 minutes now, so battery OUT and IN to re-boot.

Wonder if there might be a relation to #245 ???

dlech, 2015-01-19:
> I do not know how to get info about what task was active at a previous hangup.

You need a serial port adapter and a kernel that has lockup detection turned on.

> Wonder if there might be a relation to #245 ???

Yes, what you are describing seems to be more related to that issue and not this issue. Or perhaps a new issue.

BertLindeman, 2015-01-20:
Thanks David.

Have opened a new issue #252 for "my" hangup as it occurs again after
upgrade to kernel v3.16.7-ckt4-ev3dev1

dsharlet, 2015-01-26:
Just to add more information about this: I re-wrote my nxtcam class to use a different method of I2C communication to try to workaround this issue (it is becoming quite annoying). I still get the same intermittent hard lockups by using the I2C_SLAVE + read/write API for talking to I2C devices.

BTW, thanks for the tip about holding back + enter to reboot. This is MUCH better than removing the battery on my robot :)

dlech, 2015-01-26:
Can you post a sample program that causes a lockup all the time (or at least most of the time) so that I can dig into this more?

dsharlet, 2015-01-26:
I put everything relevant into a single C++ file, the only non-standard/non-linux dependency is ev3dev.h.

http://pastebin.com/hVRXJ62R

I had to run this about 10 times before I got a lockup. That seems like it's maybe a bit less often than I see with my main project, which is using two NXTcams. But, it's small sample size, so I can't say conclusively that the likelihood of this occurring is related to the number of nxtcam instances.

The lockups occur either immediately upon exiting (Ctrl+C), or a few seconds later. Sometimes I can run another program after exiting for a few seconds before the lockup occurs.

Let me know if there is anything else I can do to help.

dlech, 2015-01-26:
I've run this maybe 30 times and haven't had a hang up yet. I did make one change - I am using input port 4 instead of input port 1 so that I can have a serial port attached to input port 1. Can you try this and see if it makes a difference? Any other ideas on how to reproduce the problem? Do I have to let it run a long time before CTRL+C?

dsharlet, 2015-01-26:
My actual project actually has two nxtcams, one on port 1 and one on port 4. I will try again with only one NXTcam connected (to port 4) and see if that makes a difference.

I don't have to let it run for a long time before hitting Ctrl+C. I basically run it just long enough to see its actually reading measurements from the camera, then terminate it.

dlech, 2015-01-26:
OK, that is what I have been doing.

Note to self:

    gcc -std=c++11 -D_GLIBCXX_USE_NANOSLEEP -I. -lstdc++ nxtcam_ev3dev_hang.cpp

dsharlet, 2015-01-26:
I was just able to reproduce the problem after 10 executions, with nothing attached (no motors or sensors) except the NXTcam to port 4.

More info about my environment/setup:

- I'm running everything via SSH (windows 7, secure shell tab in chrome).
- I'm using a USB wifi dongle to connect to my computer (but I was using the USB/RNDIS networking in the past and observed the same issues).

dlech, 2015-01-26:
Hey! I finally got it to crash. :boom:

dsharlet, 2015-01-26:
By crash, do you mean it gave a segfault? I just got that behavior for the first time trying to reproduce this from a fresh boot with no devices but the NXTcam attached.

A few execs later, it locked up.

dlech, 2015-01-26:
I mean the kernel crashed, which is what causes the lockup. But there does seem to be a segfault involved, now that you mention it. I'm still trying to decipher the stack track. It's weird.

dsharlet, 2015-01-26:
Cool, I'm glad you were able to reproduce it. Sorry I could not make a more reliable reproducer.

Thank you so much for looking at this!! 

If there's anything else I can do to help let me know.

dsharlet, 2015-02-03:
I'm not sure how helpful this will be since it seems the reproducibity is different for you than for me, but after working with this bug for a while, it really doesn't seem quite fully random. For example, it has *never* crashed with fewer than 4 or 5 runs, and it also never seems to make it past 8-10. That doesn't seem to fit any simple probability distribution one might except if the crashes were independent of previous runs. In other words, I think that there is some kind of state that is accumulating between runs, and eventually going bad somehow.

Hopefully I'm not just imagining this :)

dlech, 2015-02-13:
I have absolutely no idea how, but I think I have fixed this. I fixed some other stuff that needed to be done in https://github.com/ev3dev/ev3dev-kernel/commit/f07bdba4758d049201a452008d9cec9eab9f176d and this problem seems to have gone away.

You can download and install [this kernel](https://www.dropbox.com/s/ljn3pfg46omebsc/linux-image-3.16.7-ckt4-ev3dev2-rc1-dlech%2B_1_armel.deb?dl=0) and try it out. (It does not include some wifi drivers, so your wifi might not work.)

dsharlet, 2015-02-17:
To install this, would I just use dpkg -i like any other .deb? Is it possible to roll this change back if the wifi drivers I need are gone?

Also, how long until this commit makes its way to the apt-get repository?

dlech, 2015-02-17:
Yes, you install it with `dpkg -i` and remove it with `apt-get remove`. When you install or remove it, it will trigger `flash-kernel` which will install the most recent kernel it can find. 

Installing this package does not write over your existing kernel, so when you uninstall it, your previous kernel will still be there.

If you test it and it works better for you, I will probably do another kernel release by the end of the week. I've got a couple of other fixes lined up already, but I am waiting on feedback on this issue.

dsharlet, 2015-02-18:
Great, thanks! I will test it and report back later tonight.

And, thank you for working on this issue! I really appreciate it.

dsharlet, 2015-02-18:
The issue appears to be resolved with the .deb you posted!! I ran my app probably 40-50 times without a hang. 

However, it is indeed missing my wifi driver, so I want to revert it... what exactly do I pass to apt-get remove? Are these the package names (from dpkg --info)?

    linux-image, linux-image-2.6, linux-modules-3.16.7-ckt4-ev3dev2-rc1-dlech+

Thanks again for working on this!

dlech, 2015-02-18:
Sweet! If only I could figure out what exactly I did that fixed it. :blush: 

To uninstall, I think this should do the trick:

    sudo apt-get remove linux-image-3.16.7-ckt4-ev3dev2-rc1-dlech+

dsharlet, 2015-02-18:
Huh, I thought I tried that... I must have missed the + at the end or something. Anyways, I'm back to normal with WiFi, so thanks again. 

I would stick with USB connection, but the USB cable obstructs the motion of the arms of my robot :(

dlech, 2015-02-22:
Fix has been release in [v3.16.7-ckt6-ev3dev1](http://www.ev3dev.org/news/2015/02/22/Kernel-Release-v3.16.7-ckt6-ev3dev1/).

closed 2015-02-22 by dlech

dsharlet, 2015-02-22:
Awesome, thank you! Installing now.

BTW, what I've been trying to build is now finally working: https://www.youtube.com/watch?v=Y0w_cRt7RzI Thanks again for your help on this and my other issues!

dlech, 2015-02-22:
Super!

You just need to teach it to throw the ball back. :laughing:

dlech, 2015-02-22:
I just realized I forgot to update the `linux-image-ev3dev` package. It won't upgrade automatically until I do this, so give it another 10 minutes or so to get that pushed out.

rhempel, 2015-02-23:
Wow! I'll be adding this to my Facebook updates tomorrow - great project - all using ev3dev?

dsharlet, 2015-02-23:
All of the code is written in C++ running on ev3dev :) I was somewhat surprised but the EV3 processor is powerful enough to do the trajectory estimation, even without floating point hardware. I'm even running my camera calibration problem onboard the EV3! That takes a few minutes to solve though. 

I'm working on making the code available, hopefully within a week or two I will make it public. I'm working on pulling some of the code out into a library for supporting stereo vision in ev3dev. The library also has some other bits, like the custom motor controller from my other issue (#261).