ev3dev/ev3dev #244: Brickman shows connected even after usb cable removed

ameysutavani, 2015-01-08:
When I remove the usb connection this happens :
[image]
[image]

dlech, 2015-01-08:
The IP address should probably go away when you disconnect the cable. However, the checkbox next to Gadget in the Networking menu just means that Gadget is *enabled*, it does not have anything to do with the connection state.

P.S. You can use `fbgrab <some-name>.png` to get screenshots on the EV3.

ameysutavani, 2015-01-09:
ok

ameysutavani, 2015-01-09:
no the ip address does not go away when i disconnect the usb cable

I have now updated to 3.16.7 ckt with brickman 0.3.0, still the bug persists

dlech, 2015-01-21:
I've dug into this a bit. The problem is actually on the kernel level, so nothing we can do about it in brickman. Basically the problem is [this](http://e2e.ti.com/support/embedded/linux/f/354/t/65332). The code mentioned in the link is the `da8xx_musb_set_mode` function [here](https://github.com/ev3dev/ev3dev-kernel/blob/ev3dev-jessie/drivers/usb/musb/da8xx.c#L386). The change would be deleting everything in the function body except for `return 0`. However, I am concerned that making the changes might cause a regression on #43.

If anyone wants to compile their own kernel and try this, let me know.

dlech, 2015-01-21:
This was easy enough that I decided to test it out myself, so I am going to go ahead and plan on including these changes next kernel release. It would be nice if 1 or 2 people using Windows would try it out before then to make sure we haven't caused a regression, but I think it should be OK.

rhempel, 2015-01-21:
I've got a Windows 8 Tablet  to test this out on, and a Windows 7 VM as well

WasabiFan, 2015-01-21:
@dlech Small nit-pick: I think "commited" is supposed to be spelled with two "t"s (in the label name).

dlech, 2015-02-22:
Fix has been release in [v3.16.7-ckt6-ev3dev1](http://www.ev3dev.org/news/2015/02/22/Kernel-Release-v3.16.7-ckt6-ev3dev1/).

closed 2015-02-22 by dlech