ev3dev/ev3dev #245: issue with minitacho motor (kernel 3.16.7-ckt3-ev3dev1)

caldes, 2015-01-11:
I have installed ev3dev-jessie-2014-12-01

After installation minitacho motors work fine.

Then I run:
apt-get update
apt-get install g++
apt-get install make
apt-get install libdbus-1-3 (to avoid issue with libdbus during upgrade)
apt-get upgrade
reboot

At this point I have tested again and minitacho motors still work fine.

However, if I run apt-get dist-upgrade and reboot, now the minitacho motors are loaded as normal tacho motor and not minitacho (type attirbute of /sys/class/tacho-motor). So I cannot connect them in my code.

Thanks for your help!

dlech, 2015-01-11:
Just to be clear here, we are talking about kernel version 3.16.7-ckt3-ev3dev1. I am not able to reproduce this problem. If I plug in a LEGO Medium motor that gets registered as motor0, then `cat /sys/class/tacho-motor/motor0/type` returns `minitacho` as expected.

Can anyone else reproduce this problem on this kernel version? There were some changes in the motor auto-detection code, so it is possible that something broke, but I can't fix it unless I can reproduce it.

caldes, 2015-01-11:
Yes, kernel is 3.16.7-ckt3-ev3dev1. I can upload the two images of the 8Gb MicroSD card (before and after dist-upgrade) if this could help identify the problem.

dlech, 2015-01-12:
> I can upload the two images...

No, this would not help.

It would be helpful of others could test this with this kernel version and confirm what they are seeing one way or the other.

rhempel, 2015-01-12:
I'll upgrade the kernel next, but with
```
Linux ev3dev 3.16.1-8-ev3dev #2 PREEMPT Sat Dec 27 14:31:18 CST 2014 armv5tejl GNU/Linux
```
The type is reported correctly as minitacho

rhempel, 2015-01-12:
```rhempel@ev3dev:~$ cat /sys/class/tacho-motor/motor0/type
minitacho
rhempel@ev3dev:~$ uname -a
Linux ev3dev 3.16.7-ckt3-ev3dev1 #2 PREEMPT Tue Jan 6 19:36:52 CST 2015 armv5tejl GNU/Linux
rhempel@ev3dev:~$ ^C
rhempel@ev3dev:~$ 
```
Works fine here with the new kernel as well - are you sure you are looking at the right motor/port?

caldes, 2015-01-12:
It works fine with 3.16.1-7-ev3dev on mine, too. I have two SD cards (the second with 3.16.7-ckt3-ev3dev1) so I'm pretty sure that I'm looking at the right motor/port since I'm not changing the hardware configuration.

caldes, 2015-01-12:
Yes, I'm pretty sure. I have two SD cards, one with 3.16.7-ckt3-ev3dev1 and one with 3.16.1-7-ev3dev. I'm not changing the hardware configuration (motor/port). Only the software.

rhempel, 2015-01-12:
Please look at the `port_name` attribute for the mini-tacho motor, and confirm it's the right port. Also, please remove all the sensors and motors from the brick, reboot, then plug in JUST the mini-tacho motor, and check the `type` attribute for `motor0`.
You don't normally have to reboot for this kind of test, but we want to be sure we're starting at the same page :-)

caldes, 2015-01-14:
I have one minitachomotor connected on A and 2 tachomotors on B and C. I start with my regular image (kernel 3.16.1-7-ev3dev). These are the commands:

root@ev3dev:/sys/class/tacho-motor# ls
motor0  motor1  motor2
root@ev3dev:/sys/class/tacho-motor# cat motor0/type
minitacho
root@ev3dev:/sys/class/tacho-motor# cat motor0/port_name
outA
root@ev3dev:/sys/class/tacho-motor# cat motor1/type
tacho
root@ev3dev:/sys/class/tacho-motor# cat motor1/port_name
outB
root@ev3dev:/sys/class/tacho-motor# cat motor2/type
tacho
root@ev3dev:/sys/class/tacho-motor# cat motor2/port_name
outC
root@ev3dev:~# apt-get dist-upgrade (I had already done update/upgrade)
root@ev3dev:~# reboot (I'm not changing any connections, i.e., minitachomotor connected on A and 2 tachomotors on B and C)
root@ev3dev:/sys/class/tacho-motor# ls
motor0
root@ev3dev:/sys/class/tacho-motor# cat motor0/type
minitacho
root@ev3dev:/sys/class/tacho-motor# cat motor0/port_name
outA

The first time I did this test (my first comment on this issue) the outcome was that the type of motor0 was tacho instead of minitacho (port_name was A) but I could see all three motors. Now the problem is that the two motors on B and C are not connected even though they are plugged in. 
If I unplug the minitacho from A only:

root@ev3dev:/sys/class/tacho-motor# ls
motor1
root@ev3dev:/sys/class/tacho-motor# cat motor1/type
tacho
root@ev3dev:/sys/class/tacho-motor# cat motor1/port_name
outB

so now the second motor is connected but not the third. If I unplug also the second motor, no motors are connected even though there is still the third motor plugged in on C.

I tried to repeat this same test as suggested without any motor connected and the outcome is similar. When I plug in minitachomotor on A and tachomotors on B and C after rebooting:

root@ev3dev:~# apt-get dist-upgrade (I had already done update/upgrade)
root@ev3dev:~# reboot (no motors are plugged in)
root@ev3dev:/sys/class/tacho-motor# ls (After reboot I plugged in the three motors as above)
motor0  motor1
root@ev3dev:/sys/class/tacho-motor# cat motor0/type
minitacho
root@ev3dev:/sys/class/tacho-motor# cat motor0/port_name
outA
root@ev3dev:/sys/class/tacho-motor# cat motor1/type
tacho
root@ev3dev:/sys/class/tacho-motor# cat motor1/port_name
outC

No sign of the motor connected on B.

dlech, 2015-01-15:
I probably broke something when I was fixing something else in this commit: https://github.com/ev3dev/lego-linux-drivers/commit/49a1f89e184a6618cac8bb77689b170109efd483

dlech, 2015-01-20:
I have just released [kernel v3.16.7-ckt4-ev3dev1](http://www.ev3dev.org/news/2015/01/19/Kernel-Release-v3.16.7-ckt4-ev3dev1/) which should fix this issue. Please try it out and close the issue if it fixes the problem.

dsharlet, 2015-01-21:
I'm not sure if this problem was 100% repeatable for me before (exactly which motors/ports failed to autodetect varied), but it seems to be fixed now. I haven't seen an issue here after a few boots, and it frequently occurred before.

caldes, 2015-01-30:
I did some tests and now it works fine. Thank you!

closed 2015-01-30 by caldes