ev3dev/ev3dev #246: How to start my programs from the Brickman menu?

jesper-friis, 2015-01-12:
A typical situation is that you build a robot and make a few programs for controlling it. Thereafter it would be nice to be able to select and run the programs in off-line mode.

The most natural would be a "Programs" menu in Brickman from which you can select and run your program. Is that possible to accomplish without rebuilding Brickman? If not, do you have suggestions for other ways to start your programs in off-line mode?

My current "poor man" solution is to start a small program in /etc/rc.local that listen to the touch sensor. A single press starts program 1, and a double press starts program 2. I am sure that this
can be done in a more natural and elegant way.

PS. Many thanks for your great work on ev3dev!

dlech, 2015-01-12:
It has not been implemented yet. See #129.

Closing as duplicate.

closed 2015-01-12 by dlech

dlech, 2015-01-12:
Also see #238 for another way to make a startup script.