ev3dev/ev3dev #247: The EV3 infrared sensor doesn't show up in /sys/class/msensor

AndyQin, 2015-01-13:
The touch sensor show up no problem at all. The infrared sensor (that comes with the package) doesn't.
I did apt-get update/upgrade, but problem persists.
I tried using the infrared sensor with Lego native code, it works without issue.
Is this a known problem?

dlech, 2015-01-13:
Please follow the suggestions [here](https://github.com/ev3dev/ev3dev/blob/master/CONTRIBUTING.md) when submitting an issue.

Also check your version of the `ev3dev-base` package (`dpkg-query -s ev3dev-base`).

I am guessing that you did `apt-get upgrade` without also doing `apt-get dist-upgrade`. `ev3dev-base` version 8.0 requires kernel version 3.16.1-8-ev3dev or higher.

AndyQin, 2015-01-15:
Thanks! The apt-get dist-upgrade solved the problem. Now all sensors appears under the lego-sensor.

closed 2015-01-15 by AndyQin