ev3dev/ev3dev #248: Issue detecting motors with kernel v3.16.7-ckt3-ev3dev1_1

sharived, 2015-01-14:
I just updated to the latest code yesterday and now I'm having trouble with my tacho-motors.  (I had noticed the changes for sensors and updated my code to reflect these).  I didn't see any changes to the motor code that - but maybe I missed something.

I am using ev3 large motors and nxt motors.  when I look in /sys/class/tacho-motor I am not seeing all of the motors that are connected.
First I saw motors in B & C (didn't show A), I rebooted and now it only likes the motor in C.  If I move the working motor, I will see it in the other ports.  So it seems like all of the ports are working.  It just no longer likes my motors.

So having tested 8 motors (6nxt and 2 ev3) - only 1 (nxt) is working now.  I feel like this is intermittent because yesterday I tested a motor that worked and today the same motor in the same port doesn't work.

Any suggestions...

rhempel, 2015-01-14:
Did you swap cables at the same time, or just the motors?

sharived, 2015-01-14:
Different cables

dlech, 2015-01-15:
I probably broke something when I was fixing something else in this commit: https://github.com/ev3dev/lego-linux-drivers/commit/49a1f89e184a6618cac8bb77689b170109efd483

dlech, 2015-01-15:
Also, this sounds just like the behavior described in #245, so we should close this issue as a duplicate if that is the case.,

sharived, 2015-01-15:
Sure this sounds similar - in particular the comment today about B and C disappearing.  I'll close this issue.

closed 2015-01-15 by sharived

sharived, 2015-01-20:
Just wanted to let you know that the fix in v3.16.7-ckt4-ev3dev1 fixed my issue.  Thanks!