ev3dev/ev3dev #249: How can I write C++/C programs on the ev3 dev

Sherev3, 2015-01-16:
I kinda followed every step and the motors are working on the bash but i need to write c/c++ programs

rhempel, 2015-01-16:
Well, first I would consider writing your program in Lua, Python, node.js, scheme, Go, or any of the other scripting languages in ev3dev.
Then you can do one of two things - one is to set up a Linux VM on your host computer, install brickstrap, get build-essential and then you can compile on your host and transfer to the EV3.
Or you can consider installing C con the brick
`apt-get install build-essential`
But compiling on the brick will be slowwww

dlech, 2015-01-16:
You can also find a list of existing libraries [here](http://www.ev3dev.org/docs/libraries/). Keep in mind, ev3dev is under heavy development and the libraries may not be up to date with the latest kernel.

Sherev3, 2015-01-21:
I am sorry guys here i am still new, but i have programmed in C before thats why i want to programme with it. So can someone explain how the system work like where should the compiler be how does the compiler communicate with brick and  where to put the libraries

Sherev3, 2015-01-21:
I have tried ralph's way but i think iam doing something wrong

dlech, 2015-01-21:
> where should the compiler be

You have 3 choices:

1. [Use brickstrap](https://github.com/ev3dev/ev3dev/wiki/Using-brickstrap-to-cross-compile-and-debug). Pros: You can install libraries using the Debian package system (i.e. `apt-get`) and it is faster the running the compiler on the EV3 itself. Cons: It is slower and you have copy the executable or mount a remote file system to run the program on the EV3.
2. Do everything on the EV3 itself. Pros: No copying files or mounting remote file systems. Cons: Installing tools/libraries is slow and compiling is slow (although not bad for small programs).
3. Use a cross-compiler like [this](http://www.mentor.com/embedded-software/sourcery-tools/sourcery-codebench/editions/lite-edition/) on a host machine. Pros: Runs on windows. Compiles very fast. Cons: Harder to use non-standard libraries and have to copy executable to or mount remote file system on the EV3 itself.

> does the compiler communicate with brick

It doesn't. You have to either compile on the brick itself or manually copy the compiled files (i.e. scp) or mount a remote file system (i.e. nfs, sshfs, etc.)

> where to put the libraries

If you install libraries using Debian packages, the will be installed in the correct place. If you are writing your own libraries, they should go in `/usr/local/lib`.

If you don't want to start from scratch, check out <https://github.com/in4lio/ev3dev-c>. It is the only C library for ev3dev that I know of so far.

Sherev3, 2015-01-22:
[image]

Sherev3, 2015-01-22:
I run into this problem which doesnt complete the installation later

ddemidov, 2015-01-22:
Did you also do
```
user@host:~$ sudo apt-key adv --keyserver pgp.mit.edu --recv-keys 2B210565
user@host:~$ sudo apt-add-repository http://ev3dev.org/debian
user@host:~$ sudo apt-get update
```
as described in the [wiki](https://github.com/ev3dev/ev3dev/wiki/Using-brickstrap-to-cross-compile-and-debug)?

Sherev3, 2015-01-22:
yeah

ddemidov, 2015-01-22:
Looks like you have ubuntu precise (12.04), and the wiki requires ubuntu trusty (14.04). Could this be a problem?

Sherev3, 2015-01-22:
As it might take sometime to to get ubuntu trusty can someone show a guide on how i can compile on the brick itself iam not doing that would take much time in compiling

ddemidov, 2015-01-22:
I think running `sudo apt-get install gcc` on the brick is enough to get the compiler.
Now you can compile a program with
```
gcc -o hello hello.c
```
and run it with
```
./hello
```
This could work for C, but I've hit a memory limit (and hence an internal compiler error) even for simple C++ programs.

rhempel, 2015-01-22:
@sherev3 - if you can install and run the C compiler on a regular Debian or Ubuntu machine then you can do it on the EV3 - as has already been mentioned, just `apt-get install build-essentials` and you're good to go.

closed 2016-02-25 by dlech