ev3dev/ev3dev #250: Strange motor port behavior after upgrading to latest kernel

dsharlet, 2015-01-17:
I just did an apt-get dist-upgrade and got a new kernel (3.16.7-ckt3-ev3dev1) 

At first, I couldn't find the large motors anywhere, and they weren't being detected by the C++ API. The medium motors seem to work as they did before. I unplugged and replugged all the motors, and now I could find the motors expected on ports B, C, and D. The motor on port A was still missing. I tried lots more unplugging/replugging/rebooting without success. 

Eventually, I found in the new device browser that I could set the mode of the port to ev3-tacho-motor. I did that, and now everything is working as expected.

dlech, 2015-01-18:
Sounds like #245.

wwildey, 2015-01-18:
I have been doing some tests with the kernel version:
[root@ev3dev][H:1018][/sys/class/tacho-motor]$ >  uname -a
Linux ev3dev 3.16.7-ckt3-ev3dev1 #2 PREEMPT Tue Jan 6 19:36:52 CST 2015 armv5tejl GNU/Linux

It appears that the auto detect for motors is broken.  When plugging in a large motor after a reboot with nothing plugged into the ports, dmesg reports:

Port A:
[  206.345670] lego-port port4: Trying to register an invalid motor type on port4.
Port B:
[  310.126504] lego-port port5: Trying to register an invalid motor type on port5.
Port C:
[  347.077425] lego-port port6: Trying to register an invalid motor type on port6.
Port D:
[  377.418356] lego-port port7: Trying to register an invalid motor type on port7.

I then plugged in a small motor in each port and they seemed fine.

I then plugged in small motors in ports A and B, and large motors in C and D and this was dmesg result:
[  537.547973] tacho-motor motor1: Bound to 'outA:ev3-tacho-motor'.
[  537.551518] lego-port port4: Added new device 'outA:ev3-tacho-motor'
[  596.218934] tacho-motor motor2: Bound to 'outB:ev3-tacho-motor'.
[  596.231613] lego-port port5: Added new device 'outB:ev3-tacho-motor'
[  698.537426] lego-port port6: Trying to register an invalid motor type on port6.
[  704.728414] lego-port port7: Trying to register an invalid motor type on port7.

[root@ev3dev][H:1024][/sys/class/tacho-motor]$ >  cat motor2/port_name; cat motor2/type 
outB
minitacho
[root@ev3dev][H:1025][/sys/class/tacho-motor]$ >  cat motor1/port_name; cat motor1/type 
outA
minitacho

I then went to "device browser" on the brick itself, selected ports, and selected outC and outD to manually set the ports to the eve taco motor.  Both ports show status as "no-motor".  I then set outC and outD to ev3-tacho-motor manually:

[root@ev3dev][H:1051][/sys/class/tacho-motor]$ >  cat motor3/port_name; cat motor3/type 
outC
tacho
[root@ev3dev][H:1052][/sys/class/tacho-motor]$ >  cat motor4/port_name; cat motor4/type 
outD
tacho

There was also an instance where I rebooted with only large motors plugged in, manually set them to eve-tacho-motor, then plugged in two small motors.  After plugging in the small motors, one was set to type tacho and not minitacho.

Setting the mode in the browser only lasts until next boot.

rhempel, 2015-01-18:
Hmmmm - duplicate of #245 ???
Looks like we'll be looking through recent commits to see what broke - sorry for the issues folks, and thanks for trying ev3dev and reporting the defect.

dsharlet, 2015-01-18:
I thought it might have been related to #245, but there was a lot of specific information in #245 that I couldn't reproduce (e.g. my problems have all with large motors, but mini motors work fine). I started adding a comment to that issue, but it seemed like I was explaining more differences in behavior than I was adding useful information.

Anyways, thanks for looking at it. This issue isn't a showstopper, you can work around this problem by going to the device browser and manually setting the mode of a motor port.

dlech, 2015-01-20:
I have just released [kernel v3.16.7-ckt4-ev3dev1](http://www.ev3dev.org/news/2015/01/19/Kernel-Release-v3.16.7-ckt4-ev3dev1/) which should fix this issue. Please reopen this issue if it is not a duplicate of #245.

closed 2015-01-20 by dlech