ev3dev/ev3dev #251: issue with large motor

topikachu, 2015-01-18:
After apt-get dist-upgrade, I cannot use large motor any more.

dmesg
[  766.950818] lego-port port4: Trying to register an invalid motor type on port4.
[  839.561652] lego-port port5: Trying to register an invalid motor type on port5.
[  946.992588] lego-port port6: Trying to register an invalid motor type on port6.
[  958.616845] lego-port port7: Trying to register an invalid motor type on port7.

uname -a
Linux ev3dev 3.16.7-ckt3-ev3dev1 #2 PREEMPT Tue Jan 6 19:36:52 CST 2015 armv5tejl GNU/Linux

rhempel, 2015-01-18:
Duplicate of #250 which is likely a duplicate of #251

Folks, just a quick reminder - the Issues section does have a search feature :-) You can see if your issue has already been reported.

topikachu, 2015-01-18:
Well. Thanks. Really a duplicated one  ;)

dlech, 2015-01-20:
I have just released [kernel v3.16.7-ckt4-ev3dev1](http://www.ev3dev.org/news/2015/01/19/Kernel-Release-v3.16.7-ckt4-ev3dev1/) which should fix this issue. Please reopen this issue if it is not a duplicate of #245.

closed 2015-01-20 by dlech