ev3dev/ev3dev #253: Connecting Raspberry Pi to Ev3Dev

CharlesMontesquieu, 2015-01-22:
I've recently bought a RPi B+ board, a small 3.5 TFT screen and a camera module. And I was struck by an idea to combine those two things. I mean, if I can easily build robots with motors and sensors thanks to LEGO Technic and if I could join that with a camera and a small, but fully-functional display, than I can make... wow, just, WOW! :)

So I wanted to ask a question, is such thing possible? I guess so, although I am not quite sure. If a RPi is just a tiny computer, than I can use it to make a SSH session with Ev3Dev via USB ports, right?
Usually one establishes connection with RPi via SSH with other computer, but it can be vice versa, isn't it? I have looked for the manual on Help page, but it states only how to make a connection with
Ev3Dev using Network Manager in Linux. I couldn't find that on Raspbian. Can I use command line instead? 

Thanks!

moriarty, 2015-01-22:
Hi, 

If you want to connect from your RasberryPi to your Ev3, and they're both on the same network it's possible. I leave my RPi and Ev3 both on 24/7, they're currently both connected via ethernet cable but sometimes I use a wifi dongle on the Ev3. 

I can log into my Pi from campus, it's the only device accessible from the internet, and from there into my Ev3. The only real use I have for this is things that take a long time to compile. But I plan on setting up a USB camera to test with the Ev3 remotely and monitor it. 

I will try via USB connection this afternoon. I think it should work.

MetallianFR68, 2015-01-22:
I'm working on a very similar project.
I created a 6 axis STOP MOTION RIG to shoot brickfilms
It is made only with LEGO parts, except the camera which is a standard RASPICAM with a modified lens

My setup works as follows:
2 EV3 bricks with 8 motors attached
1 raspberry PI B+ with the camera module

I can control everything using a remote computer since everything is connected to the same WIFI network. Even better, I can control everything using a bluetooth PS3 controller !

moriarty, 2015-01-22:
@MetallianFR68, you're Ev3s are connected to the RPis using USB network adapters, or via a USB cable directly to the Pi? 

I usually use a USB ethernet adapter or USB wifi adapter. Like @Metallian said, it makes everything easier because all the devices are on the same network. 

But it sounds like @CharlesMontesquieu wants use USB cable between the RPi and Ev3. Have you tried this?

moriarty, 2015-01-22:
I can confirm that USB network from RPi to Ev3 works. 

Follow the steps here: http://www.ev3dev.org/docs/tutorials/setting-up-ethernet-over-usb-on-linux-networkmanager/

Then skip steps 3,4 & 5.

Instead run this on the RPi: 

```sudo ip ad add 10.42.0.1/24 dev usb0```

rhempel, 2015-01-22:
One thing I need to make clear in an upcoming video is that the SSH session is used mainly to run commands interactively on the EV3 - you can write scripts that control the motors, read sensors etc and they run right on the brick.
You do not need ssh to control the brick remotely, either. Most scripting languages allow you to send arbitrary data over network sockets, so to communicate between the RPI and the EV3 (for example) you could have a Python program on the EV3 listening on an IP socket for data that is  coming from a C program on the RPI.

CharlesMontesquieu, 2015-01-22:
Thank you for all your answers!
I will try this soon.

crowndale1, 2015-01-27:
I am planning to do a similar thing also - RPI control EV3 motor through a camera on RPI.

I installed following through USB window's connection:
image: ev3dev-jessie-2014-12-01 
3.16.7-ckt4-ev3dev1 #2 PREEMPT Mon Jan 19 21:08:06 CST 2015

Can I use the same SD card for EV3 and RPI connection? I do not think I need to have an internet connection to do the update for this application. Right? Thanks.

dlech, 2015-01-28:
> Can I use the same SD card for EV3 and RPI connection?

Yes.

> I do not think I need to have an internet connection to do the update for this application. Right?

Not exactly sure what you mean here, but ev3dev will work just fine when not connected to the Internet. You only need an Internet connection for `apt-get update|upgrade`. You can also have multiple network connections at the same time, so it doesn't have to be either one or the other.

crowndale1, 2015-01-30:
by following above instructions RPI connected to EV3 through USB. Thanks much. 
How can I remotely access the RPI, using a laptop for instance, when RPI wired connected to EV3?
I found that when RPI connected to EV3, the putty on PC (previously connected to RPI) was not working any more.

crowndale1, 2015-02-01:
Forget about Question. I know how to do it now. Thanks.

moriarty, 2015-03-01:
@CharlesMontesquieu, @crowndale1, has this question been answered? If so does the answer need clarification or documentation in case other sumble across the same question? Or can the issue be closed?

crowndale1, 2015-03-02:
From my side, this can be closed. Thanks much.

CharlesMontesquieu, 2015-03-03:
Sure, it can be closed. Thank you very much.

closed 2015-03-03 by dlech

HockeyKS, 2016-11-14:
Hi,
I have a Raspberry Pi with Jessie and an ev3 with all the required images installed. At first it the connection via usb worked out really well. After trying to connect the ev3 and the Raspberry via Bluetooth (did not work out) I wanted to reconnect the two devices via usb. Every time I try to set up an ssh I get an error message now:

`ssh: connect to host ev3dev.local port 22: No route to host`

the output of `ifconfig` is:

`usb0      Link encap:Ethernet  HWaddr 12:16:53:4b:73:79  
          inet addr:169.254.105.213  Bcast:169.254.255.255  Mask:255.255.0.0
          inet6 addr: fe80::2d40:56e0:34e8:3e89/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:131 errors:0 dropped:0 overruns:0 frame:0
          TX packets:70 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:12076 (11.7 KiB)  TX bytes:15008 (14.6 KiB)
`

Did I forget to add the IP in some file on Raspberry?
I would really appreciate your help! Thanks a lot!

dlech, 2016-11-14:
@HockeyKS, the issue you have commented on has been closed for a long time. Your question here is likely to get lost because of this. Please open a new issue and explain exactly what you did that was working, which image files you used, etc. The more information that you can give, the more likely that we can help you.

HockeyKS, 2016-11-17:
@dlech thank you, I will do that!