ev3dev/ev3dev #254: Need help with Mindsensors EV3 Sensor Multiplexer (EV3SensorMUX)

JorgePe, 2015-01-22:
High.
I've received a Mindsensors EV3 Sensor MUX a couple of months ago. At the time, I used i2cdump to read the sensor values on each channel but then I saw that you were adding support for it in ev3dev so I decided to wait a bit and changed my attention to other projects.

I now have '3.16.7-ckt4-ev3dev1 #2'. 

Just the MUX ath EV3 Input 1 with an EV3 Touch Sensor ath MUX CH 1:

`lego-port port0: Added new device 'in1:nxt-i2c-host'`
`i2c-legoev3 i2c-legoev3.3: registered on input port 1`
`lego-port port8: Bound to device '3-0050'`
`lego-port port8: Added new device 'in1:mux1:lego-ev3-uart-29'`
`lego-sensor sensor0: Bound to device '3-0050'`
`lego-port port9: Bound to device '3-0051'`
`lego-port port9: Added new device 'in1:mux2:lego-ev3-uart-29'`
`lego-sensor sensor1: Bound to device '3-0051'`
`lego-port port10: Bound to device '3-0052'`
`lego-port port10: Added new device 'in1:mux3:lego-ev3-uart-29'`
`lego-sensor sensor2: Bound to device '3-0052'`
`lego-sensor sensor3: Bound to device 'in1:mux1:lego-ev3-uart-29'`
`lego-sensor sensor4: Bound to device 'in1:mux2:lego-ev3-uart-29'`
`lego-sensor sensor5: Bound to device 'in1:mux3:lego-ev3-uart-29'`

I have now 6 sensors:

`root@ev3dev:~# ls /sys/class/lego-sensor/ -l`
`lrwxrwxrwx  1 root root 0 Jan 22 12:47 sensor0 -> ../../devices/platform/legoev3-ports/lego-port/port0/i2c-legoev3.3/i2c-3/3-0050/lego-sensor/sensor0`
`lrwxrwxrwx  1 root root 0 Jan 22 12:47 sensor1 -> ../../devices/platform/legoev3-ports/lego-port/port0/i2c-legoev3.3/i2c-3/3-0051/lego-sensor/sensor1`
`lrwxrwxrwx  1 root root 0 Jan 22 12:47 sensor2 -> ../../devices/platform/legoev3-ports/lego-port/port0/i2c-legoev3.3/i2c-3/3-0052/lego-sensor/sensor2`
`lrwxrwxrwx  1 root root 0 Jan 22 12:47 sensor3 -> ../../devices/platform/legoev3-ports/lego-port/port0/i2c-legoev3.3/i2c-3/3-0050/lego-port/port8/in1:mux1:lego-ev3-uart-29/lego-sensor/sensor3`
`lrwxrwxrwx  1 root root 0 Jan 22 12:47 sensor4 -> ../../devices/platform/legoev3-ports/lego-port/port0/i2c-legoev3.3/i2c-3/3-0051/lego-port/port9/in1:mux2:lego-ev3-uart-29/lego-sensor/sensor4`
`lrwxrwxrwx  1 root root 0 Jan 22 12:47 sensor5 -> ../../devices/platform/legoev3-ports/lego-port/port0/i2c-legoev3.3/i2c-3/3-0052/lego-port/port10/in1:mux3:lego-ev3-uart-29/lego-sensor/sensor5`

so Ev3 Touch Sensor on CH1 must be sensor3 (i.e. mux1), right?

but 

`root@ev3dev:~# cat /sys/class/lego-sensor/sensor3/mode'
`COL-REFLECT'
`root@ev3dev:~# cat /sys/class/lego-sensor/sensor3/modes'
`COL-REFLECT COL-AMBIENT COL-COLOR'

(it thinks it is a Color Sensor?)
and 
'root@ev3dev:~# cat /sys/class/lego-sensor/sensor3/value0'
always returns '0' even when pressed

How do I change the device or am I reading at the wrong place?

dlech, 2015-01-22:
It looks like I need to update the docs at <http://www.ev3dev.org/docs/sensors/mindsensors.com-ev3-sensor-multiplexer/>. It should have a link to <http://www.ev3dev.org/docs/ports/ms-ev3-smux-port/>. There is a `lego-port` class device that is loaded for each port on the MUX. Currently, there is no automatic detection. You have to tell it what sensor is attached by using `mode` and `set_device` on the port.

It should be possible to get automatic detection working, but I was having problems with timing, so I left it out for now.

P.S. I edited you comment so that I could read it easier. You might want to have a look at <https://help.github.com/categories/writing-on-github/>.

JorgePe, 2015-01-22:
Thanks, it works (after searching for set_device)
```
echo analog > /sys/class/lego-port/port8/mode
echo lego-ev3-touch > /sys/class/lego-port/port8/set_device
```

sensor6 mode is now TOUCH and I can read its value with
```
cat /sys/class/lego-sensor/sensor6/value0
```
Sorry for recurrent markdown problems, I find markdown very tiresome.

closed 2015-01-22 by JorgePe

dlech, 2015-01-22:
I've updated the page I linked to.

BTW, since the `lego-ev3-touch` sensor is the only Analog/EV3 sensor, you don't have to write it to `set_device` - it is loaded automatically when you set the mode. See the footnotes in <http://www.ev3dev.org/docs/ports/ms-ev3-smux-port/>.