ev3dev/ev3dev #255: dpkg-reconfigure locales issue

crowndale1, 2015-01-22:
Here is my issue:
 
1) download ev3dev-jessie-2014-10-07 -OK
2) follow the instructions to connect the ev3 to a window PC using USB -OK
3) in Putty, user name: root, password: r00tme - OK
3) in Putty, dpkg-reconfigure tzdata - OK
4) in Putty, dpkg-reconfigure locales using default en-US - Error
 bus error 
5) if skip 4)
6) apt-get update - OK
7) apt-get upgrade - error

locale: Cannot set LC_CTYPE to default locale: No such file or directory   
locale: Cannot set LC_MESSAGES to default locale: No such file or directory   
locale: Cannot set LC_ALL to default locale: No such file or directory

dlech, 2015-01-22:
The image comes with en-US already, so there is no need to run `dpkg-reconfigure locales` unless you want a different language.

Also, you are not using the latest image file. You should be using `ev3dev-jessie-2014-12-01`.

crowndale1, 2015-01-23:
still have an issue;

1) format sd card using SDFormatter (quick format, format size adjustment off) -OK
2) FAT32, 32 GB - OK
3) write ev3dev-jessie-2014-12-01 -OK
4) dpkg-reconfigure tzdata -OK
5) apt-get update -OK
6) apt-get upgrade - error as below

...
Unpacking systemd (215-8) over (215-6) ...
Setting up systemd (215-8) ...
Failed to set capabilities on file `/usr/bin/systemd-detect-virt' (Invalid argument)
The value of the capability argument is not permitted for a file. Or the file is not a regular (non-symlink) file

setfacl: /var/log/journal: Operation not supported
setfacl: /var/log/journal/494c2307740f40188e9f5eaa764e72b2: Operation not supported
setfacl: /var/log/journal/494c2307740f40188e9f5eaa764e72b2/system.journal: Operation not supported
setfacl: /var/log/journal/494c2307740f40188e9f5eaa764e72b2/system@bad94b78cef54dba894c14661bd12099-0000000000000001-00035d013be29509.journal: Operation not supported
(Reading database ... 27486 files and directories currently installed.)
Preparing to unpack .../systemd-sysv_215-8_armel.deb ...
Unpacking systemd-sysv (215-8) over (215-6) ...
dpkg: dependency problems prevent processing triggers for dbus:
 dbus depends on libdbus-1-3 (>= 1.7.6); however:
  Package libdbus-1-3:armel is not configured yet.

dpkg: error processing package dbus (--configure):
 dependency problems - leaving triggers unprocessed
dpkg: dependency problems prevent processing triggers for dbus:
 dbus depends on libdbus-1-3 (>= 1.7.6); however:
  Package libdbus-1-3:armel is not configured yet.

dlech, 2015-01-23:
The dbus package error is a known issue: #239.

However the other errors make me think perhaps there is something wrong with your SD card. Do you have another microSD card that you can try?

Also, the formatting of the SD card you are doing is not necessary. Writing the image to the SD card writes over any partitions that were made be the formatting tool.

crowndale1, 2015-01-23:
Do I have to run;

lvextend --extents +100%FREE --resizefs /dev/ev3devVG/root /dev/mmcblk0p3

It takes time. Thanks.

dlech, 2015-01-23:
You don't *have* to, but you will quickly run out of room when you install new packages if you do not. On all of my SD cards, it runs fairly quickly, so if it is taking a long time (more than a minute or so), this also leads me to believe that there is something wrong with your SD card.

crowndale1, 2015-01-23:
I tried another sd card (class 10, 64 GB), exact the same issue. Shoudl I try another SD card?

Do you think class 4, 8 GB sd will work? Any recommendation for the SD card? Thanks.

JorgePe, 2015-01-23:
It should worlk. I've been using two Kingston 8 GB microSD HC I (class 10) cards without issues [had that libdbus issue also]

crowndale1, 2015-01-23:
To overcome libdbus issue, should I 
1) ignore it, or
2) Run  apt-get install -f  to patch things up and then run  apt-get update  and  apt-get upgrade  again. or
3) run apt-get update, then apt-get install libdbus-1-3, and then apt-get upgrade ?
Thanks.

JorgePe, 2015-01-23:
option 3) should work

dlech, 2015-01-23:
The EV3 only supports SDHC/version 2.0 of the SD specification, which means a maximum size of 32GB. You could try replacing the `100%` with something like `40%` in

    lvextend --extents +100%FREE --resizefs /dev/ev3devVG/root /dev/mmcblk0p3

This should make the total size less than 32GB and you my have better results, but no guarantee. If you already have a class 4, 8GB it is the safer choice.

dlech, 2015-01-23:
I have updated the Getting Started page about the 32GB limit.

crowndale1, 2015-01-24:
it finally went through (using a new class 10, 8 GB sd, using the option 3 above to overcome libdbus issue). Still got following issue?? in apt-get upgrade. I will try to run some applications (running a motor, sensor, etc) to see if there is a problem because of that. thanks much for all the supports. 

...
Preparing to unpack .../systemd_215-8_armel.deb ...
Unpacking systemd (215-8) over (215-6) ...
Processing triggers for dbus (1.8.12-1) ...
Setting up systemd (215-8) ...
Failed to set capabilities on file `/usr/bin/systemd-detect-virt' (Invalid argument)
The value of the capability argument is not permitted for a file. Or the file is not a regular (non-symlink) file
setfacl: /var/log/journal: Operation not supported
setfacl: /var/log/journal/494c2307740f40188e9f5eaa764e72b2: Operation not supported
setfacl: /var/log/journal/494c2307740f40188e9f5eaa764e72b2/system@7e84c9483dec4361aa593cd1d6607bbe-0000000000000001-00035d013bb2d763.journal: Operation not supported
setfacl: /var/log/journal/494c2307740f40188e9f5eaa764e72b2/system.journal: Operation not supported
(Reading database ... 27486 files and directories currently installed.)
Preparing to unpack .../systemd-sysv_215-8_armel.deb ...
Unpacking systemd-sysv (215-8) over (215-6) ...
Setting up systemd-sysv (215-8) ...
(Reading database ... 27486 files and directories currently installed.)
Preparing to unpack .../libuuid1_2.25.2-4_armel.deb ...
Unpacking libuuid1:armel (2.25.2-4) over (2.25.2-3) ...
Setting up libuuid1:armel (2.25.2-4) ...
Processing triggers for libc-bin (2.19-13) ...

dlech, 2015-01-24:
It looks like you have found a bug in our image file. I am going to open an new issue for it so that it does not get lost in the discussion here.

dlech, 2015-01-26:
As far as I can tell, the `setfacl` errors can safely be ignored. It just means that users in the `adm` group won't be able to view the systemd journal. Is reconfiguring locales still an issue? If not, we can close this one.

crowndale1, 2015-01-26:
There is no issue for locales. I think we can close this one. Thanks much for the support.

closed 2015-01-26 by crowndale1