ev3dev/ev3dev #256: problem with apt-file update

BertLindeman, 2015-01-23:
Running kernel ` 3.16.7-ckt4-ev3dev1`

`apt-file update`
fails on:

```
Downloading complete file http://ev3dev.org/debian/dists/jessie/main/Contents-armel.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 406 Not Acceptable
Download of http://ev3dev.org/debian/dists/jessie/main/Contents-armel.gz failed
Command exited with code 22
```

Do I need to change some setup file for apt-file?

dlech, 2015-01-23:
If you follow the link <http://ev3dev.org/debian/dists/jessie/main/> you will see that `Contents-armel.gz` is not there, so it not you, it me. I will see if I can change some settings so that this file is generated.

dlech, 2015-01-23:
Give it a try now.

BertLindeman, 2015-01-23:
On the web I now see the file
` http://ev3dev.org/debian/dists/jessie/main/Contents-armel.gz`
And I can download OK with `wget  http://ev3dev.org/debian/dists/jessie/main/Contents-armel.gz`

But `apt-file update` still complains the same.
Would it help if I removed the cache of apt-file?

dlech, 2015-01-23:
Hmm. Must have something to do with my web host. It works on my local mirror.

dlech, 2015-01-23:
I've sent in a support request to my hosting provider. Hopefully they will get it fixed. Basically, what is happening is if the user agent string has `curl` or `wget` in it, then it returns the 406 error. So, you should be able to create a `~/.curlrc` file with something like

    user-agent = "Mozilla/4.0"

and it should work.

BertLindeman, 2015-01-23:
I like it!  and it works:

```
Downloading complete file http://ev3dev.org/debian/dists/jessie/main/Contents-armel.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  695k  100  695k    0     0   362k      0  0:00:01  0:00:01 --:--:--  370k
```

Thanks David.
Keeping this open untill your provider responds.

dlech, 2015-04-16:
My hosting provider did add an exception for this, but I think they added it to the wrong directory. They weren't too thrilled about it either, so I'm just going to call the "won't fix" since we have a reasonable workaround. And really, `apt-file` should be setting the user agent anyway, so that would be the proper fix.

closed 2015-04-16 by dlech

dlech, 2016-08-07:
I just moved the ev3dev package repository to a new hosting service, so this might/should work now.

BertLindeman, 2016-08-08:
On my rpi3 moved `~\.curlrc` out of the home directory and ran `apt-get update`.
Results in
```
Ign http://httpredir.debian.org jessie InRelease
Hit http://httpredir.debian.org jessie Release.gpg
Hit http://httpredir.debian.org jessie Release
Hit http://httpredir.debian.org jessie/main armhf Packages
Hit http://httpredir.debian.org jessie/contrib armhf Packages
Hit http://httpredir.debian.org jessie/non-free armhf Packages
Hit http://archive.ev3dev.org jessie InRelease
Hit http://archive.ev3dev.org jessie/main armhf Packages
Reading package lists... Done
```
So looks fine to me ;-)
Thanks @dlech
 
Ah, `~.curlrc` was:
```user-agent = "Mozilla/4.0"```

dlech, 2016-08-08:
I thought the problem was with `apt-file update` rather than `apt-get update`.

BertLindeman, 2016-08-08:
Oooops. Sorry. :blush: 

But also OK :smile: 
```
apt-file update
Downloading Index http://httpredir.debian.org/debian/dists/jessie/main/Contents-armhf.diff/Index:
No Index available.
Downloading complete file http://httpredir.debian.org/debian/dists/jessie/main/Contents-armhf.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
File is up-to-date.
Downloading Index http://httpredir.debian.org/debian/dists/jessie/contrib/Contents-armhf.diff/Index:
No Index available.
Downloading complete file http://httpredir.debian.org/debian/dists/jessie/contrib/Contents-armhf.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
File is up-to-date.
Downloading Index http://httpredir.debian.org/debian/dists/jessie/non-free/Contents-armhf.diff/Index:
No Index available.
Downloading complete file http://httpredir.debian.org/debian/dists/jessie/non-free/Contents-armhf.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
File is up-to-date.
Downloading Index http://archive.ev3dev.org/debian/dists/jessie/main/Contents-armhf.diff/Index:
No Index available.
Downloading complete file http://archive.ev3dev.org/debian/dists/jessie/main/Contents-armhf.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
File is up-to-date.
```