ev3dev/ev3dev #258: How to run an ev3 motor from a PC but wirelessly

crowndale1, 2015-01-25:
image:  ev3dev-jessie-2014-12-01 
3.16.7-ckt4-ev3dev1 #2 PREEMPT Mon Jan 19 21:08:06 CST 2015

I can run an ev3 motor from PC with a USB connection (through Putty, ev3-python). How can I do the same thing without PC wired connected? I do not have a solid computer science background. A step by step instruction will be greatly appreciated!

rhempel, 2015-01-25:
Same as with a USB connection - you need to follow the directions for setting up the EV3 wireless connection vie a USB WiFi adapter, then ssh in to the EV3

crowndale1, 2015-01-25:
Do you mean following the steps in below:

Bluetooth Network on Windows #236 

Thanks.

topikachu, 2015-01-25:
@crowndale1 It's pissible run the Brick wiressely. I suggest you
1. use usb connection to setup the dev env, for example apt-get, pip install ...
2. pair bluetooth by brickman
3. setup bluetooth network and connect to pc
4. as I test, my windows cant assign correct ip to its bluetooth network adapter, so I need to setup static ip address on windows.
If everything is ok, you are able to putty to ev3 via wireless network.

crowndale1, 2015-01-26:
Here is what I did
1) In Netwroking, check Bluetooth - OK
2) In Bluetooth, check Visible - OK
3) In Bluetooth, check start Scan - found my PC -OK
4) Press my PC, go to Pair -OK
5) Press Pair, ask for Enter Pin for my PC

what is the Pin - 1234? and how to input the code?

Thanks.

dlech, 2015-01-26:
You can enter the pin code by pressing Enter on the EV3 when the text box is selected. This will open a virtual keyboard.

crowndale1, 2015-01-26:
Making some progress
6) enter 1234 pin, sounds OK
7) In Network Connections, see my PC
8) press my PC, go to IPv4
9) following USB connection steps - changing, and then load windows default
10) check Connect automatically
11) press connect, showing busy, and then timeout was received, connect-failed

need help

dlech, 2015-01-26:
Check `journalclt -r` on the EV3 to see what failed. Look for entries that say `connmand` or `bluetoothd`.

crowndale1, 2015-01-26:
Do you mean using USB connection and Putty to type   journalclt -r  to find the error? Thanks.

dlech, 2015-01-26:
Yes, that is what I mean.

crowndale1, 2015-01-27:
1) in SSH:
root@ev3dev:~# journalclt -r
-bash: journalclt: command not found
2) In step 11 above, on EV3 shows:
GDBus.Error:net.connman.Error. Failed: Input/output error

dlech, 2015-01-27:
oops, I mistyped, it is `journalctl`.

crowndale1, 2015-01-27:
A lot of info. I am not sure if below is what we are looking for:

Jan 01 00:00:39 ev3dev connmand[302]: Method "ListAdapters" with signature "" on interface "org.bluez.Manager" doesn't exist
Jan 01 00:00:39 ev3dev bluetoothd[399]: sap-server: Operation not permitted (1)
Jan 01 00:00:39 ev3dev bluetoothd[399]: Sap driver initialization failed.
Jan 01 00:00:39 ev3dev bluetoothd[399]: Bluetooth management interface 1.6 initialized

dlech, 2015-01-27:
Any errors related to connmand or bluetoothd. The errors you listed so far are expected and can be ignored. Try connecting again and then check journalctl for any new errors.

dlech, 2015-01-27:
Also see #236. I don't think the person ever got connected, but there are lots of suggestions of things to try.

crowndale1, 2015-01-28:
Here is the journalctl after redo

Jan 01 00:38:57 ev3dev systemd[1]: Stopping LSB: Raise network interfaces....
Jan 01 00:38:57 ev3dev systemd-rfkill[615]: Failed to get rfkill device 'rfkill0': No such file or directory
Jan 01 00:38:57 ev3dev systemd[1]: Stopping Encrypted Volumes.

Jan 01 00:31:04 ev3dev bluetoothd[400]: 00:11:22:98:76:54: error updating services: Connection timed out (110)
Jan 01 00:30:49 ev3dev bluetoothd[400]: 00:11:22:98:76:54: error updating services: Connection timed out (110)
Jan 01 00:28:38 ev3dev bluetoothd[400]: 00:11:22:98:76:54: error updating services: Connection timed out (110)
Jan 01 00:26:38 ev3dev bluetoothd[400]: Agent /org/ev3dev/brickman/bluez5_agent replied with an error: org.freedesktop.DBus.Error.NoRepl
Jan 01 00:21:09 ev3dev systemd-udevd[521]: Failed to apply ACL on /dev/input/event3: Operation not supported
Jan 01 00:21:09 ev3dev kernel: hid-generic 0003:045E:00B0.0002: input,hidraw1: USB HID v1.11 Device [Microsoft Microsoft® Digital Media
Jan 01 00:21:09 ev3dev kernel: input: Microsoft Microsoft® Digital Media Pro Keyboard as /devices/platform/ohci.0/usb1/1-1/1-1:1.1/0003:
Jan 01 00:21:09 ev3dev kernel: hid-generic 0003:045E:00B0.0001: input,hidraw0: USB HID v1.11 Keyboard [Microsoft Microsoft® Digital Medi
Jan 01 00:21:09 ev3dev kernel: input: Microsoft Microsoft® Digital Media Pro Keyboard as /devices/platform/ohci.0/usb1/1-1/1-1:1.0/0003:
Jan 01 00:21:08 ev3dev kernel: usb 1-1: new low-speed USB device number 2 using ohci
Jan 01 00:19:57 ev3dev bluetoothd[400]: Agent /org/ev3dev/brickman/bluez5_agent replied with an error: org.freedesktop.DBus.Error.NoRepl
Jan 01 00:15:02 ev3dev systemd[1]: Started Cleanup of Temporary Directories.
Jan 01 00:15:02 ev3dev systemd[1]: Starting Cleanup of Temporary Directories...
Jan 01 00:13:03 ev3dev bluetoothd[400]: Agent /org/ev3dev/brickman/bluez5_agent replied with an error: org.freedesktop.DBus.Error.NoRepl

dlech, 2015-01-28:
It looks like either the pairing is failing or the connection is not trusted. To save me from having to explain everything again, please see #236 (and https://wiki.archlinux.org/index.php/Bluetooth if you need more info). Use `bluetoothctl` to inspect your windows computer Bluetooth connection. Make sure `Paired: yes` and `Trusted: yes` and `UUID: PANU` (or NAP or GN). You can also type `agent on` and `default-agent` to use `bluetoothctl` for pairing instead of brickman if you need to. You could also try enabling tethering instead as described [here](https://github.com/ev3dev/ev3dev/issues/232#issuecomment-69468871).

closed 2016-02-25 by dlech