ev3dev/ev3dev #259: Lua 5.3 Works

andrewstarks, 2015-01-25:
Not an issue so much as a report. I compiled and installed the release version of Lua 5.3 and it works without modification.

I don't believe that a package is available for it, yet, but I wanted to give you that 'fyi'.

(also, luarocks installed without issue, as well)

-Andrew

rhempel, 2015-01-25:
Cool, I'm going to be AFK (more like Away From Internet) for a week, and I'm going to look at how "busted" Lua unit test framework will integrate with my normal workflow. Thks for the note about 5.3 working out of the box - I'm looking forward to the binary math operators and the inbuilt fixed/floating numbers

dlech, 2015-01-25:
Well, I'll make it an issue then...

1. I see that we install lua5.1 by default, but there is a lua5.2 package available. Should we use lua5.2 in the default install?
2. Not sure how long it will take for a debian package for 5.3 to appear since debian is currently in a package freeze, but should we include it in the default install instead of 5.1/5.2 when it does become available?

andrewstarks, 2015-01-25:
My opinion is one of someone who uses Lua all of the time and usually stays with the latest. Someone coming to this for the first time or someone trying to install Lua packages may see things differently:

I'd suggest going to 5.2 now and 5.3, when available. 

I will continue to hand install Lua 5.3 and it is very simple to do so. The install process for Lua is "make linux & make install". 

If you decide to go this route (wget the Tarbell, unpack, make...), prior to a proper package being released, you may define various "LUA_COMPAT_*" flags to make things "outoftheboxeasy" for those with older Lua libraries. 

I do think that 5.2 is better than 5.1. There is very little reason to stick with 5.1. 

-Andrew

rhempel, 2015-01-27:
I agree with Andrew - let's move to 5.2 and get 5.3 up when it's available as a standard Debian package. Andrew, do you know who the Debian maintainer is for Lua? I dimly recall Enrico Colombini, but I might be wrong.

andrewstarks, 2015-01-27:
A quick search on Lua's mailing list reveals that perhaps Enrico Tassi (gares@fettunta.org) is the contact.

However, I'm very ignorant of most of the Linux world and probably doubled my knowledge by traversing your very effective tutorials (augmented by a touch of google). So I'm not aware of the etiquette of that request. I'm happy to make it, if it is simply a matter of doing so.

dlech, 2015-01-28:
All Debian packages have a page in the Debian Package Tracking System. For example, you can find the one for lua5.2 here: <https://packages.qa.debian.org/l/lua5.2.html>. Given the fact that Lua 5.3 was just officially released a couple weeks ago and that Debian is currently in a package freeze until Jessie goes stable, it could be a while before we see an official lua5.3 package. Given the popularity of Lua, I don't think you need to bug the maintainer about it.

BTW, I updated brickstrap so the next release will include lua5.2 instead of lua5.1.

closed 2015-06-09 by dlech