ev3dev/ev3dev #260: How to close or minimize the terminal?

donwojtallo, 2015-01-25:
Hi.
I wrote a simple Snake game using ev3dev and NodeJS. You play with buttons and see game on lcd screen. 
I'm closing Brickmap at the start using exec('systemctl stop brickman'). However, when I click any button a characters like "^CCA" blinks on the screen, because of the terminal running on background. How to close or minimize the terminal?

dlech, 2015-01-25:
1. It is best not to run as root. It is too easy to accidentally break something.

2. You don't need to stop brickman. You can run your program with `openvt`. It will automatically switch back to brickman when your program ends.

        sudo openvt -s -w -- sudo --user <username> -- <program>

    Or if you set the sticky bit on `openvt`, it is simpler.

        sudo chmod +t $(which openvt)
        openvt -s -w -- <program>

3. You need to put the terminal in graphics mode when you start your program and exit graphics mode when your program exits. (pseudo-code)

        vtfd = open("/dev/tty", "r");
        ioctl(vtfd, KDSETMODE, KD_GRAPHICS);
        // do stuff with framebuffer
        ioctl(vtfd, KDSETMODE, KD_TEXT);
        close(vtfd);
        // exit program

    If your program crashes before it restores text mode, the console will be stuck in graphics mode and will not be able to switch to another console.

dlech, 2015-01-25:
I tried the sticky bit thing and it does not seem to work, (I had just read about it yesterday and it sounded promising).

donwojtallo, 2015-01-25:
Thank You for the answer.
I have problem with ioctl. I don't know if it's because of npm module or my bad usage.
First, i've installed gcc, make and src-ioctl module:
```
apt-get install gcc
apt-get install build-essential
npm install src-ioctl
```
Then i use openvt as You explained to run NodeJS in new window and use this script:
```
var fs = require('fs');
var ioctl = require('src-ioctl');

var KDSETMODE = 0x4B3A;
var KD_GRAPHICS = 0x01;
var KD_TEXT = 0x00;
// ...
var vtfd = fs.openSync("/dev/tty", "r");
ioctl.ioctl(vtfd, KDSETMODE, KD_GRAPHICS);
// ...
```
It seems to not working, I still see blinking cursor and characters. No matter if I run it from .js file or manually type it in NodeJS console using usb keyboard.
The function ioctl.ioctl(); in the last line returns 0. The source of function is:
```
static Handle<Value> Ioctl(const Arguments& args){
    HandleScope scope;
    
    int sfd = Handle<Number>::Cast(args[0])->Value();
    int com = Handle<Number>::Cast(args[1])->Value();
    Handle<External> obj = Handle<External>::Cast( args[2] );
    
    ifreq *req = (ifreq*) External::Unwrap(obj);
    
    int res = ioctl( sfd, com, req );
    
    return scope.Close(Integer::New(res));
  }
```
I also tried to install "ioctl" module (an alternative), but it fails with "gyp" error.

dlech, 2015-01-25:
Looks like the error when installing the `ioctl` package is because there is no `node` executable. You can fix this by installing the `nodejs-legacy` Debian package.

donwojtallo, 2015-01-26:
Perfect! Its working!
Now the "ioctl" module works well. "src-ioctl" must be broken (it's probably not maintained anymore). 
Thank You for your  help!

dlech, 2015-01-26:
Great. I've added `nodejs-legacy` to the list of pre-installed packages that will be on the next SD card image release. Be sure to close this issue if you feel it is resolved.

closed 2015-01-26 by donwojtallo