ev3dev/ev3dev #261: Continuously varying the motor position without oscillation

dsharlet, 2015-01-27:
Has anyone managed to successfully vary the position continuously over time without oscillation from the motors? If so, how?

I've been attempting to do this without much success. I've tried using regulation mode and tuning the built in PID parameters. I've also tried building my own controllers, but it is hard to get low enough latency to build a good controller. 

At least with the stock PID parameters, I've found that even simply repeatedly setting the same setpoint will cause pretty serious oscillation/instability in the motor position. This almost seems like a bug.

While on the subject, what is 'K' in the controller parameters? It seems to basically do nothing, unless it is set to exactly 0, in which case the motors simply do not generate any torque at all.

dlech, 2015-01-27:
This is @rhempel's area of expertise, but he is out for a week, so in the mean time, have a look at #106 and #149 if you have not already.

WasabiFan, 2015-01-27:
@dsharlet That seems like the behavior that I would expect if the PID history was being reset every time you changed the target.

@dlech Do you know if the built-in controllers do this? I seem to recall having a conversation with someone about this a while ago, but can't seem to find the thread.

sharived, 2015-01-27:
I believe I'm having the same issue with my large EV3 motor.  I am running at a slow speed (pulses per second of 300) and I'm using regulation_mode on, but I'm not using a ramp up or ramp down, and I'm using relative position_mode and a hold stop_mode. 
My issue is that if I move 50 then -50 a couple of times, I don't wind up back in my beginning position.  This is pretty critical to my application, so I'd really like to figure this out.  What do you guys mean by PID?  Are these the speed_regulation_P,I,D values?  Is there any documentation on them or any pointers you can provide.  I'm not seeing anything at the moment.

rhempel, 2015-01-27:
@shari - you need to be careful if you're trying to do relative motion. I'm on vacation right now and I'm away from my EV3, but try to use `relative` position mode. when you say move +50 or -50, it's relative to the previous target position - not the current position. You can get the same effect by using absolute mode too, you just need to calculate your new absolute position.
The key thing to remember is that when you read the current position, it may be one or two counts from the desired position. So you don't want to calculate your next position as 50 counts from current position, you want to use 50 counts from the previous target position.
Maybe this warrants a better tutorial page :-)

rhempel, 2015-01-27:
@dsharlet, what exactly do you mean by varying position continuously over time? If you don't wait for the motor to go idle between move commands, then things will get pretty unpredictable. We struggled with how much control to give to the motor drivers, it's pretty hard to make a one size fits all solution. On the other hand, if you can read and write the motors fast enough, you can write your own control algorithm - just leave the motors in continuous run mode and vary the `duty_cycle_sp` as needed

dlech, 2015-01-28:
> Do you know if the built-in controllers do this?

I don't understand what you are asking.

> I seem to recall having a conversation with someone about this a while ago, but can't seem to find the thread.

I think you mean #100. Everyone else could benefit from reading that discussion as well.

dlech, 2015-01-28:
> what exactly do you mean by varying position continuously over time?

What @rhempel *really* means (I am putting words in his mouth :wink:) is "can you please provide a sample program so that we can reproduce what you are seeing".

dsharlet, 2015-01-28:
Thanks for the info all. 

@dlech 

> What @rhempel really means (I am putting words in his mouth :wink:) is "can you please provide a sample program so that we can reproduce what you are seeing".

I can't easily provide a sample program for this, the behavior is (probably) going to be a function of the load (i.e. the robot). I have a video of my robot: https://www.youtube.com/watch?v=IE5Z7MibYiI It's WIP, but it demonstrates the issue I'm trying to solve. Virtually all of the noise in the robot positioning is due to the motor controllers, not the cameras/stereo estimation which is quite stable/continuous.

I realize I am asking a lot of the controllers by directly attaching my robot's arms to the motors, but I need this to move very quickly. The motors are plenty powerful enough, I just need to smooth out the controller behavior when it is given position setpoints that vary rapidly (but in a differentiable way, i.e. I can properly update the PID state if I have access) over time.

@rhempel:

> If you don't wait for the motor to go idle between move commands, then things will get pretty unpredictable.

This is exactly what I need to do. If I had full control over the controller, what I would do is add my own computed derivative/integral terms to the controller state to avoid the discontinuity that occurs when the position setpoint changes. I'm trying to figure out how to approximate that behavior. 

> On the other hand, if you can read and write the motors fast enough, you can write your own control algorithm - just leave the motors in continuous run mode and vary the duty_cycle_sp as needed

I tried this very early on in my project and found it difficult to get low enough latency (BTW, this was the motivation for https://github.com/ev3dev/ev3dev-lang/pull/25 ). I will try again now that the rest of my project is more stable. 

Surely your controller is operating in a much tighter loop than is possible from user space? The braking/stop mode appears to be operating in a much tighter loop than user space would allow. The main challenge I was having with my own controller is attempting to use the brake mode when I wanted to stop, and my own controller otherwise.

@sharived : 

> What do you guys mean by PID? Are these the speed_regulation_P,I,D values? Is there any documentation on them or any pointers you can provide. I'm not seeing anything at the moment.

Yes, they are the speed regulation PID values. PID controllers are used in *many* applications, like controlling the EV3 motor output. The wikipedia article is good: http://en.wikipedia.org/wiki/PID_controller

dlech, 2015-01-28:
> I can't easily provide a sample program for this

:astonished: I can see why! That is amazing!

If someone can come up with a *simple* test case...

dsharlet, 2015-01-29:
A custom PID controller totally works. I'm not sure what I did wrong last time I tried to do this, I must have set one of the motor settings incorrectly (it was like the first thing I tried to do with ev3dev...).

Anyways, the movement of my robot is now nice and smooth. Here's a video of the otherwise same robot (the same stereo vision code): https://www.youtube.com/watch?v=jHaB2zdfhHg

Thanks for all the help and sorry to lead everyone on a bit of a wild goose chase there.

closed 2015-01-29 by dsharlet

sharived, 2015-01-29:
I'm doing better as well.  I switched to absolute and put a delay between my up and down motions.  I think my problem was that I wasn't quite going all of the way with relative because my up and down is actually a hammer and therefore I prevent the down motion from completing.  Thanks @dsharlet for the PID article.  Might need that for some of my other projects.  BTW, your robot looks really great.  What type of stereo camera are you using?  I'm working with cameras as well.  I developed a derivate of the pixy camera, which I'm hoping to kickstart next month (fingers crossed).

dsharlet, 2015-01-30:
I'm using NXTcams, which have been really difficult. The quality is quite low, but the biggest issue I have is that the field of view is so narrow (~45 degrees). Pixy looks like it is almost double this. Also, it would be great to be able to synchronize the cameras for stereo purposes. Some of my vision algorithms are much more complex than they would be if I could guarantee synchronization between the cameras.

The biggest challenge with stereo on any of these serial object recognition cameras is that stereo calibration is very difficult. I'm thinking about putting together a library for helping with this because it requires non-standard techniques (since you can't get pixels from these cameras) that were a bit of a challenge to figure out.

sharived, 2015-01-30:
Agreed on NXTcam.  I am actually increasing the field of view from the Pixy
to 130 degrees and I do output the raw data.  Unfortunately since I'm using
USB I'm guessing it won't work for you since you need stereo.  Also I'm
primarily supporting Snap! and python because this is targeted for school
age children.  Again your project looks great!  Thanks for sharing.