ev3dev/ev3dev #262: problem with update to 3.16.7-ckt4-ev3dev1

in4lio, 2015-01-28:
**# uname -r**
`3.16.1-7-ev3dev`

**# apt-get update**

**# apt-get upgrade**
...
`dpkg: error processing package dbus (--configure):`
` dependency problems - leaving triggers unprocessed`
`dpkg: dependency problems prevent processing triggers for dbus:`
` dbus depends on libdbus-1-3 (>= 1.7.6); however:`
`  Package libdbus-1-3:armel is not configured yet.`
...
`dpkg: too many errors, stopping`
`Errors were encountered while processing:`
` dbus`
` dbus`
...
` dbus`
`Processing was halted because there were too many errors.E: Sub-process /usr/bin/dpkg returned an error code (1)`

**# apt-get dist-upgrade**
Same result...

ddemidov, 2015-01-28:
Duplicate of #239

WasabiFan, 2015-01-28:
See #239.

in4lio, 2015-01-28:
Thanks,
the solution is:
`apt-get install libdbus-1-3`
before upgrade.
Maybe we should mention about it in Getting Started?)

closed 2015-01-28 by in4lio