ev3dev/ev3dev #263: Wrong $PWD in Bash script while running from Brickman's File Browser?

donwojtallo, 2015-01-29:
When I run Bash script from File Browser the $PWD variable is set to '/home/<user>' instead of path to this script. Is this a bug? If not, how do I get path to the script?

ddemidov, 2015-01-29:
If this is a shell script, `$(dirname "$0")` should work.

dlech, 2015-01-29:
Currently, brickman uses `sudo` to run programs as another user (brickman runs as root). It includes the `--login` option, which might affect this. I will see if there is a way to set the current directory to the one selected in the File Browser.

In the mean time, @ddemidov's suggestion will work.

dlech, 2015-01-29:
Also see #129. We have plans for an alternate method of launching programs that will offer more control over the environment.

From the `sudo` man page's `--login` section: 

> __sudo__ attempts to change to that user's home directory before running the shell, so I don't think we are going to be able to set `$PWD` to something else here - at least not without adding a wrapper script to call the program.

donwojtallo, 2015-01-29:
Thank You! 
```$(dirname "$0")``` is working.
I was confused and have done some tests. When I use ```nodejs $(dirname "$0")/test.js``` the script was failing too, because in JavaScript the working directory still the same as $PWD in shell script. I changed in JavaScript every ```'./'``` to ```__dirname``` (see [docs](http://nodejs.org/docs/latest/api/globals.html#globals_dirname)) and now it is workin well.
Problem solved.

closed 2015-01-29 by donwojtallo