ev3dev/ev3dev #265: Gyro and Color EV3 Sensor doesn't conntect

andrewstarks, 2015-02-08:
Is there a current issue with the Gyro or Light sensor? I can't seem to get either to show up. I'm using the Lua binding and running the included test script. My hacking around has turned up nuthin. :( 

Thanks!

dlech, 2015-02-08:
1. Please see <https://github.com/ev3dev/ev3dev/blob/master/CONTRIBUTING.md>. (It is the link in the big yellow box which you open a new issue.)
2. Lua bindings are probably not been updated for the last few kernel releases, which included some breaking changes.

rhempel, 2015-02-08:
This is exactly why we have the `sysfs` attributes - so we can develop and test independent of any language - if the attributes are working and you're language binding isn't - then the binding needs to get updated.
It's also why we appreciate having people report issues, and also jumping in and helping out with specific tasks.
It's a community project!

andrewstarks, 2015-02-08:
ACK! I do remember seeing these "submit an issue" instructions somewhere at some point, but had forgotten that they had existed by the time that I had an issue[1]....

1: SD image == `ev3dev-jessie-2014-12-01.img`
2: Kernel version == `3.16.7-ckt4-ev3dev1 #2 PREEMPT Mon Jan 19 21:08:06 CST 2015`
3: Don't think my issue is with brakeman, but i'm not certain about the role that brickman plays, so:

```
● brickman.service - Brick Manager
   Loaded: loaded (/lib/systemd/system/brickman.service; enabled)
   Active: active (running) since Sun 2015-02-08 17:09:48 CST; 42s ago
 Main PID: 483 (brickman)
   CGroup: /system.slice/brickman.service
           └─483 /usr/sbin/brickman
```
`lua ev3dev-lang-test.lua `  output runs fine and even sees the large motor that I have plugged in. It also sees and works with my mindsensors servo multiplexer and the lego touch sensor. It does not seem to see the sonar, gyro or light sensors.

I'd be happy to dig into the details but this is the first time that I've seen this style of driver (`/sys/class`) and while I think it is awesome and easy to understand, I would need a bit of patience from the willing as I plowed ahead.

Is there something I'm doing wrong? Is there an up-to-date-ish document that I could follow and then fix and the submit a patch for?

BTW: Awesome project! I'm having a great time poking through it!

--Andrew

[1] @dlech I might suggest that "Contributing" is a sub-optimal title for a document that I would expect to be titled "Before Asking For Help" Also, I got here from the ev3dev.org web site from the "issues" link. Any user following that path is going to miss the very helpful document that you linked.  Hopefully that is helpful feedback.

dlech, 2015-02-08:
> Is there something I'm doing wrong?

What is the version of the `ev3dev-base` package? (`dpkg-query -s ev3dev-base`). If it is < 8.0, then you need to upgrade. (This affects the EV3 UART sensors only, so if some sensors are working, but not others, this may be the problem).

However, I can see that the lua bindings in ev3dev-lang have not been updated for this kernel. For example, it still has "msensor" instead of "lego-sensor" [here](https://github.com/ev3dev/ev3dev-lang/blob/develop/lua/ev3dev.lua#L702).

> Is there an up-to-date-ish document that I could follow and then fix and the submit a patch for?

* Up-to-date docs for all of the kernel drivers are at <http://www.ev3dev.org/docs/>.
* The latest spec for ev3dev-lang is at <https://github.com/ev3dev/ev3dev-lang/blob/develop/wrapper-specification.md>
* The kernel changelog is at <https://github.com/ev3dev/ev3dev-kpkg/blob/ev3dev-jessie/debian/changelog>.

IIRC, lua was last updated for kernel 3.16.1-6-ev3dev, so you can use the changelog to see what has changed since then (mostly renames). The other docs should help if you need more clarification. You can use this info to fix up the lua library and submit a pull request to ev3dev-lang. FYI, @fdetro (the lua/C++ maintainer) mentioned recently he has been too busy for ev3dev lately.

> ACK! I do remember seeing these...

It seems like no one sees the box, so its not just you. :wink:
Maybe GitHub should add some flashing lights  :bulb: or :zap:  something :boom:.  

> I might suggest that "Contributing"...

The name `CONTRIBUTING` is some GitHub magic, so we don't have a choice there. Although the actual text of the [new issue](https://github.com/ev3dev/ev3dev/issues/new) box is:

    Please review the guidelines for contributing to this repository.

WasabiFan, 2015-02-09:
@andrewstarks I don't know how much time you are willing to put in to it, but if you find the lua binding helpful and have some free time that you are willing to spend on it, we'd love for some help porting that binding to our [autogen system](https://github.com/ev3dev/ev3dev-lang/tree/develop/autogen)! After that happens, very little work will be needed to adapt to kernel changes and keep it up-to-date.

andrewstarks, 2015-02-09:
It is version 8.0

This is great information and I'm excited to go digging. Thanks for the pointers. I know myself well enough not to promise that I'll actually get anything done in a reasonable time, but I shall at least poke.

-Andrew

andrewstarks, 2015-02-09:
@WasabiFan I shall poke twice!

dlech, 2015-05-21:
Lua bindings were updated this week, so should be working again.

closed 2015-05-21 by dlech