ev3dev/ev3dev #266: Feedback wanted on brickman changes

dlech, 2015-02-09:
This is picking up the discussion from #232 with @sharived... but the request for feedback is addressed to __everyone__.

I've been working on some changes to brickman:

* "Networking" menu is renamed to "Wireless and Networks"
* List of ConnMan Technologies that could be powered on/off with checkbox is removed from "Wireless and Networks"
    * Bluetooth powered checkbox is moved to "Bluetooth" menu.
    * Gadget powered is always set to true.
    * Wi-Fi powered checkbox is currently missing - needs a new home.
    * Wired powered checkbox is not show (and does not need to be).
* USB and Bluetooth menus are moved under "Wireless and Networks"
* Added new "Tethering" menu to "Wireless and Networks"
    * Only USB (gadget) and Bluetooth tethering enable checkboxes are show.
    * Technically is is possible to enable tethering on Wi-Fi, but I am hiding it for now because I think it would confuse people.
* If a bluetooth device has a network service, then you can link directly to the network connection info screen from the bluetooth device info page instead of having to back out to the "Wireless and Networks" menu and find the connection in "Network Connections".

I have created a LibreOffice drawing [here](https://raw.githubusercontent.com/ev3dev/brickman/ev3dev-jessie/doc/navigation.fodg) (right-click and Save Link As..., otherwise it will just open as text in your browser). This drawing shows all of the new windows and the navigation so you can see exactly what I am talking about.

I am looking for feedback from anyone interested.

* Is the navigation intuitive?
* Is there any part that does not make sense?
* Are there any important features that are missing (other than the already [open issues](https://github.com/ev3dev/ev3dev/labels/brickman))?
* If we create a new "Wi-Fi" menu screen...
    * What should be on it? (Needs powered checkbox to turn Wi-Fi on/off for sure)
    * How should it relate to the "Network Connections" screen?
    * Where does the "Scan Wi-Fi" button belong?
* Any other ideas/suggestions?

dsharlet, 2015-02-09:
I have a small suggestion: right now, if you want to change the settings for a wifi connection (IP, DNS, etc.), you have to find your wifi network in the list, even if you are already connected to it. This can be really annoying (there are probably 100+ wifi networks in range of my EV3...). 

My suggestion would be to make it so you can get to the settings for a wifi connection without having to find your network in the list first. Of course, this only applies to a network you are already connected to. If it is the first time connecting to a network, it is of course understandable that I would have to find my network first.

As a side note: the reason I have to go to the settings often enough for this to be annoying is that ev3dev seems to periodically forget the DNS servers. I only really notice this when I try to do an apt-get update, which fails because it is unable to resolve any of the hostnames used.

BertLindeman, 2015-02-09:
Any hint on a program or extension to open the .fodg file with on win7? 
OpenOffice draw opens the file OK, but shows nothing of interest :cry: 
Using OpenOffice 4.1.1

cavenel, 2015-02-09:
You can see the pdf here: https://www.dropbox.com/s/017123ppopd28ae/navigation.pdf?dl=0

BertLindeman, 2015-02-09:
Ah, that's better. Thank you.

dlech, 2015-02-09:
> you have to find your wifi network in the list, even if you are already connected to it

This should not be the case. If you are connected to a network, that connection should be at the top of the list. See [this](https://github.com/ev3dev/connman/blob/ev3dev-jessie/doc/overview-api.txt) for more info on how ordering of the connections is supposed to work (they are called services in ConnMan). If you are not seeing the expected behavior as described by that doc, please open a new issue.

I have also made some fixes with regards to race conditions when discovering services. It could be that since you have so many wireless networks around that the list is not being reordered properly.

> ev3dev seems to periodically forget the DNS servers

I [added a workaround](https://github.com/ev3dev/brickman/blob/ev3dev-jessie/lib/connman/Service.vala#L462) to brickman that may be causing this issue. I may remove it in the next release and see if it makes a difference. Actually, you could try running without brickman for a while and see if it makes a difference (`sudo systemctl stop brickman`).

dlech, 2015-02-24:
FYI, I have added a new Wi-Fi menu and updated the LibreOffice [drawing](https://raw.githubusercontent.com/ev3dev/brickman/ev3dev-jessie/doc/navigation.fodg) (right-click and Save Link As..., otherwise it will just open as text in your browser).

cho934, 2015-03-01:
Hello,
I've just tried the last image ev3dev-jessie-2015-02-24 using new brickman, but I don't succeed to connect by USB using my laptop in ubuntu 14.04.
The new menu "Tethering > Network info" displays the IP 192.168.2.1 instead of 10.42.0.3...
The menu IPV4 shows the correct IP 10.42.0.3...
Brickman is connected, but I have no ssh connection and no ping on 10.42.0.3.

Have you got an idea ?

dlech, 2015-03-01:
I haven't had time to updated the documentation yet, but when using tethering, your EV3 acts as a network server instead of a client. You cannot use both tethering and "All Network Connections" connection at the same time. If you need to access the Internet over USB, do not use tethering. If you want no configuration on your host computer, do use tethering. When tethering is enabled, you host computer will be automatically assigned an IP address from the EV3 and will be able to connect to the address specified in the tethering network info,

dlech, 2015-03-01:
@cho934, I am moving this conversation to #283 since it seem related.

dlech, 2015-04-30:
I haven't heard any complaints since these changes have been released, so I guess it must be good enough. As usual, please open a new issue for any further comments, suggestions, problems, etc.

closed 2015-04-30 by dlech