ev3dev/ev3dev #267: IC and Color sensors are not detected

Aaylor, 2015-02-11:
Hi,

We use an EV3 for a "school project", but our IC and Color sensors are not detected when they're plugged in brickman. In fact, when we plug the color sensor, the light doesn't turn up. But if we do that after removing the SD Card and booting to the default brickman OS, it works correctly.

1. `ev3dev-jessie-2014-12-01.img`

2. `3.16.1-7-ev3dev #2 PREEMPT Tue Nov 25 11:24:38 CST 2014`

3. `Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub`

4. `● brickman.service - Brick Manager
   Loaded: loaded (/lib/systemd/system/brickman.service; enabled)
   Active: active (running) since Sat 2000-01-01 01:00:44 CET; 15 years 1 months ago
 Main PID: 416 (brickman)
   CGroup: /system.slice/brickman.service
           └─416 /usr/sbin/brickman
Jan 01 01:00:44 ev3dev systemd[1]: Starting Brick Manager...
Jan 01 01:00:45 ev3dev systemd[1]: Started Brick Manager.`

The first thing we've made: `apt-get update` `apt-get upgrade` `apt-get dist-upgrade`, and no more update are available.

We looked at the issue #265, but it doesn't help. It just shows a weird thing happening : our TOUCH Sensor (which works well), appears in the folder `/sys/class/msensor/sensor<int>` instead of `/sys/class/lego-sensor/sensor<int>`. If i really understand what it was told in #265, it should be an update problem... but no more update are available...

Thanks for help !

dlech, 2015-02-11:
I'm not sure what is going on here. In #265, his kernel updated correctly, but I have been discussing with @AleksiAhtiainen in ev3dev/ev3dev-lang#43 the same issue with the kernel upgrade not taking effect. (And @BertLindeman might have this problem too.)

I know we have issues in the past with the firstboot script not running correctly sometimes, and I am thinking this may be the problem here.

To verify that the firstboot script ran correctly, you should see the following:

1. There should be 4 ssh host key files:  `/etc/ssh/ssh_host_{rsa|dsa|ecdsa|ed25519}_key` (you would not be able to connect with ssh if you did not have at least one of these).

2. There should be a file `/etc/modprobe.d/usb-net.conf` with contents like:

         options g_cdc dev_addr==02:16:53:3F:48:38 host_addr=12:16:53:3F:48:38
         options g_ether dev_addr==02:16:53:3F:48:38 host_addr=12:16:53:3F:48:38

    `02:16:53` and `12:16:53` will match, but the last 3 numbers will be different for each EV3.

3. You should have a swap partition. Running `sudo lvscan` should show something like:

          ACTIVE            '/dev/ev3devVG/root' [7.30 GiB] inherit
          ACTIVE            '/dev/ev3devVG/swap' [64.00 MiB] inherit

     and `/etc/fstab` should have the line:

        /dev/mapper/ev3devVG-swap none            swap   sw                                  0      0

4. There should be a file `/etc/flash-kernel/db` with the contents:

        # To override fields include the Machine field and the fields you wish to
        # override.
        #
        # e.g. to override Boot-Device on the Dreamplug to sdb rather than sda
        #
        #Machine: Globalscale Technologies Dreamplug
        #Boot-Device: /dev/sdb1
        
        Machine: LEGO MINDSTORMS EV3 Programmable Brick
        Boot-Device: /dev/mmcblk0p1

I am guessing that you will find that `/etc/flash-kernel/db` is missing the last 2 lines, which is why the kernel upgrade is not taking effect. If this is the case, please check the other items in the list so that we can figure out where the failure is happening.

Aaylor, 2015-02-11:
Items 1 and 2 are correct, but not the last two.

`/etc/fstab` display this:
```
#<file system>            <mount point>   <type> <options>                           <dump> <pass>
/dev/mmcblk0p1            /media/mmc_p1   vfat   defaults,errors=remount-ro,noatime  0      1
/dev/mapper/ev3devVG-root /               ext3   defaults,errors=remount-ro,noatime  0      1
proc                      /proc           proc   defaults                            0      0
```
We don't have `/dev/mapper/ev3devVG-swap`

And as you guessed, in `/etc/flash-kernel/db`, the last two lines are missing.
Should we try to reinstall entirely the system, or is there any way to solve this problem ?

dlech, 2015-02-11:
The flash-kernel problem is easy enough to fix - just add those two lines to that file. Then run `sudo flash-kernel` and reboot.

If `/dev/mapper/ev3devVG-swap` exists (you did not say the output of `sudo lvscan`), you should be able to add the missing line to the `/etc/fstab` file and be OK. Some packages probably failed to upgrade though because there was not swap file and `apt-get` ran out of memory. For example, there were errors related to perl not being able to find the language mentioned in https://github.com/ev3dev/ev3dev-lang/issues/43#issuecomment-73837650 that are probably related to this.

The actual firstboot script is [here](https://github.com/ev3dev/brickstrap/blob/ev3dev/ev3dev-jessie/root/etc/init.d/firstboot).

Aaylor, 2015-02-11:
Ah, yes, sorry. Here is the output of `sudo lvscan`: 
```
  ACTIVE            '/dev/ev3devVG/root' [7.32 GiB] inherit
  ACTIVE            '/dev/ev3devVG/swap' [64.00 MiB] inherit
```

Thanks a lot, i'll try this as soon as possible !

BertLindeman, 2015-02-11:
@dlech 
> (And @BertLindeman might have this problem too.)

No kernel problem here, was just trying to help :smiley:

dlech, 2015-02-11:
@BertLindeman, I thought you said you had the same older kernel version, but now I see you were just quoting the previous comment. :wink:

AleksiAhtiainen, 2015-02-13:
Thank you, this debug information worked for the kernel problem also for me. 

I now got the new kernel  ```3.16.7-ckt4-ev3dev1``` to boot. I was missing the exact same lines as @Aaylor in ```/etc/fstab``` and ```/etc/flash-kernel/db``` . After adding the missing lines to the files and running ```sudo flash-kernel``` again, the new kernel now boots.

Aaylor, 2015-02-13:
Thanks you, it work very well for us too !

dlech, 2015-02-25:
In the `ev3dev-jessie-2015-02-24.img` (just released), I have change how the flash-kernel config file is updated, so that part of the problem is solved. However, it is still possible for creating the swap file to fail. Users will notice this when they try to apt-get install or upgrade certain packages because they will run out of memory, so this still needs to be fixed.

I am considering creating a new firstboot program with a user interface so that 1) The firstboot stuff is not run until after things quiet down a bit and 2) The user can be notified if there are errors.

If we do something like this, I would also include the suggestion from #280 of expanding the root partition to fill the SD card.

dlech, 2015-02-25:
Here is an instance of the ssh keys not being create: #216.

G33kDude, 2015-03-27:
I've just taken a 32G class 10 uSD card that used to have the 2014-12-01 release on it (to the best of my knowledge), and put the latest release 2015-02-24 on it. `sudo lvscan` is throwing warnings left and right

```
root@ev3dev:/etc/ssh# sudo lvscan
  WARNING: Duplicate VG name ev3devVG: YGHnF0-pRCC-SWRE-je7i-bCRs-RKwV-hdNn8F (created here) takes precedence over 42ekjh-ky3F-IEoo-MyyS-taUY-JSvH-Vmy0is
  WARNING: Duplicate VG name ev3devVG: YGHnF0-pRCC-SWRE-je7i-bCRs-RKwV-hdNn8F (created here) takes precedence over 42ekjh-ky3F-IEoo-MyyS-taUY-JSvH-Vmy0is
  WARNING: Duplicate VG name ev3devVG: Existing YGHnF0-pRCC-SWRE-je7i-bCRs-RKwV-hdNn8F (created here) takes precedence over 42ekjh-ky3F-IEoo-MyyS-taUY-JSvH-Vmy0is
  Couldn't find device with uuid kT4fXU-GFed-V2hs-W0OU-iPXa-5Ekn-JDbfa1.
  inactive          '/dev/ev3devVG/root' [29.62 GiB] inherit
  inactive          '/dev/ev3devVG/swap' [64.00 MiB] inherit
```

Perhaps it's related to #293? I haven't tried expanding my SD card yet.

dlech, 2015-03-27:
It looks like the root cause of this is re-flashing an SD card. It seems that part of the partition table, or at least the lvm partition table is surviving the process and is picked up by the fresh system. This causes the `vgextend` to fail in the firstboot script. The script exits on error, so everything after that point does not get run.

I think the ideal solution here would be to find where this information is hiding - presumably in the empty space of the SD card right after the 900MB ev3dev image. If it is always in the same place, we can easily wipe it out as part of the first boot script. Or maybe we should use `fallocate`.

cho934, 2015-03-27:
We had the same problem installing image ev3dev-jessie-2015-02-24.
We executed fdisk to remove all partitions on the sdcard. Then we wrote the image.
As a result, the unknown device disappeared.
Now we still had the duplicate error of ev3devVG.

We fixed the problem using the following command:
```
root@ev3dev:~# sudo lvscan
  WARNING: Duplicate VG name ev3devVG: YGHnF0-pRCC-SWRE-je7i-bCRs-RKwV-hdNn8F (created here) takes precedence over 42ekjh-ky3F-IEoo-MyyS-taUY-JSvH-Vmy0is
  WARNING: Duplicate VG name ev3devVG: YGHnF0-pRCC-SWRE-je7i-bCRs-RKwV-hdNn8F (created here) takes precedence over 42ekjh-ky3F-IEoo-MyyS-taUY-JSvH-Vmy0is
  WARNING: Duplicate VG name ev3devVG: Existing YGHnF0-pRCC-SWRE-je7i-bCRs-RKwV-hdNn8F (created here) takes precedence over 42ekjh-ky3F-IEoo-MyyS-taUY-JSvH-Vmy0is
  Couldn't find device with uuid kT4fXU-GFed-V2hs-W0OU-iPXa-5Ekn-JDbfa1.
  inactive          '/dev/ev3devVG/root' [864.00 MiB] inherit
  inactive          '/dev/ev3devVG/swap' [64.00 MiB] inherit
```

```
root@ev3dev:~# vgdisplay
 WARNING: Duplicate VG name ev3devVG: YGHnF0-pRCC-SWRE-je7i-bCRs-RKwV-hdNn8F (created here) takes precedence over 42ekjh-ky3F-IEoo-MyyS-taUY-JSvH-Vmy0is
  WARNING: Duplicate VG name ev3devVG: YGHnF0-pRCC-SWRE-je7i-bCRs-RKwV-hdNn8F (created here) takes precedence over 42ekjh-ky3F-IEoo-MyyS-taUY-JSvH-Vmy0is
  WARNING: Duplicate VG name ev3devVG: Existing YGHnF0-pRCC-SWRE-je7i-bCRs-RKwV-hdNn8F (created here) takes precedence over 42ekjh-ky3F-IEoo-MyyS-taUY-JSvH-Vmy0is
  Couldn't find device with uuid kT4fXU-GFed-V2hs-W0OU-iPXa-5Ekn-JDbfa1.
  --- Volume group ---
  VG Name               ev3devVG
  System ID             
  Format                lvm2
  Metadata Areas        1
  Metadata Sequence No  4
  VG Access             read/write
  VG Status             resizable
  MAX LV                0
  Cur LV                2
  Open LV               0
  Max PV                0
  Cur PV                2
  Act PV                1
  VG Size               7.36 GiB
  PE Size               4.00 MiB
  Total PE              1885
  Alloc PE / Size       232 / 928.00 MiB
  Free  PE / Size       1653 / 6.46 GiB
  VG UUID               YGHnF0-pRCC-SWRE-je7i-bCRs-RKwV-hdNn8F
   
  WARNING: Duplicate VG name ev3devVG: YGHnF0-pRCC-SWRE-je7i-bCRs-RKwV-hdNn8F (created here) takes precedence over 42ekjh-ky3F-IEoo-MyyS-taUY-JSvH-Vmy0is
  --- Volume group ---
  VG Name               ev3devVG
  System ID             
  Format                lvm2
  Metadata Areas        1
  Metadata Sequence No  2
  VG Access             read/write
  VG Status             resizable
  MAX LV                0
  Cur LV                1
  Open LV               1
  Max PV                0
  Cur PV                1
  Act PV                1
  VG Size               864.00 MiB
  PE Size               4.00 MiB
  Total PE              216
  Alloc PE / Size       216 / 864.00 MiB
  Free  PE / Size       0 / 0   
  VG UUID               42ekjh-ky3F-IEoo-MyyS-taUY-JSvH-Vmy0is
```
```

vgreduce --removemissing ev3devVG --force
grename YGHnF0-pRCC-SWRE-je7i-bCRs-RKwV-hdNn8F ev3devold
\\
root@ev3dev:~# vgs
  VG        #PV #LV #SN Attr   VSize   VFree
  ev3devVG    1   1   0 wz--n- 864.00m    0 
  ev3devold   1   1   0 wz--n-   6.52g 6.46g
```
```
root@ev3dev:~# vgremove ev3devold
Do you really want to remove volume group "ev3devold" containing 1 logical volumes? [y/n]: y
  Logical volume "swap" successfully removed
  Volume group "ev3devold" successfully removed
```
```
root@ev3dev:~# vgs
  VG       #PV #LV #SN Attr   VSize   VFree
  ev3devVG   1   1   0 wz--n- 864.00m    0 
```
```
root@ev3dev:~# vgextend ev3devVG /dev/mmcblk0p3
  Volume group "ev3devVG" successfully extended
```
```
root@ev3dev:~# vgs
  VG       #PV #LV #SN Attr   VSize VFree
  ev3devVG   2   1   0 wz--n- 7.36g 6.52g
```
```
root@ev3dev:~# lvextend --extents +100%FREE --resizefs /dev/ev3devVG/root /dev/mmcblk0p3
  Size of logical volume ev3devVG/root changed from 864.00 MiB (216 extents) to 7.36 GiB (1885 extents).
  Logical volume root successfully resized
resize2fs 1.42.12 (29-Aug-2014)
Filesystem at /dev/mapper/ev3devVG-root is mounted on /; on-line resizing required
old_desc_blocks = 1, new_desc_blocks = 1
Performing an on-line resize of /dev/mapper/ev3devVG-root to 1930240 (4k) blocks.
The filesystem on /dev/mapper/ev3devVG-root is now 1930240 (4k) blocks long.
```

Then it's working (without swap).
I hope it can help to find a better solution.

dlech, 2015-04-01:
I've just released a new image with a fix for this. Let me know if you have duplicate volume groups after installing this image. You can check by running `vgs` on the EV3.

dlech, 2015-04-01:
https://github.com/ev3dev/ev3dev/releases/tag/ev3dev-jessie-2015-03-31

closed 2015-04-01 by dlech

thedutchy, 2015-04-05:
Hello everyone!

I've had the same problem with my EV3, just updated to ev3dev-jessie-2015-03-31 and want to share the results.

Before the update, I was using ev3dev-jessie-2014-12-01 but had to reinstall. After the reinstall, everything appeared fine, but using df I no longer showed the full extent of my SD-card available. Using the lvextend command did not help either. I checked the settings above, and (3) and (4) were incorrect.

So, I decided to update to ev3dev-jessie-2015-03-31 to see if this resolves the problem. After using the new image, I found out that (3) and (4) are now correct. To my delight, lvextend --extents +100%FREE --resizefs /dev/ev3devVG/root /dev/mmcblk0p3 now also works perfectly again and I have the full size of my SD card available.

Thanks for the good work dlech!

G33kDude, 2015-04-05:
It's great to hear that our endless tinkering on IRC actually got a working fix out! How large is your SD card?

thedutchy, 2015-04-05:
Yes, I am also quite happy it is working again. It's a 32GB SD card, and before the update I only had to access to 1GB. A bit of a waste :-)