ev3dev/ev3dev #268: HiTechnic Compass Sensor and IRSeeker V2 don't work

knefel, 2015-02-11:
The HiTechnic NXT IRSeeker V2 and HiTechnic NXT Compass Sensor don't appear in /sys/class/lego-sensor. I can detect them at the correct adress (0x08 and 0x01) with i2cdetect. For example the HiTechnic Sensor Multiplexer and the HiTechnic NXT SuperPro Prototype Board connecting correctly. What should I do in order to fix the problem?

dlech, 2015-02-12:
1. Please see [guidelines for contributing](https://github.com/ev3dev/ev3dev/blob/master/CONTRIBUTING.md). (It is the link in the yellow box you see when you create a new issue). It could be that you have an old kernel version (see #133).
2. Run [this script](https://gist.github.com/dlech/d86161aea84db05f8ffb) on your EV3 and paste the output here. We need to verify that that the vendor and product strings match the kernel driver ([here](https://github.com/ev3dev/lego-linux-drivers/blob/master/sensors/nxt_i2c_sensor_defs.c#L647) and [here](https://github.com/ev3dev/lego-linux-drivers/blob/master/sensors/nxt_i2c_sensor_defs.c#L1063)).

knefel, 2015-02-12:
Some additional information:
<ol>
<li>
Image file: ev3dev-jessie-2014-10-07.img <br>
kernel: 3.16.7-ckt4-ev3dev1 #2 PREEMPT Mon Jan 19 21:08:06 CST 2015
</li>
<li>
<pre>
<code>
bus:  3  port:  1 
         Input port 1:
             Address: 0x08
             Version: 'Vers 1.4'
              Vendor: 'HiTechnc'
             Product: 'NewIRDir'
bus:  4  port:  2 
         Input port 2:
             Address: 0x01
             Version: '�V2.1   '
              Vendor: 'HITECHNC'
             Product: 'Compass '
<code>
</pre>
</li>
</ol>

dlech, 2015-02-12:
You have the most recent kernel, so that is good.

I can see in the case of the Compass sensor, the vendor string was not capitalized in the driver definition, so I have fixed that. I am not seeing anything obviously wrong with the IRSeeker.

dlech, 2015-02-12:
Looks like I had a driver name mismatch on the IRSeekerV2. It should be fixed now too.

dlech, 2015-02-22:
Fix has been release in [v3.16.7-ckt6-ev3dev1](http://www.ev3dev.org/news/2015/02/22/Kernel-Release-v3.16.7-ckt6-ev3dev1/).

closed 2015-02-22 by dlech