ev3dev/ev3dev #269: apt-get upgrade fails

gvukasin, 2015-02-12:
I do not have the tacho-motors and lego-sensor-class classes in /sys/class/. I did not initially do the 'apt-get update", 'apt-get upgrade', nor 'apt-get dist-upgrade'. So I thought the update or upgrade would have these two classes. I ran 'apt-get update', which was successful. But when I run 'apt-get upgrade' I get the following message in terminal: 
debconf: delaying package configuration, since apt-utils is not installed
dpkg: dependency problems prevent processing triggers for dbus:
 dbus depends on libdbus-1-3 (>= 1.7.6); however:
  Package libdbus-1-3:armel is not configured yet.

Then the next message repeats many times until terminal gives up:
dpkg: error processing package dbus (--configure):
 dependency problems - leaving triggers unprocessed
dpkg: dependency problems prevent processing triggers for dbus:
 dbus depends on libdbus-1-3 (>= 1.7.6); however:
  Package libdbus-1-3:armel is not configured yet.

What should I do? 

Thanks in advance.

gvukasin, 2015-02-12:
I forgot to mention that my kernel is: 3.16.1-7-ev3dev #2 PREEMPT Tue Nov 25 11:24:38 CST 2014
and the BrickManager: ● brickman.service - Brick Manager
   Loaded: loaded (/lib/systemd/system/brickman.service; enabled)
   Active: active (running) since Thu 2015-02-12 12:31:29 EST; 15min ago
 Main PID: 393 (brickman)
   CGroup: /system.slice/brickman.service
           └─393 /usr/sbin/brickman

Feb 12 12:32:13 ev3dev brickman[393]: ** (process:393): CRITICAL **: Service.vala:82: GDBus.Error:net.connman.Error.InvalidService: Invalid service

dlech, 2015-02-12:
See #239.

closed 2015-02-25 by dlech