ev3dev/ev3dev #270: No Module named ev3.lego

vhf974, 2015-02-12:
Hello,

I followed the document 'Getting Started'. Python works. I managed to run this program:
http://www.ev3dev.org/projects/2014/10/06/LED-Demo/

But when a python program there is the following line:

1 from ev3.lego import MediumMotor

I get the following message:
ImportError: No Module named ev3.lego

Thank you kindly answer me, I'm interested simple examples to start programming from avoiding programming object.

Sorry for my English.

Greetings.

dlech, 2015-02-12:
It sounds like you are trying to use <https://github.com/topikachu/python-ev3>. This library does not come pre-installed. You have to install it yourself. The link has instructions on how to install it.

vhf974, 2015-02-13:
Hello,

Thank you, yes indeed I understand a little better. But it still does not work.

root @ ev3dev: ~ # workon ev3_py27
(ev3_py27) ev3dev root @: ~ # python
Python 2.7.8 (default, October 18, 2014, 5:24:08 p.m.)
[GCC 4.9.1] is linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> From ev3.lego import MediumMotor
>>> D = MediumMotor ()
Traceback (most recent call last):
   File "<stdin>", line 1, in <module>
   file"/root/.virtualenvs/ev3_py27/local/lib/python2.7/site-packages/python_ev3-0.0.2.6-py2.7.egg/ev3/lego.py", line 172, in __init__
   file"/root/.virtualenvs/ev3_py27/local/lib/python2.7/site-packages/python_ev3-0.0.2.6-py2.7.egg/ev3/ev3dev.py", line 323, in __init__
ev3.ev3dev.NoSuchMotorError: No such motor type = port = minitacho
>>>

If I may, I think until the installation of the virtual environnment, your tutorial is very well done. Well Done.

But then I can not run the examples are here:
https://github.com/topikachu/python-ev3/tree/master/test

I keep getting error messages like this:

Attach a motor on port A continues Then
-------------------------------------------------- -------------------------
AttributeError Traceback (most recent call last)
<ipython-input-4-cdb331ca1706> in <module> ()
      46
      47 if __name__ == '__main__':
---> 48 unittest.main ()

/usr/lib/python2.7/unittest/main.pyc in __init __ (self, module, defaultTest, argv, testrunner, testLoader, exit, verbosity, failfast, catchbreak, buffer)
      92 = self.testLoader testLoader
      93 = self.progName os.path.basename (argv [0])
---> 94 self.parseArgs (argv)
      95 self.runTests ()
      96

/usr/lib/python2.7/unittest/main.pyc in ParseArgs (self, argv)
     147 else:
     148 self.testNames = (self.defaultTest,)
-> 149 self.createTests ()
     Except getopt.error 150, msg:
     151 self.usageExit (msg)

/usr/lib/python2.7/unittest/main.pyc in createTests (self)
     156 else:
     157 self.test = self.testLoader.loadTestsFromNames (self.testNames,
-> 158 self.module)
     159
     160 def _do_discovery (self, argv, Loader = None):

/usr/lib/python2.7/unittest/loader.pyc in loadTestsFromNames (self, names, module)
     128 of string specifiers. See 'loadTestsFromName ()'.
     129 "" "
-> = 130 suites [self.loadTestsFromName (name, module) for name in names]
     131 return self.suiteClass (suites)
     132

/usr/lib/python2.7/unittest/loader.pyc in loadTestsFromName (self, name, module)
      98 obj = Module
      For part 99 in parts:
-> 100 parent obj = obj, getattr (obj, share)
     101
     102 if isinstance (obj, types.ModuleType)

AttributeError: 'module' object Has No attribute '/ root /'

Sorry for my English.

Greetings.

dlech, 2015-02-13:
python-ev3 has it's own place to report issues at <https://github.com/topikachu/python-ev3/issues>.
 You may have better luck asking for help there.

dlech, 2015-04-16:
Closing since this is a problem with python-ev3

closed 2015-04-16 by dlech

GogicM, 2017-05-11:
I am sorry for jumping into closed topic, but I have to ask, is there any way to use python-ev3dev module WITHOUT ev3dev OS, I need to use only python module on Raspbian OS? Thank you for your answer.
Regards,
Milan

dlech, 2017-05-11:
Don't be sorry... open a new issue. :smiley: