ev3dev/ev3dev #272: Can't connect WIFI ubuntu

Tratost, 2015-02-13:
Hello,

I'm trying to use network on the brickman, but unfortunately, until there,
it never works really fine. I'll explain what happened. I plugged the usb cable
on my ubuntu and follow the steps of this tutorial :
http://www.ev3dev.org/docs/tutorials/setting-up-ethernet-over-usb-on-linux-networkmanager/

After that, I can connect myself on the brickman via ssh. However, I can't use network
in the brickman itself (can't make apt-get update for example). I have realized that
the network status is only "connected" and not "online". Is there a reason for that ?
I precise also that, on the same network, it work on my friend's OSX but not on my ubuntu.
How can I solve this problem ?
Thanks in advance !

dlech, 2015-02-13:
Post some screenshots of the EV3 and your computer so we can see how you have things setup. Mainly looking for IP address, gateway and DNS on the EV3 and that "Shared to other computers" is selected on your host computer. Also, you host computer has to have an Internet connection to share. If the EV3 says "connected", then you have a network connection, but not an Internet connection. If it says "online" then you have an Internet connection.

Tratost, 2015-02-16:
I have solve the problem, thanks to your advices.
For the DNS, I hadn't change the IP adress after my firends, and it was the same configuration as the OSX. Changing manually  the DNS and add the "10.42.0.1" address instead of "192.168.2.1" help me to access to my Internet connection.
Thank you.

dlech, 2015-02-16:
I'm glad you got it working.

closed 2015-02-16 by dlech