ev3dev/ev3dev #273: Absolute IMU sensor: device or resource busy

jdchristensen, 2015-02-16:
I have a Mindsensor Absolute IMU gyro/tilt/compass sensor connected to my EV3.
I can adjust the mode and read the values by directly accessing files in /sys.
But when I run the following program

    from ev3.mindsensors import AbsoluteIMU
    d = AbsoluteIMU(2)
    print(d.version)

I get the following backtrace

    Traceback (most recent call last):
      File "Abs_IMU_test.py", line 8, in <module>
        print(d.vendor_id)
      File "/usr/local/lib/python2.7/dist-packages/ev3/mindsensors.py", line 12, in vendor_id
        return self.read_byte_array_as_string(0x08, 8)
      File "/usr/local/lib/python2.7/dist-packages/ev3/ev3dev.py", line 418, in read_byte_array_as_string
        return ''.join(chr(r) for r in self.read_byte_array(reg, _len))
      File "/usr/local/lib/python2.7/dist-packages/ev3/ev3dev.py", line 415, in read_byte_array
        return [self.read_byte_data(reg + r) for r in range(_len)]
      File "/usr/local/lib/python2.7/dist-packages/ev3/ev3dev.py", line 389, in proxied_smb_method
        return getattr(self.b, method)(self.addr, *args, **kwargs)
    IOError: [Errno 16] Device or resource busy

This happens even if I insert delays into the script.

Let me know what other information I can provide.

dlech, 2015-02-16:
1. Please review the [guidelines for contributing](https://github.com/ev3dev/ev3dev/blob/master/CONTRIBUTING.md) to this repository. (This is the link in the yellow box you see when you create a new issue).
2. It sounds like you are using python-ev3. It looks like the code there is accessing the sensor directly using I2C commands instead of using the lego-sensor driver class. I would suggest submitting a pull request to python-ev3 to change this.

dlech, 2015-02-16:
Forgot to say that the reason you are getting a busy error is because the ev3dev driver has taken ownership of the device. You could unbind the ev3dev driver and use the existing python-ev3 code.

jdchristensen, 2015-02-16:
Thanks for the quick help.  I think your analysis is right, so this is not a bug with ev3dev.  I will close this issue and submit one to python-ev3.

Sorry for not reading the guidelines.  I misinterpreted "contributing" as "contributing code", so I didn't read it.  Maybe rename it to "Guidelines for posting issues", or say "Please read ... before creating a new issue"?

closed 2015-02-16 by jdchristensen

dlech, 2015-02-16:
> I misinterpreted "contributing" as "contributing code",

Actually that is what it was [originally intended for](https://github.com/blog/1184-contributing-guidelines
). I am not aware of any way to change it other than submit a feature request to GitHub.