ev3dev/ev3dev #274: Running programs with root permission (or resolving user bluetooth permissions)

andybets, 2015-02-16:
I'm using a python script which provides a bluetooth rfcomm server using python-bluez, and it works fine as root, but I get permissions errors when running as 'user' (ev3dev-jessie-2014-12-01.img).

Is it possible to change anything easily in brickman to run scripts as root (security isn't an issue)?  Alternatively, if it's simpler, any idea how might I get bluetooth working as a non-root user?

Any help would be much appreciated.

dlech, 2015-02-16:
The reason is for brickman not running programs as root is not so much security as it is to keep you from accidentally breaking things, like "oops, I just deleted some really important files".

To get around this though, you could write a wrapper script for your program that invokes sudo. I think you can setup your user account so that it can invoke sudo without asking for a password.

But it would be better for everyone if we can figure out how to do it as non-root. Can you post your code?

dlech, 2015-02-16:
I got a little distracted by this, so I figured it out. :wink: 

python-bluez is basically using BlueZ 4 and we have BlueZ 5. I am guessing that you already figured out about adding `--compat` to the ExecStart line in `bluetooth.service`. (By the way, if you didn't know, you should copy `/lib/systemd/system/bluetooth.service` to `/etc/systemd/system/bluetooth.service` and edit the file in `/etc`, otherwise the changes will be written over during an upgrade).

What the `--compat` option does is add a unix socket file at `/var/run/sdp`. So, this is the file that needs the permission changed. You can of course do this manually, but who wants to do that every time they reboot? So, systemd to the rescue. I created two unit files.

`/etc/systemd/system/var-run-sdp.path`:

    [Unit]
    Descrption=Monitor /var/run/sdp
    
    [Install]
    WantedBy=bluetooth.service
    
    [Path]
    PathExists=/var/run/sdp
    Unit=var-run-sdp.service

`/etc/systemd/system/var-run-sdp.service`:

    [Unit]
    Description=Set permission of /var/run/sdp
    
    [Install]
    RequiredBy=var-run-sdp.path
    
    [Service]
    Type=simple
    ExecStart=/bin/chgrp bluetooth /var/run/sdp

I'm not sure if I got the dependencies correct, but I enabled and ran them like this:

    sudo systemctl daemon-reload
    sudo systemctl enable var-run-sdp.path
    sudo systemctl enable var-run-sdp.service
    sudo systemctl start var-run-sdp.path

What this does is monitors for the existence of `/var/run/sdp` (the path unit file) and when is created, set the group to `bluetooth` (the service unit file).

Now, you just have to add your user to the bluetooth group and you are good to go.

    sudo usermod +aG bluetooth <username>
    newgrp bluetooth

dlech, 2015-02-17:
And now after I have done all of that, it would probably be better if you could find a library that uses the new BlueZ 5 DBus API instead of the old sdp stuff.

andybets, 2015-02-17:
Thanks very much!

Yes, I added --compat, but also tried a few other things, and slightly lost track of which may be relevant, so didn't think to mention it.

I did look at using DBus API, but couldn't seem to find a python example for rfcomm, and thought the learning curve looked too steep for right now - but I appreciate this approach is effectively deprecated.  Once I have a prototype working, I might return to this to try using BlueZ 5 DBus, as I'd like to understand Bluetooth better anyway.

Thanks again. I'll implement it as soon as the batteries are charged! - I still need to get a rechargeable ;)

andybets, 2015-02-17:
Great!  It's working now.

closed 2015-02-17 by andybets