ev3dev/ev3dev #275: Missing classes

gvukasin, 2015-02-18:
I have updated to the new kernel and I am missing the following classes and drivers:
tacho-motor-class
ev3-tacho-motor
ev3-analog-sensor
lego-sensor-class

Should I manually put them on from the github file "lego-linux-drivers-master"?

dlech, 2015-02-18:
> I have updated to the new kernel

Please be specific about exactly which version (`uname -r`)

> I am missing the following classes and drivers

If you are, there is something wrong with your installation (like bad SD card or something). More than likely though, you do have them.

Plug in a sensor and a motor then run `ls /sys/class` on your EV3. You should see `lego-port`, `lego-sensor` and `tacho-motor`. If you do, then you have all of these drivers.

gvukasin, 2015-02-18:
I was being an idiot, I found the drivers. Sorry for the premature posting.

closed 2015-02-18 by gvukasin

dlech, 2015-02-18:
I think we all do things like that sometimes. :wink: