ev3dev/ev3dev #276: Install problem

LarsBudek, 2015-02-19:
I try to install ev3dev-jessie-2014-10-07 and follow the “get started guide”… Everything seems to work. Format sdhc 8GB card. Install the image fil on the card using Win32diskimager. Boot the EV3 on the card… Flashing LED for some time and the Green LED. But the screen is blank…  What is wrong? Please give me a hit…

dlech, 2015-02-19:
Try rebooting. If that does not work, try flashing the SD card again. If that does now work, try a different SD card.

WasabiFan, 2015-02-19:
@dlech
>  If that does now work

That looks like a typo, but makes the sentence meaning change (and could be confusing)

dlech, 2015-02-19:
Thanks for pointing that out. I have corrected it.

dlech, 2015-02-25:
I noticed that you are trying to install an old release (ev3dev-jessie-2014-10-07). Please use the latest image file.

marecl, 2015-03-16:
I had the same problem, blank screen or ev3 stock OS was booting
Try to dd if=/dev/zero of=/path/to/disk and then write image to raw disk
Or in the format settings select "write disk with zeros" to make sure there will be true "free space"
It has worked for me.

dlech, 2015-03-16:
Hmm... I think this might be related to #267.

dlech, 2015-04-16:
Pretty sure this is fixed in the ev3dev-jessie-2015-03-31 release

closed 2015-04-16 by dlech