ev3dev/ev3dev #277: bind: Address already in use

artpoz, 2015-02-20:
After running brickfpv.init from this project
http://www.pabr.org/bricks/brickfpv/brickfpv.en.html
I've got error:
bind: Address already in use
I'm using latest version of ev3dev (2014-12-01)
Before update everything worked well on version ev3dev-jessie-2014-10-07.img

dlech, 2015-02-20:
Here is the scripts for reference:

brickfpv.init

```
#!/bin/sh

exec  > /tmp/brickfpv.log  2>&1

start() {
    connmanctl tether bluetooth on
    cd /root/brickfpv-1.0
    ./sixpair
    ./brickrc.sh  &
    ./streamer.sh 81 "./v4l2http -c /dev/video0 -r 15 --mjpeg -w 160 -h 120"  &
    ./streamer.sh 82 "./v4l2http -c /dev/video1 -r 15 --mjpeg -w 160 -h 120"  &
    ./httpd.sh  &
}

stop() {
    killall -9 streamer.sh brickrc.sh httpd.sh brickrc v4l2http
    fuser -k 80/tcp 81/tcp 82/tcp 83/tcp 84/tcp
}

case "$1" in
    start) start ;;
    stop) stop ;;
esac
```

httpd.sh

```
#!/bin/sh

while true; do
    nc.traditional  -l  -p 80  -c "./http.sh"; 
done
```

It looks like it is using ports 80,81,82,83 and 84. You can run `netstat --listening --numeric-ports --programs` to see what is using one of these ports already.

artpoz, 2015-02-21:
latest version after upgrade:

root@ev3dev:~/brickfpv-1.0/src# ./brickrc -v ps3_tank.config
Found ev3dev /sys/class/tacho-motor
/sys/class/tacho-motor/./port_name
5-continue/sys/class/tacho-motor/../port_name
5-continue/sys/class/tacho-motor/tacho-motor0/port_name
Found ev3dev motor B as tacho-motor0
/sys/class/tacho-motor/tacho-motor1/port_name
Found ev3dev motor C as tacho-motor1
brickrc modified by artpoz
init
sound_tone: not implemented
speed 0hid
Waiting for incoming Bluetooth connection...
**bind: Address already in use**
root@ev3dev:~/brickfpv-1.0/src# netstat --listening --numeric-ports
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State
tcp        0      0 0.0.0.0:111             0.0.0.0:*               LISTEN
tcp        0      0 192.168.0.1:53          0.0.0.0:*               LISTEN
tcp        0      0 127.0.0.1:53            0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:38998           0.0.0.0:*               LISTEN
udp        0      0 192.168.0.1:53          0.0.0.0:*
udp        0      0 127.0.0.1:53            0.0.0.0:*
udp        0      0 0.0.0.0:111             0.0.0.0:*
udp        0      0 169.254.81.95:123       0.0.0.0:*
udp        0      0 127.0.0.1:123           0.0.0.0:*
udp        0      0 0.0.0.0:123             0.0.0.0:*
udp        0      0 0.0.0.0:896             0.0.0.0:*
udp        0      0 127.0.0.1:913           0.0.0.0:*
udp        0      0 0.0.0.0:59098           0.0.0.0:*
Active UNIX domain sockets (only servers)
Proto RefCnt Flags       Type       State         I-Node   Path
unix  2      [ ACC ]     STREAM     LISTENING     875      /run/systemd/private
unix  2      [ ACC ]     STREAM     LISTENING     2927     /var/run/dbus/system_bus_socket
unix  2      [ ACC ]     STREAM     LISTENING     898      /run/lvm/lvmetad.socket
unix  2      [ ACC ]     SEQPACKET  LISTENING     901      /run/udev/control
unix  2      [ ACC ]     STREAM     LISTENING     904      /run/systemd/journal/stdout
unix  2      [ ACC ]     STREAM     LISTENING     2751     /run/rpcbind.sock

working version:

root@ev3dev:~/brickfpv-1.0# netstat --listening --numeric-ports --programs
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State                      PID/Program name
tcp        0      0 127.0.0.1:53            0.0.0.0:*               LISTEN                     335/connmand
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN                     1/init
udp        0      0 127.0.0.1:53            0.0.0.0:*                                          335/connmand
udp        0      0 192.168.1.11:123        0.0.0.0:*                                          393/ntpd
udp        0      0 127.0.0.1:123           0.0.0.0:*                                          393/ntpd
udp        0      0 0.0.0.0:123             0.0.0.0:*                                          393/ntpd
Active UNIX domain sockets (only servers)
Proto RefCnt Flags       Type       State         I-Node   PID/Program name    P               ath
unix  2      [ ACC ]     STREAM     LISTENING     876      1/init              /               run/systemd/private
unix  2      [ ACC ]     STREAM     LISTENING     2927     1/init              /               var/run/dbus/system_bus_socket
unix  2      [ ACC ]     STREAM     LISTENING     897      1/init              /               run/lvm/lvmetad.socket
unix  2      [ ACC ]     SEQPACKET  LISTENING     903      1/init              /               run/udev/control
unix  2      [ ACC ]     STREAM     LISTENING     905      1/init              /               run/systemd/journal/stdout
root@ev3dev:~/brickfpv-1.0#

dlech, 2015-02-21:
I wonder if the error message is referring to Bluetooth rather than TCP since it comes right after the "Waiting for incoming Bluetooth connection..." message. Is there a place you can select the RFCOMM channel? Maybe @pabr knows?

artpoz, 2015-02-22:
I think I found reason of error. Program was running twice - once after reboot and once running from command line. First run gives no error message of address already in use. Program works on latest version, so I'm closing this issue.

closed 2015-02-22 by artpoz