ev3dev/ev3dev #278: Default root password refused

massimiliano-mantione, 2015-02-20:
I wrote ev3dev-jessie-2014-12-01 on a micro sd card.

After boot, and after successfully setting up USB networking, ssh refuses the default root password given [here](https://github.com/ev3dev/ev3dev/wiki#DefaultRootPassword).

I am sure that networking is up because:
* I can ping the ev3, and
* ssh, on the 1st time, asked me if I trusted the new machine at that IP address.

So this is not a network issue.

I will see if I can mount the sd on a linux box and edit /etc/passwd to reset  the root password, but... am I doing something wrong?

dlech, 2015-02-20:
I has happened a few times to others in the past that the SSH host keys failed to be generated on the first boot. When it asked you if you trusted the new machine, it should have showed you a host key. You can try re-flashing the SD card and try again. As make sure your SD card is not > 32GB.

massimiliano-mantione, 2015-02-21:
It did ask me if I trusted the machine.

BTW, meanwhile I tried various things, none of them worked.

I have been able to mount the micro sd, and adit /etc/passwd removeing the "x" from the root line to reset the password, but I have still been asked for a password.

I also copied my public ssh key on /root/.ssh/authorized_keys hoping to get in with key authemtication, but that did not work, either (of course I invoked ssh as a user with the proper private key in the .ssh dir).

Then I re-downloaded ev3dev (same version), and reflashed it.
Again, the default root password is refused.

Is there another way to reset the password with the SD mounted?

dlech, 2015-02-21:
I have seen one other case where the problem was actually on the host machine configuration. Do you have another computer that you can use to try to connect with? The default root password works for everyone else, so I don't think changing will help.

bzflaglegomaniac, 2015-03-01:
I don't know enough about ssh to explain this but it r00tme didn't work as a password via ssh for me either. After connecting with ssh I accepted the provided keys, but the password failed.

I tried it three times, then used a keyboard on the ev3 directly to confirm that I had it right.

Since I had the keyboard hooked up anyway, I followed the rest of the first use instructions, including setting up a new user and adding the user to the indicated groups.

I was then able to use ssh with the new user name.

After accepting the keys with the new user name, I was also able to log in as root with the r00tme password. I'm guessing that accepting the keys with the new user also reset the keys for root?

dlech, 2015-06-09:
There were some changes to boot order dependencies in the 2015-05-01 release that should help make sure the ssh host key gets setup correctly. Let us know if this is still an issue.

dlech, 2015-10-30:
There will no longer be a default root password in the next release.

closed 2015-10-30 by dlech

rhempel, 2015-10-30:
Sooo - how do we set it :-)

dlech, 2015-10-30:
`sudo passwd`

rhempel, 2015-10-30:
Doh!

Zordid, 2016-04-14:
I got the same problem. First time boot - I cannot log on because I do not know the default root password! But, hey - HOW am I supposed to SET a password myself without being able to get the shell? sudo passwd - yeah, great - but how do I log on???

ddemidov, 2016-04-14:
Login/password is robot/maker.

I tried to find a link to the docs where this is stated, but failed.

Zordid, 2016-04-14:
Thanks - about the same time I found a hint about this, too! ;)
This is really, really awkward. The setup tutorial is so detailed - but THIS IMPORTANT information is simply missing. Just say: log on using ssh robot/maker somewhere and there will be no questions... :) Other than that - great software!

ddemidov, 2016-04-14:
Ok, found it. Every 'connecting to the internet' tutorial has it, but it would make sense to also provide the info in the 'Getting started'.

Zordid, 2016-04-14:
If you are starting like me, you got your WiFi dongle and want to use it. I could easily setup the WiFi connection on the brick itself, so I was instantly able to ping it. Thus, I was stuck because this login info is missing, I would not have searched for it in connection via USB cable as this has nothing to do with USB cables...
Sometimes it is easier to not be too explicit but have a simple overview for experts that don't want to follow a 20-step-by-step-guide just to find this tiny bit of information... :-)

dlech, 2016-04-14:
It is always best to open a new issue instead of commenting on a closed issue (unless you think the issue should be re-poened). So if you think we need to change the website, open a new issue for it. Also pull requests are welcome (it greatly increases the chance of something actually being done about it)!

MbarekErras, 2019-03-04:
ssh robot@ip and use the password "maker", it worked for me

papersail, 2021-03-14:
I also encountered this issue yesterday. As a long time Linux user, I think I subconsciously assumed the user name to be '_root_', instead of supposedly **'robot**'. That's the reason why I couldn't log in with the password.