ev3dev/ev3dev #279: how to use pressure sensor and sharp ir sensor from mindsensors

crowndale1, 2015-02-21:
Hi,

image: ev3dev-jessie-2014-12-01 
3.16.7-ckt4-ev3dev1 #2 PREEMPT Mon Jan 19 21:08:06 CST 2015

How to use pressure sensor and sharp IR sensor from mindsensors in ev3dev? Appreciate any help.

dlech, 2015-02-21:
The best way would be to write a driver for them. See <http://www.ev3dev.org/contributing/#write-a-driver>. For simple sensors, there is not actually any "coding", it is just filling in some data values.

After a quick look at the pressure sensor docs, I would suggest 7 modes

* ABS-PSI: writes 'p' to 0x41, reads from 0x43
* ABS-MB: writes 'b' to 0x41, reads from 0x43
* ABS-KPA: writes 'k' to 0x41, reads from 0x43
* REL-PSI: writes 'p' to 0x41, reads from 0x45
* REL-MB: writes 'b' to 0x41, reads from 0x45
* REL-KPA: writes 'k' to 0x41, reads from 0x45
* RAW: reads from 0x53

And one command

* ZERO: writes 'd' to 0x41

The Sharp IR sensor looks like an Analog/NXT sensor, so you can use the existing generic analog driver. You could also write a new driver that provides a linear value like the lookup table [here](https://github.com/ev3dev/lego-linux-drivers/blob/master/sensors/nxt_i2c_sensor_defs.c#L148).

dlech, 2015-02-22:
Since the pressure sensor was trivial, I have started the driver for it in <https://github.com/ev3dev/lego-linux-drivers/commit/14c655facab821e5f31ae588c1f4cb90e69e6364>. I don't have the sensor though, so someone else is going to have to finish it (or donate a sensor).

kortschak, 2016-04-27:
@dlech I have a PPS58-Nx. What is the easiest way to find the product ID? I can send a PR to finish your addition if you can point me in the right direction.

dlech, 2016-04-27:
I usually use `i2c-tools`. I'm pretty sure this package is installed by default. You can use `i2c-detect` to find the address of the sensor (should be 0x0C - needs to be added to [this page](http://www.ev3dev.org/docs/sensors/i2c-sensor-addressing/)).

Then use `i2c-dump` to find the product name. More info: www.ev3dev.org/docs/sensors/using-i2c-sensors/

Also, I just had  a look at the docs again and it says 

> When the Units are changed, the Reference pressure value is reset to zero

So the `REL-*` modes will probably need an extra helper function to set register 0x47 to standard atmospheric pressure so that the relative reading is actually relative to something other than 0.

kortschak, 2016-04-27:
Where do helpers go (file and ordering within the file relative to other things)?

dlech, 2016-04-27:
The helper function should probably be a [set_mode_post_cb](https://github.com/ev3dev/lego-linux-drivers/blob/14c655facab821e5f31ae588c1f4cb90e69e6364/sensors/nxt_i2c_sensor_defs.c#L266) so that it runs after we write to 0x41.

Since it is just one function, it can go in the `nxt_i2c_sensor_defs` file with the other similar functions for now.

kortschak, 2016-04-27:
I have the product ID and will write the helper tonight (I have rebased your commit onto master to avoid merge conflicts in the PR to come). I can squash the whole thing into one commit or keep the parts separate - what is your preference?

Also, what is the best approach to testing the driver? Is there an workflow to getting just the drivers built and inserted?

dlech, 2016-04-27:
Go ahead and squash it.

Since the driver are loadable modules, my workflow is this:

1. Unload the module I am working on (often requires unloading dependents as well - [example](https://gist.github.com/dlech/c08cd4679a98d4ad1716))
2. Compile the kernel and install all modules to the EV3 via sshfs. This is done automagically with [ev3dev-buildscripts](https://github.com/ev3dev/ev3dev-buildscripts) and some `local-env` hooks.
3. Reload the module and see what happens.
4. If there is a kernel oops, reboot
5. Repeat

Since you are just touching the `nxt_i2c_sensor.ko` module, you could just copy that file using scp.

kortschak, 2016-04-28:
@dlech Please take a look at https://github.com/kortschak/lego-linux-drivers/commit/6aa94d031ba3a848cc28c6190160b4c9123530dc before I take it for a spin (this is my first foray into drivers).

I'll squash once I'm sure it's working.

dlech, 2016-04-28:
Please go ahead and open a pull request. It will make it easier to comment on the changes.

kortschak, 2016-04-29:
It looks like the ev3dev-buildscripts/README.md is a little out of date; it says the build ends up in `../dist`, but it appears that it is put in `build-area`.

dlech, 2016-04-29:
> It looks like the ev3dev-buildscripts/README.md is a little out of date; it says the build ends up in ../dist, but it appears that it is put in build-area.

Pull request or new issue please.

dlech, 2016-04-30:
`fix committed` label only applies to the pressure sensor.

dlech, 2016-05-08:
Pressure sensor driver has been released in [kernel v4.4.9-11-ev3dev-ev3](http://www.ev3dev.org/news/2016/05/08/Kernel-Release-Cycle-11/).

I have opened #625 for the Sharp IR sensor.

closed 2016-05-08 by dlech