ev3dev/ev3dev #280: Add brickman menu for expanding SD card

G33kDude, 2015-02-25:
As well as perhaps a few of the other "Getting Started" items, such as running `apt-get upgrade`

dlech, 2015-02-25:
I wouldn't want to do apt-get upgrade this way because there wouldn't be an easy way to handle errors. Expanding the root partition on the other hand is fairly error free, so it would be a good thing to add. I'm thinking that we would have a separate firstboot program instead of incorporating it into brickman though.

G33kDude, 2015-02-25:
I suppose you're right about apt-get. However, it may still be a good idea to allow user creation/deletion/modification through brickman.

marecl, 2015-03-16:
I will appreciate option "check firstboot script". In some old updates I couldn't do anything with EV3, because apt-get upgrade or dist-upgrade couldn't work for me until I ran firstboot script again

MolarAmbiguity, 2015-04-06:
Perhaps there could be an optional "getting started" wizard, that guides you through the steps to set up ev3dev. This would run on the brick.

dlech, 2015-04-28:
I've been messing with Raspberry Pi a bit lately. They have a nifty program called [raspi-config](http://elinux.org/RPi_raspi-config) that does this kind of stuff. I'm thinking we may just [fork it](https://github.com/asb/raspi-config) and modify it to suit ev3dev.

Regarding the original suggestion related to expanding the root partition: in jessie, this runs so quickly that I am going to just add it to the firstboot script in the next image release and see how that goes, so there is one less thing to configure at the start.

rhempel, 2015-04-28:
Funny you should mention this - a buddy of mine got a 3D printer and it uses an rPi. I had to help him reflash an SD card image, and their standard image is just a tiny bit bigger than a standard 8GB SD card. Long story short, I saw a link to raspi-config and thought, hey we could use that :-)
I'm all for less configuraiton at the start - put the root partition expansion in there if it's not going to take too long.
Maybe for first boot, we (you) can also switch the console font to something bigger so we can let the users know what's going on?

dlech, 2015-04-29:
> Maybe for first boot, we (you) can also switch the console font to something bigger so we can let the users know what's going on?

Until someone fixes plymouth, there is nothing to see during boot besides flashing lights.

dlech, 2015-04-29:
> I will appreciate option "check firstboot script". In some old updates I couldn't do anything with EV3, because apt-get upgrade or dist-upgrade couldn't work for me until I ran firstboot script again

I don't think we need this anymore since we sorted out #267.

dlech, 2015-04-29:
I'm am closing this issue as "wonfix" since I don't think brickman is not the right place for these kinds of things. The display is just too small to let you know what is going on and data entry is too tedious, especially for things like entering strong passwords. Instead, I think we should take the ideas here an put them into `ev3dev-config` (see #316).

closed 2015-04-29 by dlech