ev3dev/ev3dev #281: Motor does not switch direction

aracanid, 2015-02-25:
Hi there, I have a program that waits for commands from a ROS node. When it receives the command it should act by having the robot turn left or turn right. However the robot only seems to move in one direction and not the other. Below is the code I am using as well as some initialisation code.

Init code:
```python
	self.DEFAULT_PULSES_PER_SECOND = 100
	self.DEFAULT_POSITION_SP = 10
	self.DEFAULT_NEGATIVE_POSITION_SP = -(self.DEFAULT_POSITION_SP)
	self.DEFAULT_RUN_MODE = "position"
	self.DEFAULT_HOLD_MODE = "on"
	self.DEFAULT_STOP_MODE = "brake"
	self.DEFAULT_POSITION_MODE = "relative"
	self.DEFAULT_REGULATION_MODE = True
	self.MOTOR_STARTING_POSITION = 0

	self.YAW_LEFT = "yaw_left "
	self.YAW_RIGHT = "yaw_right"
```

Move functions:
```python
def yaw_left(self, angle):
	self.left_track_motor.position_sp = -(angle)
	self.right_track_motor.position_sp = angle
	self.left_track_motor.start()
	self.right_track_motor.start()

def yaw_right(self, angle):
		
	self.left_track_motor.position_sp = angle
	self.right_track_motor.position_sp = -(angle)
	self.left_track_motor.start()
	self.right_track_motor.start()
```

I'm pretty new to using Ev3Dev so not sure what I've done wrong here, any help you could give would be much appreciated.

moriarty, 2015-02-26:
How are you calling the functions? Can I see the ros node code with the callbacks? 

In your init code, I don't know what your 

    self.YAW_LEFT = "yaw_left " 
    self.YAW_RIGHT = "yaw_right"

is used for but one has an extra whitespace..." "

moriarty, 2015-02-26:
Is this the source repository for your project? 

https://github.com/aracanid/offloadable_face_recognition/blob/master/nodes/motor_controller.py

aracanid, 2015-02-26:
Hi Moriarty, you're right I should remove the white space, I've messed up my struct there a bit but I believe it should still work anyway. Also yes that's the code for the motor controller node in ros, the code from above is located here https://github.com/aracanid/offloadable_face_recognition/blob/master/ev3/ev3_motor_server.py

moriarty, 2015-02-26:
The whitespace isn't the issue, now I see how you're using it. "yaw_left " and "yaw_right" are both 9 chars for your packing/unpacking. 

Does it move in one particular direction? only left or only right? Or does it move only the first direction it receives - but both directions work? 

In your ev3_motor_server you have some debug output for the commands it receives in the while loop. Are the commands being received? 

It seems like you're not running the rospy node on the ev3 itself. I'm not familiar with using sockets to communicate with the ev3- I installed ros_comm stacks on the ev3 itself. I jumped on the issue because I'm familiar with ROS but I might not bee much help - it doesn't look like ROS is an issue. 

I would write a small test to check if it's an issue with the sockets or a bug in your ev3 motor code. 

You could try this code: 

	def process_key_input_command(self):

		#conn, addr = self.socket.accept()
		#print "Connection address: ", addr

		while True:
			#data = conn.recv(self.unpacker.size) #data size
			key_in = list(raw_input("a,d,q: "))[0]
			#if not data:
			#	break
			if key_in = 'q':
				exit()
			elif key_in not in ['a','d']:
				break
			else:
				if key_in is 'a':
					data = self.unpacker.pack(*("yaw_left ", 90))
				elif key_in is 'd':
					data = self.unpacker.pack(*("yaw_right", 90))
				command, angle = self.unpacker.unpack(data)
				print command
				print angle
				
				with self.motor_lock:
					if command == self.YAW_LEFT:
						print "YAW_LEFT command received"
						self.yaw_left(angle)		
					elif command == self.YAW_RIGHT:
						print "YAW_RIGHT command received"
						self.yaw_right(angle)
					else:
						print "Unknown command received!"

		#conn.close()

And then just update the main function

    def main(self):
        ev3ms = Ev3_Motor_Server()
        ev3ms.process_key_input_command()

I'm away from my EV3 so I haven't tested that code

aracanid, 2015-02-26:
Hi again, thanks a lot for your help with the issue so far. The reason I'm using the sockets is because I was previously unable to get ROS working with the EV3, it would actually be a great help if you could point me in the right direction for getting that set up. I'm not in my lab at the moment to double check but if I remember correctly it only moves in the direction that it first moves in and then will not reverse to the other direction. So it is able to move both left and right, it just cannot switch between the two once one of these actions has occurred. I'm sure it's something simple I'm missing here haha.

rhempel, 2015-02-26:
Alex, thanks for stepping in here and helping out Justin. If it turns out to be a driver level issue, I'll get on it.

moriarty, 2015-03-01:
@rhempel - No problem. I'm on vacation and away from my ev3 until Tuesday. To be honest, I haven't had as much time to hack with it as I'd like. But at least on vacation I have time to relax and check issues.

@aracanid any updates on this, or progress with debug output? I did install ROS on ev3- with some caveats that I'm
still slowly debugging. On the ev3dev.org website there is a project page- you can find a link to a how to. You need to build ROS from source but it can't be compiled directly on the ev3- thankfully there is "brickstrap" which makes this easy to install everything and then create your own custom ev3dev SD card. But it's all explained in the link. It's a live "how to" hosted in a github repository where I was also planning on putting some sample code.

moriarty, 2015-03-03:
@aracanid - I'm back and had a chance to try your code with my function ```process_key_input_command(self):``` above. 

I had a small error: 

    # if key_in = 'q': # 
    if key_in == 'q': 

But otherwise it worked. So I think it's something with how you're using sockets. 

@rhempel this doesn't appear to be a driver level issue :+1: 

But I have been busy, and don't remember the last time I did a dist-upgrade, so I can try again in the morning after everything is up to date.. 
3.16.7-ckt4-ev3dev1 # 2 PREEMPT Mon Jan 19 21:08:06 CST 2015

aracanid, 2015-03-03:
Hi Moriarty, I'm sorry for the late reply to this, I've been away for a few days and haven't had access to a PC. I appreciate your help in figuring out what was the issue and I hope you had a good vaction :)

closed 2015-03-03 by aracanid

moriarty, 2015-03-03:
What was the issue? 

I just finished updating my ev3 and tried your code again. 

Your original code with the sockets seemed to work. I didn't try the whole thing, but now I'm curious what was wrong. 

This is the code I used to test the socket connection, I just used ipython. 

    import socket
    s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
    s.connect(("192.168.88.195", 5004))
    import struct
    p = struct.Struct('9s I')
    s.sendall(p.pack(*("yaw_left ", 90)))
    s.sendall(p.pack(*("yaw_right", 90)))
    s.sendall(p.pack(*("yaw_left ", 90)))