ev3dev/ev3dev #282: A critical look at the tacho-motor class.

dlech, 2015-02-27:
I am working on writing a driver for the [mindsensors.com NxtMMX](http://www.mindsensors.com/index.php?module=pagemaster&PAGE_user_op=view_page&PAGE_id=134) which of course will use the tacho-motor class. And although I don't have access to one, I am also taking into consideration the [HiTechnic motor controller](http://www.hitechnic.com/blog/wp-content/uploads/HiTechnic-Motor-Controller-Specification.pdf) (I can't write a driver since I don't have one, but just making sure that if someone were to write a driver, it will fit with what we are doing). While I am at it, I am planning on streamlining the tacho-motor class which involves making some breaking changes. So, if anyone has any suggestions for the tacho-motor class, now is an excellent time to make them heard. Here is what I am thinking...

### Speed and position related attributes

It has been suggested that we rename the `position` attribute to `pulses` to make it more obviously related to the `pulses_per_second[_sp]` attributes. (Comes from https://github.com/ev3dev/ev3dev/issues/229#issuecomment-67282237). This was also a point of confusion in #128, so maybe a good idea. Although my inclination is to call the attributes `position` and `speed[_sp]` and just make it clear in the documentation that the units are tachometer "ticks" or "counts". These two words seem to be used much more commonly than "pulses". Thoughts?

Another issue related to speed is that the motor controllers referenced here do not have speed feedback, only position. So, do we try to calculate speed or do we just return `-ENOTSUPP`?

### Commands and Modes

I would like to reconcile the `tacho-motor` class with the concepts of "commands" and "modes" that we have in the `lego-sensor` class. In the `lego-sensor` class, changing a "mode" changes which value(s) we are reading, how we read the value or units of measurement of the value. For example, for the LEGO ultrasonic distance sensors (both EV3 and NXT versions), changing the mode can change the scaling from centimeters to inches or it can change from continuous reading to single-shot reading (how we read it). "Commands" on the other hand tell the sensor to do something that does not change which value we are reading or how the value is read or the units. For example, on the LineLeader sensor, we have commands to calibrate white and black. This may change the actual numeric value that we are reading, but it doesn't change how (i.e. reflected or ambient by turning the LEDs on or off) or the units (it is still % of calibrated range).

One thing we don't have terminology for yet though is passing parameters to commands. This doesn't exist (yet) in the `lego-sensor` class. In the tacho-motor class, if a value is numeric, we are calling these parameters "setpoints", but if they are non-numeric, we are calling them "modes". I guess this is OK, but maybe we could find a different word that means both?

Getting back to motors, I think most attributes in the `tacho-motor` class should actually be called commands rather than modes. Also, there is some opportunity for some consolidation.

First, let's have a look at some commands existing motor driver implementations:

<table>
<tr>
<th>Command</th>
<th>NXT-G</th>
<th>EV3-G</th>
<th>NxtMMX</th>
<th>HiTechnic</th>
<th>ev3dev</th>
</tr>
<tr>
<td>Run forever/unlimited</td>
<td>Called "unlimited". Takes parameters of direction, power, and control (power regulation) enabled/disabled.</td>
<td>Called "On". Takes parameters of power (sign indicates direction).</td>
<td>Called "unlimited" (In NXT-G block). Takes parameters of speed and speed regulation enabled/disabled. Not sure on the units. Also doesn't seem to do anything if speed regulation is disabled.</td>
<td>Two commands called "Run with power control only" and "Run with constant speed". Takes parameter of power or degrees/second (divided by 10).
<td>Called "forever" ("run_mode" attribute). Takes parameters of "duty_cycle_sp" or "pulses_per_second_sp" and "regulation_mode"</td>
</tr>
<tr>
<td>Run for rotations/degrees</td>
<td>Two commands, called "rotations" and "degrees". Takes parameters of direction, ramping, power, control (power regulation), duration (rotations or degrees). Parameters are all relative to current position.</td>
<td>Two commands called "On for Degrees" and "On for Rotations". Takes parameters of power and degrees or rotations. Parameters are all relative to current position.</td>
<td>Three commands in NXT-G, "degrees", "rotations" and "To Encoder Position" (in NXT-G block - two commands or command with extra parameter in I2C). Takes parameters of power, duration (degrees/rotations). Degrees and rotations are relative values and encoder position is absolute value.</td>
<td>Called "Run to position". Takes parameter of max speed (degrees/second) and encoder position (absolute).</td>
<td>Called "position" ("run_mode" attribute). Takes parameters of "duty_cycle_sp" or "pulses_per_second_sp" and "regulation_mode", "position_mode" (relative or absolute).</td>
</tr>
<tr>
<td>Run for time</td>
<td>Called "seconds". Takes parameters of direction, power, control (power regulation enabled/disabled), and duration (seconds).</td>
<td>Called "On for Seconds". Takes parameters of power and seconds.</td>
<td>Called "seconds" in NTX-G or "Timed control" in I2C docs. Takes parameters of speed (regulated mode only?) and seconds (whole numbers only).</td>
<td><i>No timed mode.</i></td>
<td>Called "time" ("run_mode" attribute). Takes parameters of "duty_cycle_sp" or "pulses_per_second_sp" and "regulation_mode", and "time_sp" (milliseconds).</td>
</tr>
<tr>
<td>Stop</td>
<td>Two actions called "Brake" (active) and "Coast"</td>
<td>Two actions called "Brake" (active) and "Coast"</td>
<td>Two actions called "Brake" (passive) and "Coast". Also at the end of the run to encoder command there is the possibility of "Active encoder feedback".</td>
<td>Two special speed values "Off/Brake" (passive?) and "Off/Float". Also, The "run to position" command actively holds the position upon completion until another command is given.</td>
<td>Three "stop_mode" (attribute) values "coast", "brake" (passive) and "hold" (active).</td>
</tr>
<tr>
<td>Reset</td>
<td><i>No reset.</i></td>
<td><i>No reset.</i></td>
<td>Resets all parameters and current encoder value to 0. (Does not reset PID parameters).</td>
<td>Resets current encoder value and set to Off/Braked</td>
<td>Resets all parameters (including PID) and stops motor.</td>
</tr>
</table>

As you can see, things line up fairly well already. However, for better consistency with the lego-sensor class, I am planning on renaming the `run_modes` attribute to `commands` and renaming the `run_mode` attribute to `command`. We will drop the `run` attribute and instead writing a command to the `command` attribute will actually start the command. There will be a new `stop` command to take the place of writing `0` to `run`. Additionally, there will be a `reset` command and we will drop the `reset` attribute.
[... 92 lines cut ...]
The first question that comes to my mind is: Do we return multiple status at the same time or do we have a single state that returns only the highest priority status value. I'm leaning more towards the multiple status, but that leads to another question. Do we have a single attribute that returns multiple state flags or do we have multiple attributes that return a single state. Keeping in mind that the sysfs motto is "one thing per attribute", perhaps we need one attribute per status item. This makes for a lot of attributes, but I have seen in other driver where they add a subdirectory to group together related attributes. So, we could have a `state/` subdirectory that contains a number of attributes based on the table above. What makes the most sense here?

### Ramping

In ev3dev, we have `ramp_up_sp` and `ramp_down_sp` (which could be renamed to `*_ms` to be consistent with the other motor classes). When these are set to non-zero values, the speed will ramp up from 0 to 100% duty cycle over the specified time period. If the duty cycle is limited by the `duty_cycle_sp` or speed regulation mode is on, then the ramp time will be less. For example if we have `duty_cycle_sp` set to 50 (and speed regulation mode is off) and `ramp_up_sp` is 500ms, when we start the motor, it will ramp up from 0 to 50% duty cycle over the span of 250ms and then stay at 50% duty cycle until there is a stop command.

[... cut ...] .. The new type will actually be a new driver class, but we'll keep it in the same `/sys/class/tacho-motor` directory, but will be named "pair0", "pair1"...

I will modify the existing ev3_tacho_motor driver so that output ports B/C will be a pair and output ports A/D will be a pair. The pair will only show up if both ports have the same motor driver loaded (i.e. both are EV3 Large motors). The NxtMMX driver will always create a pair for its motors since it doensn't know about other motor types.

dlech, 2015-02-27:
@rhempel, I know this is an epic novel, but since you are the motor man, I hope you can make some time in the near future to plow through it and offer feedback, otherwise, I'm just going to go for it.

rhempel, 2015-02-27:
I have plowed through most of David, and I'm seeing things that we visited and struggled with months ago, like separating the polarity modes, and distinguishing modes from setpoints.
Overall, the unification makes a lot of sense, especially separating out modes from commands.
I am glad that we had a number of users that worked through issues with the motors, notable the Figelli tacho polarity and the confusion of updating setpoints before the motor was in the idle state.
Let's talk about this tomorrow (maybe we can have a Skype session in the afternoon) and divide up the task - bear in mind I am still working on getting my full dev environment up. I can build the kernel modules, but not a full SD image.
Thanks for taking the time to assemble the backgorund informatio and posting it here for review.
I encourage other users of the existing motor drivers to leave some comments before we plow ahead with this - but in the end I think it's the right direction, minor course corections will be considered, of course.

dlech, 2015-02-28:
> the confusion of updating setpoints before the motor was in the idle state.

I forgot to touch on this. The way the NxtMMX handles this is that uses the parameters that were set when a command was started. If a parameter is changed while the motor is running, it has no effect until a new command is issued. I think the ev3_tacho_motor driver should do the same. When we start a command, make a copy of the parameter variables and uses those in the PID and timer calculations.

Tomorrow afternoon sounds good for Skyping.

rhempel, 2015-02-28:
Cool, I have to go in to the office (day job) in the morning, I have some time free in the afternoon

in4lio, 2015-03-01:
**Motor pairs** also are necessary in case of the remote access to the brick over LAN. The original Lego VM allows mask of output ports in direct commands for this purpose, if I understood correctly, you would like to do something in the same style. Here are some other tentative ways to realize synchronization:

**`slave` attribute** (works only for tacho-motors of same types)
All commands for the host will be retranslated to the slave. _"Run together"_ example:
```
$ ls /sys/class/tacho-motor/
motor2/
motor3/

$ echo motor3 > /sys/class/tacho-motor/motor2/slave
$ cat /sys/class/tacho-motor/motor2/slave
motor3

$ echo 1 > /sys/class/tacho-motor/motor2/run

$ cat /sys/class/tacho-motor/motor2/state
run
$ cat /sys/class/tacho-motor/motor3/state
run
```
Moreover, it allows to create a chain more than two motors.

**`slave` attribute and pending command** (tacho-motors may have different types)
Slave receives the command but executes it only when the host will get any command. _"Host runs the slave stops"_ example:
```
$ echo motor3 > /sys/class/tacho-motor/motor2/slave
$ cat /sys/class/tacho-motor/motor2/slave
motor3

$ echo 0 > /sys/class/tacho-motor/motor3/run
$ cat /sys/class/tacho-motor/motor3/state
run

$ echo 1 > /sys/class/tacho-motor/motor2/run

$ cat /sys/class/tacho-motor/motor3/state
idle
$ cat /sys/class/tacho-motor/motor2/state
run
```

**`group` attribute or an execution of any motor commands synchronously**
_"Different commands for different motors"_ example:
```
$ ls /sys/class/tacho-motor/
motor2/
motor3/

$ ls /sys/class/dc-motor/
motor1/

$ echo 10 > /sys/class/tacho-motor/motor2/group
$ echo 10 > /sys/class/tacho-motor/motor3/group
$ echo 10 > /sys/class/dc-motor/motor1/group

$ ls /sys/class/motor-group/
group10/

$ echo sync > /sys/class/motor-group/group10/mode

$ echo 0 > /sys/class/tacho-motor/motor3/run
$ echo 1 > /sys/class/tacho-motor/motor2/run
$ echo coast > /sys/class/dc-motor/motor1/command

$ cat /sys/class/tacho-motor/motor2/state
idle

$ echo 1 > /sys/class/motor-group/group10/exec

$ cat /sys/class/tacho-motor/motor2/state
run
```

dlech, 2015-03-01:
I actually proposed some master/slave motor attributes in <https://github.com/ev3dev/ev3dev/issues/40#issuecomment-36183880>. However, what I am proposing now is different. In the case of motors controlled via I2C, this sort of chaining will still be less than ideal because it would still send two separate commands to start each motor. Sending the commands takes some amount of time, so the motors would not actually start at the *exact* same time. The controllers I am looking at have special commands to start both motors at the same time, which is what the motor pair will utilize. This is why pairs will not be configurable - because they depend on the specific underlying drivers.

I like the idea of a motor group. I it doesn't seem useful enough to spend time implementing it right now though (well, I'm not going to do it, but pull requests are always welcome - and wait until we have made these other changes). And, I think it should be done in configfs rather than using sysfs attributes. Then (assuming we have made the other proposed changes) it might look like:

```
$ cat /sys/class/tacho-motor/motor0/state/running
0
$ cat /sys/class/dc-motor/motor0/state/running
0

$ cd /sys/kernel/config/motor-group
$ ls
command
dc-motor/
servo-motor/
tacho-motor/

$ cd tacho-motor
$ mkdir motor0
$ cd ../dc-motor
$ mkdir motor0

$ cd ..
$ echo run-forever > command

$ cat /sys/class/tacho-motor/motor0/state/running
1
$ cat /sys/class/dc-motor/motor0/state/running
1
```

Mixing motor types might get a little tricky though since they won't all have the same commands.

MetallianFR68, 2015-03-02:
Hi,
I just read this article.
There are 2 things i really want to see in future versions:
- motor groups (not only pairs)
- ability to detect the FIRGELLI L12 automatically
So I give a big +1 to this suggestions

rhempel, 2015-03-02:
While it would be nice to be able to detect Firgelli motors automatically, I'm afraid that it's impossile because they were made electrically identical to the standard servo motor.
Well, not exactly compatible becuase the motor tacho counts "backwards" relative to the standard LEGO motor, given the same direction setting, which is why we have an invert tacho attribute.
Most ev3dev based creations know which port a motor is plugged into, so it is not too big of a deal to force the Firgelli mode when you start your program.
Is there another specific use case where automatic detection would be a bonus?

dlech, 2015-03-05:
I've been making progress. See commits above.

dlech, 2015-03-05:
To be a little more verbose, I have pretty much made all of the changes I described except for doing something with motor state. Any one that wants can give this a test and see how badly I broke things. :grin:

I'm going to start on the actual implementation of the NxtMMX now and see how it fits into the "new" tacho-motor class.

MetallianFR68, 2015-03-05:
Hi,

I see that you're working hard on the motor class. ! good job !

Here's another feature suggestion : min and max values.

Example : you set a "max position" value for the motor and it stops when it reaches it (or even better it ramps down and stops at the value)

dlech, 2015-03-05:
What you are describing sounds like the run to position commands with a ramp up/down setpoint, which are features we already have (although possibly not working correctly).

rhempel, 2015-03-05:
I think @MetallianFR68 is asking for hard stops so that he can turn the motor on and check position and adjust output duty cycle and direction as long as the motor position is inside the stops. If the motor hits either min or max setpoints - stop the motor.

dlech, 2015-03-05:
OK, I see. I still think you can accomplish this using the run to position command. You just have to check the sign of your duty_cycle_sp to know whether to use the min or max value as the position_sp. Keep in mind that we are modifying the driver so that changing a setpoint while the driver is running has no effect until you send another command.

MetallianFR68, 2015-03-06:
Yes that's what I meant. The idea was to implement LIMITS value.
For example the limit is set to 5000
If i try to send the motor to position 10000, it stops at 5000 (the upper limit)
But I agree this can be done in another way using the already available commands...

dlech, 2015-04-15:
I've just pushed a new kernel package with many of these proposed changes. To save me from having to type everything again, read the [announcement](http://www.ev3dev.org/news/2015/04/15/Kernel-Release-v3.16.7-ckt9-ev3dev1/), check out the new tutorial and let me know how it goes.

A few notable things I have *not* done yet...
1. Anything with ramping - I have no idea if it works, so don't be surprised if it does not.
2. Implemented motor pairs yet, so still nothing available with that.
3. The "stalled" flag does not work with the `legoev3-motor` driver.
4. The settings for the Firgelli actuators need to be looked at [here](https://github.com/ev3dev/lego-linux-drivers/blob/master/ev3/legoev3_motor_defs.c#L41). I don't own one of these, so someone else will have to fix it (or send me one). 

If you find any problems with the "new" tacho-motor class, please open a new issue rather than posting here.

rhempel, 2015-04-16:
I'll look at the new drivers tomorrow or Friday - I have a Firgelli actuator to test with. I also have a bunch of test scripts that need to be rewritten but that's OK

MetallianFR68, 2015-09-14:
Is there any update on motor groups ?
I'm really looking forward for this feature

dlech, 2015-09-14:
Sorry, not yet.

rhempel, 2015-09-14:
I'll probably be poking around and fixing a few small issues in the tacho motor driver soon - the `run-to-abs-pos` is not hitting the target very reliably at all.

ericpascual, 2015-10-18:
First of all, kudos to everybody in the ev3dev team for such an amazing work. I've been using LeJOS a lot, (and contributing a small bit to) but I really enjoy having Debian running on the EV3, and being able to program it with languages such as C/C++ and Python (I manage a volunteer association in amateur robotics, and could note that the last one being IMHO the best suited for teaching programming to youngsters). Once again, many thanks to all of you.

While discovering and experimenting with the latest ev3dev version, using Python for programming, I could observe a constant behaviour with `run-to-abs-pos` which is perhaps the one you are referring to in the previous post. Note that the programming language has nothing to do here, since echoing to the `/sys/sclass/tacho-motor` pseudo files gives the same result.

When the `stop` command is `brake`, the final position is always past the target. When `stop` is set to `hold`, it reaches the target perfectly, but with an overshoot every time (i.e. it first pass the target and then goes back to it). I have tested with the large motor only for the moment, but using various speeds (from 50% to 100%). 

I haven't played with the PID settings, not being an expert in the domain, but I guess it would be possible for people mastering the topic to find the right triplet for reaching the target without overshoot. Maybe this overshoot has something to do with the brake stop behaviour.

Observing the behaviour in brake stop mode, it seems that the motor starts to decelerate when reaching the target position. Even with no speed down ramp, this has chances to too late for stopping at the right position. But I cannot ensure that the observation is precise enough.

I'll try to have a look to the source code of motor regulation, but am not sure to be competent enough for finding clues. In case I could, I'll share them here of course.

rhempel, 2015-10-18:
Note that the Python API is under active development right now and may change slightly.
The overshoot depends a lot on things like motor loading, speed, how far the absolute position is from the start position and so on. If you really need the precision, `hold` mode is the way to go, and you can also play with the `ramp_down_sp`. For fine tuning, the PID parameters may help along with a tutorial that you can probably find via Wikipedia.
Some users just read the `position` property and handle the motor position themselves.
I may look at the motor drivers in between other tasks.

ericpascual, 2015-10-18:
Thanks Ralph for replying so quickly. 
I too came to the conclusion that the hold mode was the way to go for the moment.
By the way, I've found somewhere a proposal related to linking motors in some master-slave relation. It would be great since I could also notice a small glitch at the end of straight moves, one motor stopping a small delay after the other, leading to the robot heading getting wrong. This is maybe caused by the small delays created by controlling motors with separate calls.

dlech, 2015-12-12:
This issue is getting rather old. Most items have been implemented, but there are a few things still left to do. I am opening individual new issues for these so we can close this one.

Ramping: #453 
Motor pairs (aka synchronized starting of motors): #454
Stall detection on EV3: #455

If I left out anything important, feel free to open a new issue.

closed 2015-12-12 by dlech