ev3dev/ev3dev #283: ssh connection timeout

LeiG, 2015-03-01:
I tried to install the ev3dev-jessie-2015-02-24 and followed the [http://www.ev3dev.org/docs/tutorials/setting-up-ethernet-over-usb-on-mac/] to set up connection. 

Everything works fine and ev3dev was successfully connected after Step 6 (192.168.2.1 as suggested). I then jumped to Step 8 and got my EV3 connected too. 

But, when I tried *ssh*, I got the following message:
```
 ssh: connect to host 192.168.2.3 port 22: Operation timed out
```
Tried *ping*:
```
user: ping 192.168.2.3
PING 192.168.2.3 (192.168.2.3): 56 data bytes
Request timeout for icmp_seq 0

```

I am not sure what happened. Any inputs are appreciated!

LeiG, 2015-03-01:
BTW, tried to reboot as well.

bzflaglegomaniac, 2015-03-01:
I just bought an ev3 yesterday and I'm having the same issue.

The first clue that something is different is that, in step 2, there is no "networking" option in the main menu. You can find the "Gadget" checkbox listed in the "Tethering" menu off the main menu.

Having set the Gadget checkbox, and following all the other steps, I could not connect. Nor could I ping the ev3.

On the computer I manually set the IP address for USB0 to 10.42.0.1, but I still couldn't ping the ev3.

I then plugged a usb keyboard into the ev3 and hit ctrl-alt-F1 to bring up a terminal window, entered username=root and password=r00tme (note that it's "r/zero/zero/t/m/e") and logged in.

On the ev3, running...

ifconfig usb0

... showed the ev3 had the address of 10.43.0.3, as expected, but I could not ping my computer (ping 10,43,0.1 said the network was unreachable).

Solution:

On a hunch, I went back to the ev3's GUI interface (ctrl-alt-F2) and under Tethering, I turned gadget off.

I then went back to the terminal window (ctrl-alt-F1) and confirmed I could ping my PC.

From my PC I confirmed I could ping the ev3.

Trying to connect the ev3 with ssh worked up until the root password (r00tme) was rejected, even though it worked in the terminal (I logged out and back in to confirm).

There is another bug report that addresses this problem. I've only just arrived at this point, so I haven't read it yet.

My work-around was to use the keyboard plugged into the ev3 with the ctrl-alt-F1 terminal window to complete the steps listed in the Getting Started page.

Doing that, I created a user account for myself, after which I was able to connect to the ev3 via ssh using the new username and password and I now have an ev3 command prompt available on my PC.

Hope that helps.

cho934, 2015-03-01:
After installing the ev3dev-jessie-2015-02-24, same issue using Ethernet over USB on Linux Ubuntu 14.04. No ping and no ssh connection. 
I don't have a compatible USB keyboard, so I search another solution.

dlech, 2015-03-01:
In reply to https://github.com/ev3dev/ev3dev/issues/266#issuecomment-76603778, I said:

>  haven't had time to updated the documentation yet, but when using tethering, your EV3 acts as a network server instead of a client. You cannot use both tethering and "All Network Connections" connection at the same time. If you need to access the Internet over USB, do not use tethering. If you want no configuration on your host computer, do use tethering. When tethering is enabled, you host computer will be automatically assigned an IP address from the EV3 and will be able to connect to the address specified in the tethering network info.

I will look into this some more today and see if I can repeat the problem you are having and update the Getting Started documentation.

cho934, 2015-03-01:
ok, effectively it's working for me without activating Tethering > Gadget

rex-j, 2015-03-01:
I just unchecked Tethering/Gadget as dlech said above, and problem solved.

LeiG, 2015-03-01:
Thank you guys. Unchecked Tethering > Gadget and problem solved as well.

dlech, 2015-03-01:
I've updated the documentation on setting up the Ethernet over USB connection to remove the misleading "enable gadget" step that was from the previous version of brickman. I think we can close this issue.

closed 2015-03-01 by LeiG