ev3dev/ev3dev #287: Documentation contributors wanted.

dlech, 2015-03-02:
I have just created a couple new pages on Bluetooth network connections for Mac, [Internet connection sharing](http://www.ev3dev.org/docs/tutorials/setting-up-ethernet-over-bluetooth-on-mac/) and [tethering](http://www.ev3dev.org/docs/tutorials/using-bluetooth-tethering-on-mac/).

It would be nice if someone could copy these and modify them for Windows.

And if you are really ambitious, Linux as well. I couldn't get Bluetooth working on Ubuntu 14.10 using the default System Settings tools, so it will probably have to be command line instructions (for BlueZ 4).

moriarty, 2015-03-05:
On Arch Linux I also wasn't able to get it working through the Gnome environment, I didn't put any time into trying via CLI because I have BlueZ 5, and it makes sense to target Ubuntu 14.04.

I'll try to get my hands on a USB bluetooth adapter and try Ubuntu. As for Windows- I won't be of much help.

Also- what are you using to take screenshots of the brick?

rhempel, 2015-03-05:
I think he's just doing grabs of the framebuffer .... and then running them through a program to get the colors to look like an old school LCD. They look great!

dlech, 2015-03-05:
On Linux, I would think that it should be easy (as in using GUI only) to get a connection if you enable tethering on the EV3 (EV3 is "server", host computer is "client"). The other way around though (EV3 is "client", host computer is "server"), you will have to set up a network bridge interface and then use bluez-tools (assuming BlueZ 4) to set up a network connection.

For the screenshots, I am actually using the "desktop test" version of brickman. It is the brickman UI with a different back end that lets me fake any information on the screen to get exactly the screenshot I want. You can read more about it [here](https://github.com/ev3dev/brickman).

Alternately, you can use `fbgrab` on the EV3 to get a screenshot saved to a png file. But then to make them look like the other screenshots, you will have to change white to `#adb578` and scale to 2x without using any interpolation.

One should be able to reuse most of the EV3 screenshots that are already available though.

jhi0417, 2015-03-11:
I am working on this part to connect EV3 with windows. Once I am successful, I will post things here; by the way, it seems like pairing is successful however I can't get EV3 connected to the wireless network that I am sharing - any suggestions?

dlech, 2015-03-11:
> by the way, it seems like pairing is successful however I can't get EV3 connected to the wireless network that I am sharing - any suggestions?

Please open a new issue for this and include more details (see [issues](http://www.ev3dev.org/issues/)).

And thanks for working on the Windows tutorial :+1:

aacco, 2018-05-03:
I don't know whether below info is needed... Anyway, I'll comment.
Connecting via bluetooth(blueman) in Archlinux succeeded:
  Archlinux(pc) (May 3, 2018)
  brickman v0.8.1
  kernel 4.4.87-22-ev3dev-ev3
Note: Archlinux requires additional packages: net-tools, dnsmasq
I'm beginner, so it was hard to find above and I had trouble connecting.

dlech, 2018-05-03:
Feel free to create a new page on the [wiki](https://github.com/ev3dev/ev3dev/wiki) for Bluetooth and add this information.

aacco, 2018-05-04:
I added.
[Archlinux packages for bluetooth connecting](https://github.com/ev3dev/ev3dev/wiki/Archlinux-packages-for-bluetooth-connecting)
Please contact me if anything is faulty.

GiovanniSM20, 2018-06-23:
@aacco https://wiki.archlinux.org/index.php/bluetooth ?

dlech, 2018-06-23:
The page got moved: https://github.com/ev3dev/ev3dev/wiki/Bluetooth

aacco, 2018-06-24:
Sorry, I forgot revising my comment.

bemxio, 2020-03-03:
@dlech can i help with bluetooth tethering on windows?

dlech, 2020-03-03:
You can certainly try. However the best information I could find is that it only works with Windows phone and not any other version of Windows 10. There are settings in "mobile hotspot" that make it look like it should be possible to share an Internet connection over Bluetooth, but I've never been able to make it work.

bemxio, 2020-03-03:
ok, so:
1. how to do it?
2. i mean tethering so no internet, and there isn't a tutorial for bluetooth tethering on website

dlech, 2020-03-03:
"tethering" in the context of the EV3 brick means sharing the Internet connection of the EV3 brick. So, for example, if the EV3 brick was connected directly to the Internet with Wi-Fi, tethering can be used to share that connection from the EV3 to other devices via USB or Bluetooth. I'm guessing this is probably not what you want to do.

bemxio, 2020-03-03:
i don't wanna argue, but the official page says different:

> **Tethering to a Host PC (no Internet access)**
If you don’t need to connect your brick to the Internet, you can use a technology called “tethering” to connect it directly to a PC. This can sometimes require less configuration than other options. Some of these guides may be partially incomplete on certain platforms.

dlech, 2020-03-03:
Try it and let us know what you find out :smile:

bemxio, 2020-03-03:
ok, maybe just not. i lost the track now