ev3dev/ev3dev #289: Connecting in Linux Mint

MolarAmbiguity, 2015-03-06:
I've installed ev3dev on my brick, and was able to connect fine under Ubuntu. Now when I try in Mint I cannot connect over both USB and Bluetooth. Has anyone had any success in Mint before?

rex-j, 2015-03-06:
Yes, I am using Mint 17.1 and I can connect by USB.

MolarAmbiguity, 2015-03-06:
It seems I cannot find the option in the ipv4 settings called "shared to other computers". Could you see this option?

dlech, 2015-03-06:
Which version of Mint are you using? 17.x? Cinnamon? MATE? LMDE?

MolarAmbiguity, 2015-03-06:
1.7.1 32-bit Cinnamon

dlech, 2015-03-06:
Looks like you have to go to "Network Connections" instead of "Network Settings", then you can find "shared to other computers".

MolarAmbiguity, 2015-03-06:
Ok that worked (after I fiddled around with turned of tethering). I will update the tutorial with this info.

closed 2015-03-06 by MolarAmbiguity

dlech, 2015-03-06:
FYI, I sent a pull request to fix Network Settings in Linux Mint.

https://github.com/linuxmint/cinnamon-control-center/pull/109

MolarAmbiguity, 2015-03-06:
Ah ok cool, I'll hold off for now. Should probably just make sure people are using the latest version of cinnamon control center.

dlech, 2015-05-26:
Bug also exists in gnome-control-center.

https://bugs.launchpad.net/ubuntu/+source/gnome-control-center/+bug/1385674

dlech, 2015-07-01:
I saw that Mint 17.2 was released today. I wonder if my patch was included?