ev3dev/ev3dev #290: Enabling Bluetooth by on default?

andybets, 2015-03-07:
Is it possible/easy to change some configuration setting so that Bluetooth is enabled by default when the BrickMan menu loads?  Starting a script automatically would be really great too - if possible.

dlech, 2015-03-07:
What do you mean by "enabled"? Do you mean powered on? If so, once you power it on, it should stay powered on across reboots. If not, this is a bug. If by "enabled" you mean connected to something else, please explain further.

andybets, 2015-03-07:
Hi David - Yes, I think it's must be a bug then.  I'm having to go to 'Network' and select 'Bluetooth' every time the BrickMan menu loads up.

dlech, 2015-03-07:
It sounds like you don't have the latest version of brickman - not that it would make a difference for this issue. Please upgrade your packages (at least `brickman`, `connman` and `bluez`) and may sure you install the new packages `iptables` and `ev3dev-connman-config` and then try again.

Then have a look at `journalctl -r` to see if there is anything related connmand or bluetoothd that may give a hint as to why it is not staying powered on.

andybets, 2015-03-07:
Thanks.  I'll have a look in the next few days, and get back to you.  What about automatically running scripts automatically when it loads?  What's the best way to arrange it?

dlech, 2015-03-07:
Search is your friend: https://github.com/ev3dev/ev3dev/issues?q=is%3Aissue+startup+script

dlech, 2015-05-21:
Is this still an issue with the latest release?

andybets, 2015-05-21:
Sorry, not sure. I haven't had a chance to work on the project since we
last spoke.

DataDevil Limited (Company Number: 8979820) Telephone: 016973 32492 |
Website[2] Registered Office: DataDevil Limited, Fairview House,
Victoria Place, Carlisle, Cumbria, UK CA1 1HP

DataDevil Limited has taken reasonable steps to ensure that outgoing
communications do not contain malicious software. Please carry out your
own checks on this email before accepting it and opening attachments.
Any views or opinions expressed may not necessarily represent those of
DataDevil Limited.

Links:

  1. https://github.com/ev3dev/ev3dev/issues/290#issuecomment-104113264
  2. http://www.datadevil.co.uk

dlech, 2015-05-21:
OK. I'm going to close the issue since it is working for me and presumably others. Feel free to reopen if you find the problem still exists next time you have a chance to use ev3dev.

closed 2015-05-21 by dlech