ev3dev/ev3dev #291: What is the smallest SD card you know that will work?

sharived, 2015-03-08:
I am looking at buying SD cards in bulk for my kickstarter and I was wondering if you know what is the smallest (cheapest) size that will work.  I currently use 4GB for development and it is fine.  Do you think 2GB will work for non-development users or should I stick with 4GB?

dlech, 2015-03-08:
Technically, our current install image will fit on a 1GB card, but it doesn't leave much room for installing too many extras. 2GB should be more than plenty for most people.

dlech, 2015-03-08:
Another thing to consider is random access read/write speeds. Most cards only advertise sequential read/write speeds, not random, so you have to dig a bit. If you buy the cheapest card and it has really bad random read/write times, your customers won't be very happy because it will be really slow. (Compared to the fastest SD card which are only kind of slow. :wink:)

cho934, 2015-03-09:
Exactly, and try to use SDCard Class 10.

sharived, 2015-03-10:
Thanks!  I'll try a few 2GB cards.

closed 2015-03-10 by sharived