ev3dev/ev3dev #292: Bluetooth in Mint

MolarAmbiguity, 2015-03-11:
When I try to connect in mint 17.1 32-bit cinnamon edition, I get the error:
'GDBus.Error:org.bluwz.Error.Failed: No such file or directory. This error is displayed on the brick.

Bluetooth is turned on and when I do sudo apt-get install bluez on both my laptop and EV3 I am told that I have the latest version.

dlech, 2015-03-11:
Setting up Internet connection sharing using Bluetooth on Linux is not trivial. This is why we [don't have a tutorial](http://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-bluetooth/?tabs-0=ubuntu-14-04). You have to set up a network bridge and then setup a Bluetooth network service that uses it on your host computer. `bt-network` from the `bluez-tools` takes care of the latter part, but you will have to do some research to figure out how to setup the network bridge (bluez does not do this).

Alternately, if you don't need to access the Internet from your EV3 over Bluetooth, then you could [use tethering](http://www.ev3dev.org/docs/tutorials/using-bluetooth-tethering/) instead. We don't have any Linux-specific instructions for it yet, but you can probably figure it out from the Mac instructions.

MolarAmbiguity, 2015-03-18:
Ah ok, I will have a go at setting up a network bridge and service. I think I was having the same issue as sharived in issue https://github.com/ev3dev/ev3dev/issues/222. I suggest a renaming of the bluetooth options in brickman?

rhempel, 2015-03-18:
Any particular suggestions .... ?

MolarAmbiguity, 2015-03-19:
"Pair with bluetooth gadget"
"Pair with a device that is not a computer"
Something along those lines.

dlech, 2015-03-20:
Pairing any Bluetooth device works the same no matter what you are connecting to, so there is no reason to have multiple options that do the exact same thing.

MolarAmbiguity, 2015-03-20:
I got the impression that you only pair from the EV3 if you are using a bluetooth keyboard or mic? To connect to a computer you pair from there. I might be a bit confused.

dlech, 2015-03-20:
With Bluetooth, you can initiate pairing from any device that supports it. Keyboards and headsets tend to be so simple that they cannot initiate pairing, but the EV3 is basically a regular computer, so it can do just about anything with Bluetooth (assuming you have the software for it).

MolarAmbiguity, 2015-10-20:
Sorry for the late response...

I can pair with my ev3 using blueman as per the ubuntu tutorial.

closed 2015-10-20 by MolarAmbiguity