ev3dev/ev3dev #293: "WARNING: Duplicate VG name" when attempting to expand SD card

dhmmjoph, 2015-03-15:
When attempting to expand the SD card as per the [Getting Started Guide](http://www.ev3dev.org/docs/getting-started/#grow-your-root-file-system), I get the following error:

````
root@ev3dev:~# lvextend --extents +100%FREE --resizefs /dev/ev3devVG/root /dev/mmcblk0p3
  WARNING: Duplicate VG name ev3devVG: YGHnF0-pRCC-SWRE-je7i-bCRs-RKwV-hdNn8F (created here) takes precedence over 42ekjh-ky3F-IEoo-MyyS-taUY-JSvH-Vmy0is
  Couldn't find device with uuid kT4fXU-GFed-V2hs-W0OU-iPXa-5Ekn-JDbfa1.
  Cannot change VG ev3devVG while PVs are missing.
  Consider vgreduce --removemissing.
  WARNING: Duplicate VG name ev3devVG: YGHnF0-pRCC-SWRE-je7i-bCRs-RKwV-hdNn8F (created here) takes precedence over 42ekjh-ky3F-IEoo-MyyS-taUY-JSvH-Vmy0is
root@ev3dev:~# 
````

I've attempted to rewrite the latest image to the SD card 2 or three times now, each time formatting the card beforehand, with the same result each time.

dlech, 2015-03-15:
What size of SD card are you using?

What are the results from Step 4.5 in Getting Started?

What is output of `pvdisplay` on the EV3?

dhmmjoph, 2015-03-15:
I am using an 8GB SD Card of class 2 or 4, I forget which.

I went through the steps you outlined in #267 (as linked to in step 4.5) and my results were as follows:
1. I was able to connect via SSH.
2. My /etc/modprobe.d/usb-net.conf was correct.
3. Running ```sudo lvscan``` returned the following:
 ```
 root@ev3dev:/# sudo lvscan
   WARNING: Duplicate VG name ev3devVG: YGHnF0-pRCC-SWRE-je7i-bCRs-RKwV-hdNn8F (created   here) takes precedence over 42ekjh-ky3F-IEoo-MyyS-taUY-JSvH-Vmy0is
   WARNING: Duplicate VG name ev3devVG: YGHnF0-pRCC-SWRE-je7i-bCRs-RKwV-hdNn8F (created here) takes precedence over 42ekjh-ky3F-IEoo-MyyS-taUY-JSvH-Vmy0is
   WARNING: Duplicate VG name ev3devVG: Existing YGHnF0-pRCC-SWRE-je7i-bCRs-RKwV-hdNn8F (created here) takes precedence over 42ekjh-ky3F-IEoo-MyyS-taUY-JSvH-Vmy0is
   Couldn't find device with uuid kT4fXU-GFed-V2hs-W0OU-iPXa-5Ekn-JDbfa1.
   inactive          '/dev/ev3devVG/root' [7.30 GiB] inherit
   ACTIVE            '/dev/ev3devVG/swap' [64.00 MiB] inherit
 root@ev3dev:/# 
 ```
 and my ```/etc/fstab``` looks like this:
 ```
 root@ev3dev:/etc# cat fstab
 #<file system>            <mount point>   <type> <options>                           <dump> <pass>
 /dev/mmcblk0p1            /media/mmc_p1   vfat   defaults,errors=remount-ro,noatime  0      1
 /dev/mapper/ev3devVG-root /               ext3   defaults,errors=remount-ro,noatime  0      1
 proc                      /proc           proc   defaults                            0      0
 root@ev3dev:/etc# 
 ```
My ```/etc/flash-kernel/db``` exactly matched the one in the issue.

The output of ```pvdisplay``` was as follows:
```
root@ev3dev:/# pvdisplay
  WARNING: Duplicate VG name ev3devVG: YGHnF0-pRCC-SWRE-je7i-bCRs-RKwV-hdNn8F (created here) takes precedence over 42ekjh-ky3F-IEoo-MyyS-taUY-JSvH-Vmy0is
  Couldn't find device with uuid kT4fXU-GFed-V2hs-W0OU-iPXa-5Ekn-JDbfa1.
  --- Physical volume ---
  PV Name               unknown device
  VG Name               ev3devVG
  PV Size               867.94 MiB / not usable 3.94 MiB
  Allocatable           yes (but full)
  PE Size               4.00 MiB
  Total PE              216
  Free PE               0
  Allocated PE          216
  PV UUID               kT4fXU-GFed-V2hs-W0OU-iPXa-5Ekn-JDbfa1
   
  --- Physical volume ---
  PV Name               /dev/mmcblk0p3
  VG Name               ev3devVG
  PV Size               6.52 GiB / not usable 4.00 MiB
  Allocatable           yes (but full)
  PE Size               4.00 MiB
  Total PE              1669
  Free PE               0
  Allocated PE          1669
  PV UUID               8JaI65-Nm1A-5IDC-9s3G-JY4p-sHKQ-66FjUJ
   
  WARNING: Duplicate VG name ev3devVG: Existing YGHnF0-pRCC-SWRE-je7i-bCRs-RKwV-hdNn8F (created here) takes precedence over 42ekjh-ky3F-IEoo-MyyS-taUY-JSvH-Vmy0is
  --- Physical volume ---
  PV Name               /dev/mmcblk0p2
  VG Name               ev3devVG
  PV Size               867.94 MiB / not usable 3.94 MiB
  Allocatable           yes (but full)
  PE Size               4.00 MiB
  Total PE              216
  Free PE               0
  Allocated PE          216
  PV UUID               kw5jbH-ngBx-HgNq-44d0-ypDx-lHz9-zu0iTF
   
root@ev3dev:/# 
```

dlech, 2015-03-16:
That's weird. I wonder if it is picking up an old partition table or something. Flashing the SD card should write over the partition table, but strange things happen sometimes.

You could try remove the offending extra partition using `pvremove`. (`pvremove kT4fXU-GFed-V2hs-W0OU-iPXa-5Ekn-JDbfa1` maybe?)

You could also try filling the whole SD card with zeros (using `dd`) and then flash the image again.

dlech, 2015-03-27:
Closing as duplicate of #267. I think this is the root cause of that issue.

Also, for the record, here is how we fixed a broken system in a round about way: http://ev3dev.org/irclog/#id-16571

closed 2015-03-27 by dlech