ev3dev/ev3dev #294: Bluetooth Internet Sharing

marecl, 2015-03-16:
I was having a lot of fun triyng to get into my EV3 without PC.
Firstly I plugged my phone as wired AP. It worked. Yes. It's good, but it's consuming too much power.
The second idea - bluetooth, and there are problems.
---IP adress is always different (pretty annoying, but easy to repair. I think so...)
---Bluetooth name is stuck at "ev3dev"
---When I want to setup new connection I must remove host device pair (my phone from EV3) and remove EV3 pair from my phone. Then I must pair both and just after pair says "connected" I can go fast to internet setup, blah blah, but this is ok. When I disconnect and try to connect again EV3 shows an error "GDBus.Error:org.bluez.Error.Failed: No such file or directory"
*UPDATE* I can go to Network Setup every time, but to connect properly (no timeout, no errors ot sth) both devices MUST be set as visible.

I don't care about third one much, but taking my android AP every time to work (switch on bt, visibility, port forwarding and stuff) is very tiring

dlech, 2015-03-16:
Have you tried enabling Bluetooth tethering on the EV3?

marecl, 2015-03-16:
Here are some new infos:
When I reboot EV3 brick I cant connect to phone by bluetooth (connect returns an error, "Network Connection" button dosent work)
Tethering from EV3 - once enabled, you must reboot do disable (ev3 won't connect by bluetooth to anything). 
To connect after reboot - remove phone-brick and brick-phone pair. Then pair again and connect (or it will connect automatically)
---
Tethering from EV3 works but in my situation it's rather for offline work. I need internet connection now because I'm just after reflash and I cant use PC much, so I will have running client on phone

dlech, 2015-03-16:
I think there is a bug in the ConnMan network manager that we use related to Bluetooth connections. I have notices some things like this before, so I will see what I can do about it.

dlech, 2015-04-18:
Some things I failed to address:

> -IP adress is always different

You can set a static IP address on the EV3.

> Bluetooth name is stuck at "ev3dev"

It uses the network hostname for the Bluetooth name, so running `sudo hostnamectl set-hostname <new-name>` will change the name.

ensonic, 2015-09-08:
If there is space we could show the hostname in the brickman title too. Thats what mindstorms and lejos is showing.

closed 2017-10-03 by dlech