ev3dev/ev3dev #298: is ev3dev-simulink connection possible?

cemyclgn, 2015-03-24:
Hi All. We used the ev3dev brickman program on lego mindstorm ev 3 to reach from matlab\ simulink to robot via wifi. We reached from computer to robot by pinging the ip address of it. I appreciated it. No netgear obligation. However, the connection from matlab/ simulink to the robot cannot be established. Is ev3dev-simulink connection possible? Any suggestion or advice would be very helpful. Thanks a lot in advance.

dlech, 2015-03-25:
How does simulink communicate with other devices like the EV3? Do you have a link or code sample you could share?

cemyclgn, 2015-03-25:
[image]
[image]

I want simulink communicate with lego ev3. So, I am able to try our control algorithms on it. I tried to explain my problems in pictures.

dlech, 2015-03-25:
It looks like this software is made to work with the EV3 running the official LEGO firmware. In ev3dev, we have rewritten all of the drivers, so it won't work with simulink. Your options are to write a compatible VM that runs on ev3dev or write your own simulink communication module (or whatever it might be called).

closed 2015-04-16 by dlech