ev3dev/ev3dev #299: Two WeDo USB Hubs confusing lego-port/port instances

JorgePe, 2015-03-24:
Not really a problem but I noticed something strange today, as I'm using my second WeDo USB hub again - using python and WeDoMore but cannot use both WeDo so fast workaround is calling bash scripts.

"First WeDo" has just one PF M motor at first port, "Second WeDo" has one PF M motor at each port.
Using 3.16.7-ckt6-ev3dev1

dmesg:
```
[   20.810526] dc-motor motor0: Bound to device 'port8'
[   20.811728] dc-motor motor1: Bound to device 'port10'
[   20.812762] dc-motor motor2: Bound to device 'port11
```

but `ls /sys/class/lego-port/` :
```
port0  port1  port10  port11  port2  port3  port4  port5  port6  port7  port8  port9
```

where are pot10 and port11?

They seem to exist:

`ls /sys/bus/usb/devices/1-1.4.3:1.0/lego-port`:
```
port8  port9
```
`ls /sys/bus/usb/devices/1-1.4.4:1.0/lego-port`:
```
port10  port11
```

and 

`ls /sys/class/dc-motor/ -l`:
```
motor0 -> ../../devices/platform/ohci.0/usb1/1-1/1-1.4/1-1.4.3/1-1.4.3:1.0/lego-port/port8/motor0
motor1 -> ../../devices/platform/ohci.0/usb1/1-1/1-1.4/1-1.4.4/1-1.4.4:1.0/lego-port/port10/motor1
motor2 -> ../../devices/platform/ohci.0/usb1/1-1/1-1.4/1-1.4.4/1-1.4.4:1.0/lego-port/port11/motor2
```

As I said not really a problem because I can use my bash scripts echoing to `/sys/class/dc-motor/motorx/duty_cycle_sp` as intended.
But I'm wondering if the reason for port10 and port11 not showing up at the file system can be related to using two WeDo's.

rhempel, 2015-03-24:
but `ls /sys/class/lego-port/`
`port0  port1  port10  port11  port2  port3  port4  port5  port6  port7  port8  port9`
Look hard :-)
They are right between `port1` and `port2` - wecome to unexpected sort order in `ls`!

JorgePe, 2015-03-24:
Feeling soooo stupid. I tried only once to get to port 10 after dmesg and made a typo and after that was totally blind.

closed 2015-03-24 by JorgePe